lvgl/lvgl #4261: issue with events for objects lvgl

EMABRN17, 2023-05-30:
### Introduce the problem

Hi developers, I am trying to add an event to lvgl objects, I generated the interface for arduino using SquareLine_Studio, if I don't add the event the compilation and display works properly, but I can't add the event attached to the component

### Examples and cases

This is the part of the code that gives problems:
void screen1_switch1_event_handler(lv_obj_t *switch_btn, lv_event_t event)
{  
        if(event == LV_EVENT_VALUE_CHANGED) {
        Serial.printf("State: %s\n", lv_switch_get_state(switch_btn) ? "On" : "Off");
    }
}

This is the code to link the event to this object(it is declared in another function).

ui_Screen1_Switch1 = lv_switch_create(ui_Screen1);
    lv_obj_set_width(ui_Screen1_Switch1, 70);
    lv_obj_set_height(ui_Screen1_Switch1, 35);
    lv_obj_set_x(ui_Screen1_Switch1, -155);
    lv_obj_set_y(ui_Screen1_Switch1, -40);
    lv_obj_set_align(ui_Screen1_Switch1, LV_ALIGN_CENTER);
    lv_obj_set_event_cb(ui_Screen1_Switch1, screen1_switch1_event_handler);

I initialized the serial in the ui.ino file, and included the ui.c file inside the .ino file
I get this error in compilation:

 In function 'screen1_switch1_event_handler':
ui.c:34:18: error: invalid operands to binary == (have 'lv_event_t' {aka 'struct _lv_event_t'} and 'int')
         if(event == LV_EVENT_VALUE_CHANGED) {
                  ^~
ui.c:35:9: error: 'Serial' undeclared (first use in this function)
         Serial.printf("State: %s\n", lv_switch_get_state(switch_btn) ? "On" : "Off");
         ^~~~~~

I attach the repository for those who would like to explore further
https://github.com/EMABRN17/lvgl_esp32_arduino_squareline_studio.git

I have made several attempts but don't know how to solve it, I am using a 4.3inch ESP32-4827S043 model display

pete-pjb, 2023-05-30:
Hi @EMABRN17 ,

It is preferred  to post questions on the Forum [here](https://forum.lvgl.io/) as Github is normally used for LVGL development related discussions.  You can log into the Forum with your Github account if you like. 🙂 

With regard to your question the ```lv_event_t``` type is a structure so the simplest solution is to use the provided API function to return the event code as follows:

```c
lv_event_code_t code = lv_event_get_code(event);
if(code == LV_EVENT_VALUE_CHANGED) {
...
```
Check the online documentation [here](https://docs.lvgl.io/8.3/overview/event.html?highlight=lv_event#events) for all the details...

I hope that helps...

Kind Regards,

Pete

pete-pjb, 2023-06-01:
Hi @EMABRN17 ,

I am going to assume you have solved your problem now and close this issue.  Please create a new question in the [Forum](https://forum.lvgl.io/) should you still have a problem. 🙂 

Kind Regards,

Pete

closed 2023-06-01 by pete-pjb