lvgl/lvgl #4263: Random hard fault, stuck in lv_obj_get_child_cnt(); need expert help

Mountain03, 2023-05-31:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

[image]

kisvegabor, 2023-05-31:
Hi,

A few things that you can try:
- if possible run to UI on PC with ASAN using standard malloc/free. It should find all memory corruptions.
- enable LVGL asserts with `LV_USE_ASSERT_...   1` in `lv_conf.h`
- enable the logging of LVGL as described [here](https://docs.lvgl.io/master/porting/log.html). 
- I see that you are using FreeRTOS. Be sure you followed [these](https://docs.lvgl.io/master/porting/os.html).

Mountain03, 2023-06-02:
> Hi,
>
> A few things that you can try:
>
> * if possible run to UI on PC with ASAN using standard malloc/free. It should find all memory corruptions.
> * enable LVGL asserts with `LV_USE_ASSERT_...   1` in [...]

1, I don't know how to use ASAN and deal with the environment, Can I try to move my code to VS simulator instead?
2, enable LV_USE_ASSERT_... cause very very slow in my system , even take many minutes to start up
3, layout_update_core(); or lv_obj_move_children_by(); is often use to refresh the page , randomly hard fault is really hard to catch,
4, yes indeed, any else task is removed , only LVGL task there, problem still there.

Mountain03, 2023-06-03:
I may found the real reason, maybe the sdram access error, I use MEMTESTOR to test , many errors in 550MHz, all corect in 520MHz, I put lvgl memery pool to external sdram ,when I put to internal everything seems OK, thank you for help

closed 2023-06-03 by Mountain03