lvgl/lvgl #4267: feat(sdl) add a way to release SDL and LVGL resources safely while needed

hanhsuan, 2023-06-04:
### Introduce the problem

Recently, the only way to release resources is clicking `x` button., due to the `event_handler_timer` couldn't be touched outside `lv_sdl_window.c`.

### Examples and cases

1. Normally, GUI applications will provide `ok` or `yes` button for user to confirm and close window. 
2. Catching signal while kernel sending to kill this application to release resources.

### Suggested solution

It might be able to add
 
```c
void lv_sdl_window_release(lv_disp_t * disp)
{
    clean_up(lv_disp_t * disp);
}
```

and

```c
void lv_sdl_quit()
{
    SDL_Quit();
    lv_timer_del(event_handler_timer);
    inited = false;
}
```

to let user release resources while needed to release whole SDL or not.

kisvegabor, 2023-06-05:
It makes sense but `lv_sdl_quit` should remove all `lv_disp_t`s too.

I would approach `lv_sdl_window_release` a little bit differently. What if we added an `LV_EVENT_DELETE` to each `disp` and in this event we could call `clean_up(disp)` (`lv_disp_remove(disp);` should be removed from here to avoid recursive calls).  So instead of `lv_sdl_window_release` we could call the generic `lv_disp_remove(disp);` which could send an `LV_EVENT_DELETE`, in which we can close the window. 

What do you think?

hanhsuan, 2023-06-06:
Sure. This one is better.

kisvegabor, 2023-06-06:
Great! Would you be interested in adding it to LVGL?

hanhsuan, 2023-06-06:
I love to. However, this is my side project. Please wait me a while to do that.

kisvegabor, 2023-06-07:
Sure, take you time. Thank you in advnace.

hanhsuan, 2023-06-11:
I've created [this feature](https://github.com/hanhsuan/lvgl/tree/add_release_sdl_resources_feature) and hit one issue. The `lv_free` --> `lv_free_builtin` --> `lv_tlsf_free` will make application to stop working at `tlsf_assert(!block_is_free(block) && "block already marked as free");`, if trying to free `disp` twice. 
Is this in purpose?

kisvegabor, 2023-06-12:
> if trying to free disp twice.
Is this in purpose?

How can you free the a disp twice?

hanhsuan, 2023-06-12:
The step I did is:
1. compiled with disable LV_SDL_DIRECT_EXIT
2. close the window, and keep the loop running, such as 
    ```c
        while(demo_status) {
            lv_tick_inc(1);
            lv_timer_handler();

            SLEEP(1);
        }
    ```
3. catch signal sent by ctrl + c
    ```c
    signal(SIGINT, demo_release);
    signal(SIGTERM, demo_release);
    ```
4.  send LV_EVENT_DELETE event to disp in `demo_release` function.

These steps will free disp twice.

kisvegabor, 2023-06-13:
> send LV_EVENT_DELETE event to disp in demo_release function.

The `disp` was already removed in `clean_up`. So there is no display to send the `LV_EVENT_DELETE` as it's already freed.

hanhsuan, 2023-06-13:
> 
> The `disp` was already removed in `clean_up`. So there is no display to send the `LV_EVENT_DELETE` as it's already freed.

1. In `void lv_disp_remove(lv_disp_t * disp)`, it calls `lv_free(disp);`.
2. In `size_t lv_tlsf_free(lv_tlsf_t tlsf, const void * ptr)`, it calls `block_mark_as_free(block);` to mark bit to freed and doesn't set `disp` to `NULL`.
3. In `lv_res_t lv_obj_send_event(lv_obj_t * obj, lv_event_code_t event_code, void * param)`, it only checks `obj` shouldn't be `NULL`.
```c
lv_res_t lv_obj_send_event(lv_obj_t * obj, lv_event_code_t event_code, void * param)
{
    if(obj == NULL) return LV_RES_OK;

    LV_ASSERT_OBJ(obj, MY_CLASS);

    lv_event_t e;
    e.current_target = obj;
    e.original_target = obj;
    e.code = event_code;
    e.user_data = NULL;
    e.param = param;
    e.deleted = 0;
    e.stop_bubbling = 0;
    e.stop_processing = 0;

    _lv_event_push(&e);

    /*Send the event*/
    lv_res_t res = event_send_core(&e);

    /*Remove this element from the list*/
    _lv_event_pop(&e);

    return res;
}
```

Do I miss something?

kisvegabor, 2023-06-13:
> to mark bit to freed and doesn't set disp to NULL.

Your" `disp` variables e.g. in `main.c` are not known by LVGL so it can't NULL them. 

> In lv_res_t lv_obj_send_event(lv_obj_t * obj, lv_event_code_t event_code, void * param), it only checks obj shouldn't be NULL.

For a display you should use `lv_disp_send_event`.

> send LV_EVENT_DELETE event to disp in demo_release function.

Is it really required to delete the display here? The program will be closed anyway and all allocated memories will be dropped. Calling `lv_deinit()` seems more appropriate.

hanhsuan, 2023-06-14:
Ok, I got it.

closed 2023-06-20 by hanhsuan