lvgl/lvgl #4268: Corrupted display on unhide/hide flag

mmar22, 2023-06-05:
Reported https://forum.lvgl.io/t/troubles-with-layered-objects/11947 [https://forum.lvgl.io/t/troubles-with-layered-objects/11947](url)

v 8.3.7
Platformio
ESP32

kisvegabor, 2023-06-06:
I've answered on the Forum

closed 2023-06-06 by kisvegabor

mmar22, 2023-06-06:
OK but this is bug and forum dont have bug part

kisvegabor, 2023-06-07:
I doubt that it's really a bug in LVGL as the hide feature is used a lot and a lot of people would have reported if it wasn't working well.

mmar22, 2023-06-07:
I agree YES and NO
1. Only some uses DMA drivers. And bug is partly in lovyan
2. LVGL part of bug is that calling set changes on invisible objects invoke flush_cb. You can test Hide bar aside other object and call update random value into bar .

Perfect will be when lvgl create driver for ESP i80 with right DMA.