lvgl/lvgl #4271: calender focus fails when input group is changed after creation

pm-mic, 2023-06-07:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When changing the input group of the calendar after it has been created, the calendar will fail to focus when the associated input group is active.

Because the documentation mentions that the buttonmatrix is the base of the calendar, I also checked with the buttonmatrix. However, the buttonmatrix shows the expected behaviour.

This bug is present both in v8.3 and v9.0.

### To Reproduce
```c
void lv_example_calendar_1(void)
{
    lv_group_t *group = lv_group_create();
    //lv_group_set_default(group);                                                           /* <= this works */
    lv_obj_t  * calendar = lv_calendar_create(lv_scr_act());

    lv_group_add_obj(group, calendar);                                                   /* <= this doesn't work */
    lv_win32_add_all_input_devices_to_group(group);

    lv_obj_set_size(calendar, 185, 185);
    lv_obj_align(calendar, LV_ALIGN_CENTER, 0, 27);
    lv_obj_add_event_cb(calendar, event_handler, LV_EVENT_ALL, NULL);

    lv_calendar_set_today_date(calendar, 2021, 02, 23);
    lv_calendar_set_showed_date(calendar, 2021, 02);

    /*Highlight a few days*/
    static lv_calendar_date_t highlighted_days[3];     
    highlighted_days[0].year = 2021;
    highlighted_days[0].month = 02;
    highlighted_days[0].day = 6;

    highlighted_days[1].year = 2021;
    highlighted_days[1].month = 02;
    highlighted_days[1].day = 11;

    highlighted_days[2].year = 2022;
    highlighted_days[2].month = 02;
    highlighted_days[2].day = 22;

    lv_calendar_set_highlighted_dates(calendar, highlighted_days, 3);

#if LV_USE_CALENDAR_HEADER_DROPDOWN
    lv_calendar_header_dropdown_create(calendar);
#elif LV_USE_CALENDAR_HEADER_ARROW
    lv_calendar_header_arrow_create(calendar);
#endif
    lv_calendar_set_showed_date(calendar, 2021, 10);
}

#endif
```

### Expected behavior

When I switch the input group and use a rotary encoder to go over the components, the calendar should focus as usual.

### Screenshots or video

kisvegabor, 2023-06-13:
Sorry, I've missed this issue.

It seems the documentation of the calendar is not updated... I'll address it soon. 

In the newer versions of LVGL the calendar became a little bit more complex and you can get the button matrix with `lv_calendar_get_btnmatrix(calendar);`

pm-mic, 2023-06-13:
@kisvegabor 

This works. However this project also includes the arrow header, which needs to make the buttons accessible as well then. 

So I made some changes, and I created #4281 in case anyone else might be interested. (Not sure why the tests fail, they fail on stuff I didn't touch... the fix works though).

I'm working with a project which is generated by squareline studio, which causes some problems with this as well. I'm not really sure if this is the right place to address these, but I don't know where else to go. 

For the fix to work squareline studio needs to add the header object to the "ui.h" and the "ui.c" globals in order to be able to call the lv_calendar_header_arrow_get_*_btn(). 

Also, when setting the initial date for the calendar in squareline studio, squareline studio calls "lv_calendar_set_shown_date()". Which is the wrong call, and I manually need to change this to "lv_calendar_set_showed_date()", every time the project is exported.

kisvegabor, 2023-06-13:
Really there is no example for that but to get the buttons you can use `lv_obj_get_child(header, 0 or 1)`. It think it's really simple so I'd rather show this use case instead of adding a new API.

What do you think?

pm-mic, 2023-06-13:
The addition of an API is indeed overkill when the fix is this easy. I wasn't aware of this solution. 

But I think both are a workaround for this particular issue. 

I think it should work as expected, calling lv_group_add_obj(lv_calendar_t *) should yield the same result as when it's being added to the default group at creation. 

The second best solution is indeed documenting this workaround as explicitly as possible.

kisvegabor, 2023-06-14:
> I think it should work as expected, calling lv_group_add_obj(lv_calendar_t *) should yield the same result as when it's being added to the default group at creation.

The calendar is made of several object (container, button matrix, header, and the content of the header) and `lv_group_add_obj` adds only a single object to the group. So we are left with documenting.  :slightly_frowning_face:

pm-mic, 2023-06-14:
Thank you for the clarity, and of course all the attention to this issue.

closed 2023-06-14 by pm-mic

pm-mic, 2023-06-23:
For anyone running into this issue as well, the proposed solution doesn't work because the header itself is the first child. So the solution is: 
- get header from calendar
- get child 0 from header
- get child -1 from header

```
    calendar_btns = lv_calendar_get_btnmatrix(ui_date_calendar);
    cal_header = lv_obj_get_child(ui_date_calendar, 0);
    next_btn = lv_obj_get_child(cal_header, -1);
    prev_btn = lv_obj_get_child(cal_header, 0);
```