lvgl/lvgl #4272: Compiler warnings with unused static functions using STM32 2D DMA

jacquesdmernok, 2023-06-08:
### Perform all steps below and tick them with [x]
- [ x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x ] Update lvgl to the latest version
- [ x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### Describe the bug
When enabling STM32 2DDMA, I get a persistent warning about the unused static draw callbacks

### To Reproduce

Using lvgl 8.3.7, enable the STM32 2D DMA with LV_USE_GPU_STM32_DMA2D in lv_conf.h.

The functions defined in the header ("lv_gpu_stm32_dma2d.h") are defined as follows:

```c
static void lv_draw_stm32_dma2d_blend(lv_draw_ctx_t * draw_ctx, const lv_draw_sw_blend_dsc_t * dsc);
static void lv_draw_stm32_dma2d_buffer_copy(lv_draw_ctx_t * draw_ctx,
                                            void * dest_buf, lv_coord_t dest_stride, const lv_area_t * dest_area,
                                            void * src_buf, lv_coord_t src_stride, const lv_area_t * src_area);
static lv_res_t lv_draw_stm32_dma2d_img(lv_draw_ctx_t * draw_ctx, const lv_draw_img_dsc_t * img_dsc,
                                        const lv_area_t * src_area, const void * src);
static void lv_gpu_stm32_dma2d_wait_cb(lv_draw_ctx_t * draw_ctx);
static void lv_draw_stm32_dma2d_img_decoded(lv_draw_ctx_t * draw_ctx, const lv_draw_img_dsc_t * img_dsc,
                                            const lv_area_t * coords, const uint8_t * src_buf, lv_img_cf_t color_format);
static dma2d_color_format_t lv_color_format_to_dma2d_color_format(lv_img_cf_t color_format);
static lv_point_t lv_area_get_offset(const lv_area_t * area1, const lv_area_t * area2);
```

Although I have tried disabling the static using LV_STM32_DMA2D_TEST, the define does not apply to these functions.

As far as I can see, the static prototypes should be in the .c file, and not the header. They are only referenced within the c file in lv_draw_stm32_dma2d_ctx_init etc.

### Expected behavior
Expected that the program can compile without unused static function warnings

kisvegabor, 2023-06-09:
Hi,

> As far as I can see, the static prototypes should be in the .c file, and not the header. They are only referenced within the c file in lv_draw_stm32_dma2d_ctx_init etc.

I can't test it now on hardware, but it should work. Could you try it out and send a PR?

kisvegabor, 2023-06-26:
The PR is merged, closing.

Thank you!

closed 2023-06-26 by kisvegabor