lvgl/lvgl #4277: Freezing image and font files on lvgl v9

sam0910, 2023-06-12:
I've converted fonts and images on lvgl online tools.
and I've copied them to lv_bindings/lvgl/src, lv_bindings/lvgl/src/font
to compile with firmware, there are error on that

1.Image
img.cf = LV_IMG_CF_TRUE_COLOR_ALPHA on the converted image.c file
I changed it to LV_COLOR_FORMAT_NATIVE_ALPHA, 
```
error: 'LV_IMG_PX_SIZE_ALPHA_BYTE' undeclared here (not in a function); did you mean
'LV_IMG_BUF_SIZE_ALPHA_1BIT'?
.data_size = 1482 * LV_IMG_PX_SIZE_ALPHA_BYTE,
```
tried LV_IMG_BUF_SIZE_ALPHA_1BIT, but still got exit.

 
2.Font
I've copied _custom_font.c_ file to _lv_bindings/lvgl/src/font_ folder.
I've declare the font name in lv_conf.h
**#define LV_FONT_CUSTOM_DECLARE LV_FONT_DECLARE(custom_font_14)**
```
error: conflicting type qualifiers for 'custom_font_14'
lv_font_t custom_font_14 = {
          ^~~~~~~~~~~~~~
In file included from 
../misc/lv_style.h:18, from lv_bindings/lvgl/src/core/lv_obj.h:20,
/lv_micropython/lib/lv_bindings/lvgl/lvgl.h:34,
/lv_micropython/lib/lv_bindings/lvgl/src/font/custom_font_14.c:10:
/lv_bindings/lv_conf.h:420:48: note: previous declaration of 'custom_font_14' was here
 #define LV_FONT_CUSTOM_DECLARE     LV_FONT_DECLARE(custom_font_14)
```

I've used to done in many times in previous version v8, 
but I don't know what I am missing.
Thank you :)

kisvegabor, 2023-06-12:
Hi,

Note that LVGL v9 is not released yet and tools are not updated.

I suggest using v8 for now.

sam0910, 2023-06-12:
Thank you for the reply.

sam0910, 2023-06-12:
I am using lvgl/micropython. as you know there is major updated for mp1.20.
It heard major updated version of mp1.20 is about to be aligned lv_micropython repo.
I assume that will be aligned with lvgl master branch. 

Could you give me some tips to freeze a font on lvgl master branch for using mp1.20?

I found the solution of images, simply changing like below.
```
LV_IMG_CF_TRUE_COLOR -> LV_COLOR_FORMAT_NATIVE
LV_IMG_PX_SIZE_ALPHA_BYTE -> LV_COLOR_DEPTH
```

closed 2023-06-12 by sam0910