lvgl/lvgl #4279: stuck when using the encoder to focus to the hidden msgbox

fivekernels, 2023-06-12:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I have an msgbox in my screen and an encoder registered in LVGL; When I hide the Button matrix in it(or hide the whole msgbox and before hide the msgbox, the encoder already focus on it), the encoder will still focus on the Button matrix, and once I press the encoder's key and rotate the encoder, the program will stop in the while loop because it can't find the next element to focus on.

### To Reproduce

```c
// hide the btnmatrix
lv_obj_t * msgbut = lv_msgbox_get_btns(msgbox_1);
lv_btnmatrix_set_btn_ctrl_all(msgbut, LV_BTNMATRIX_CTRL_HIDDEN);
// or hide the msgbox by software but the btnmatrix in msgbox already been focused
lv_obj_add_flag(msgbox_1, LV_OBJ_FLAG_HIDDEN);
```

### Expected behavior

The program actively jumps out of the hidden button matrix.

### Screenshots or video

[image]

kisvegabor, 2023-06-15:
Hi,

I've tried to reproduce it like this, but it worked well for me:
```c
void lv_example_msgbox_100(void)
{
    static const char * btns[] = {"Apply", "Close", ""};

    lv_obj_t * msgbox_1 = lv_msgbox_create(NULL, "Hello", "This is a message box with two buttons.", btns, true);
    lv_obj_add_event(msgbox_1, event_cb, LV_EVENT_VALUE_CHANGED, NULL);
    lv_obj_center(msgbox_1);

    // hide the btnmatrix
    lv_obj_t * msgbut = lv_msgbox_get_btns(msgbox_1);
    lv_group_focus_obj(msgbut);
    lv_btnmatrix_set_btn_ctrl_all(msgbut, LV_BTNMATRIX_CTRL_HIDDEN);
    lv_obj_add_flag(msgbox_1, LV_OBJ_FLAG_HIDDEN);
}
```

fivekernels, 2023-06-16:
hi, i just run the code in visual studio for lvgl v8.3:

```c
static const char * btns[] = { "Apply", "Close", "" };

lv_obj_t * msgbox_1 = lv_msgbox_create(NULL, "Hello", "This is a message box with two buttons.", btns, true);
lv_obj_center(msgbox_1);

// hide the btnmatrix
lv_obj_t * msgbut = lv_msgbox_get_btns(msgbox_1);
lv_group_focus_obj(msgbut);
lv_btnmatrix_set_btn_ctrl_all(msgbut, LV_BTNMATRIX_CTRL_HIDDEN);
lv_obj_add_flag(msgbox_1, LV_OBJ_FLAG_HIDDEN);
```

[image]

when the application lanuched, just press "right arrow" in my keyboard the application will stay in the while here in "lv_btnmatrix":

[image]

pete-pjb, 2023-06-16:
Hi @kisvegabor ,

I think there is a bug here or at least the need to prevent the code getting stuck in a loop...

If the button is hidden or inactive the code will hang....

Looking at the latest master branch I think you/or someone has already addressed the problem:

```c
                if(btnm->btn_id_sel == btn_id_start) {
                    btnm->btn_id_sel = LV_BTNMATRIX_BTN_NONE;
                    break;
                }
```
https://github.com/lvgl/lvgl/blob/1bfdecf585bb50fbb22ed1c38450b5c9eb600196/src/widgets/btnmatrix/lv_btnmatrix.c#LL597C1-L600C18

I think this needs to be back ported to Version 8 ?

Let me know what you think?

cc @fivekernels

Kind Regards,

Pete

pete-pjb, 2023-06-16:
Hi @kisvegabor and @fivekernels ,

It seems there is a commit [here](https://github.com/lvgl/lvgl/commit/7ae90e43d22c6b54787713419d587803af749a08) in May this year to the master branch which fixed this issue.

If you don't have time to back port this before Monday Gabor, I can do it Monday morning if that helps.

Kind Regards,

Pete

pete-pjb, 2023-06-16:
Hi again @kisvegabor and @fivekernels ,

I have gone ahead and added the fix to Release 8.3 [here](https://github.com/lvgl/lvgl/commit/07bce7466d471a15ba241a9db3239f7a116b0b46)

Perhaps @fivekernels you can give this a try now with the updated Release 8.3.

@kisvegabor I hope you don't mind me doing this, I know you are busy right now and I thought it would be helpful.  I had a bit of time so I have done it anyway, if there is any issues with what I have done let me know.

Kind Regards,

Pete

fivekernels, 2023-06-17:
Hi @pete-pjb ,

As you said, I do use the code under release/v8.3, and I have tested the updated release 8.3. It does improve the issue, but it seems that there are still some minor issues and  I'm not sure if this is a bug.

When I just hide the benmatrix but keep the msgbox displayed, if I rotate my encoder(mouse wheel in lvgl simulator), the focus point will still appear on the hidden button matrix (but it is not visible on the screen), after I press my encoder when focusing on the invisible Button Matrix, LVGL will focus inside Button Matrix, but since Button Matrix has been hidden, no focus can be seen on the screen at this time, and rotate the encoder will not be able to make the focus jump out of the Button Matrix, only "long press" the encoder can make the focus jump out of the Button Matrix.

```c
static const char * btns[] = { "Apply", "Close", "" };

 lv_obj_t * msgbox_1 = lv_msgbox_create(NULL, "Hello", "This is a message box with two buttons.", btns, true);
//lv_obj_add_event(msgbox_1, event_cb, LV_EVENT_VALUE_CHANGED, NULL);
lv_obj_center(msgbox_1);

// hide the btnmatrix
lv_obj_t * msgbut = lv_msgbox_get_btns(msgbox_1);
lv_group_focus_obj(msgbut);
lv_btnmatrix_set_btn_ctrl_all(msgbut, LV_BTNMATRIX_CTRL_HIDDEN);
// do not hide the msgbox
// lv_obj_add_flag(msgbox_1, LV_OBJ_FLAG_HIDDEN);
```

Here are the screenshots:

Focusing on the close button:  
[image]

Focusing on the button matrix(not visible):  
[image]

By the way, when I was testing the above code using the emulator, I found another issue. By default, LVGL is focused on Msgbox's Button Matrix. Once I use the encoder (mouse wheel) or the tab key on my keyboard to switch focus to another position (e.g. Msgbox's close button), then switch back to the hidden "Button Matrix", now when I press the "left arrow" on the keyboard, it will make the code stuck in a loop in "lv_btnmatrix".

[image]

At that time the "left arrow" on the keyboard is pressed, the value of btnm->btn_id_sel becomes an invalid "3", which causes the value of the btn_id_start to become an invalid "2" (because btnm->btn_cnt = 2, the statement "btnm->btn_id_sel == btn_id_start" will never become true), so it will not jump out of the loop.

Thank you.

kisvegabor, 2023-06-19:
Thank you Pete! 
This fix might be also related: edd5ad28a

pete-pjb, 2023-06-19:
Hi @fivekernels ,

Please can you try the latest V8.3 branch with:

> This fix might be also related: [edd5ad2](https://github.com/lvgl/lvgl/commit/edd5ad28ad74d1025db0b442cfa390d8c01ae006)

Which @kisvegabor committed about an hour ago.

If you still have issues please come back to us and we will look into the problem further.

Thank you in advance! 🙂 
Kind Regards,

Pete

fivekernels, 2023-06-19:
Hi @pete-pjb ,

Unfortunately, after I tested the latest commit [edd5ad2](https://github.com/lvgl/lvgl/commit/edd5ad28ad74d1025db0b442cfa390d8c01ae006), the issue is still there.

Thank you.

pete-pjb, 2023-06-19:
Hi @fivekernels ,

I just saw your reply...  Yes,  what do you expect is the correct behaviour

Kind Regards,

Pete

kisvegabor, 2023-06-19:
Okay, I can see the issue now and I think there are the indepent things:
1. The button matrix now really crashes as it can't find a non-hidden button, and `b++` will index the array out of the bounds. It's not trivial how to fix it on all places but it's doable in general.
2. The group shouldn't focus a button matrix if all of it's buttons are hidden. Is this true? Really this is what we want? Can be a case when we want to focus an all-button-hidden button matrix?

pete-pjb, 2023-06-19:
Hi @kisvegabor & @fivekernels ,

Having looked around the internet a bit,  I can see no application where a **_'Hidden'_** button or control would be focusable.  

But there maybe instances where a GUI built for accessibility might require **_'Disabled'_** buttons or controls to be focusable.

Personally I would think **_'Hidden'_** items definitely should not be focused at all...  To me it in unintuitive and confusing for the user when the focus suddenly disappears.  

I think with regard to **_'Disabled'_** objects it should be optional for them to be focusable or not.

What do you think?

Kind Regards,

Pete

pete-pjb, 2023-06-19:
Hi @kisvegabor ,

Without diving into the code and looking myself do you know if a viable solution to this problem might be to set the ```LV_OBJ_FLAG_CLICK_FOCUSABLE``` flag to false also, when setting the ```LV_OBJ_FLAG_HIDDEN``` flag?

Just a thought...

cc @fivekernels 

Kind Regards,

Pete

pete-pjb, 2023-06-19:
Hi Gabor ,

I have tried my question above and it does not work...

Looking at the solution going forwards here are some thought provoking suggestions (made without having a full appreciation of the Indev and group code right now):

Can this issue be handled by the indev code?

If an object or it's parent which is focused has a change of state to ```LV_OBJ_FLAG_HIDDEN``` should it perhaps un-focus itself and focus the next object in the group. (Not just the button matrix).  Also if ```lv_group_focus_obj()``` is called on a already hidden object should it just return without setting the focus...

It also seems the states/flags of the button matrix defined by ```lv_btnmatrix_ctrl_t``` might be a part duplication of some of the normal object flags ```lv_obj_flag_t``` is there perhaps scope to rework the button matrix to use the normal object flags for consistency and may be add some more if necessary?

To selectively decide if a disabled object can be focused/selected is also another conversation I guess,  currently it seems you cannot visibly focus a disabled object, but I expect that is by design.  I am not sure if embedded GUIs really need to focus disabled buttons personally, but others may not agree. 🙂 

If you already have something else in mind for this maybe I could help out with it if you have time to share your thoughts, things are a little calmer in my day job at the moment so I should be able to spend sometime if required...

cc @kisvegabor, @fivekernels

Kind Regards,

Pete

kisvegabor, 2023-06-20:
> If an object or it's parent which is focused has a change of state to LV_OBJ_FLAG_HIDDEN should it perhaps un-focus itself and focus the next object in the group. (Not just the button matrix). Also [...]

I think it's a good idea. In the `lv_obj_add/remove_flag` function we can handle it on one central place. 

When the selected button of button matrix is hidden, it could select the next button automatically.

> It also seems the states/flags of the button matrix defined by lv_btnmatrix_ctrl_t might be a part duplication of some of the normal object flags lv_obj_flag_t is there perhaps scope to rework the button matrix to use the normal object flags for consistency and may be add some more if necessary?

`lv_btnmatrix_ctrl_t` is a special thing. E.g. its lower bits are used to store the width of the button. So even they look similar, I wouldn't mix it with `lv_obj_flag_t`. 

> If you already have something else in mind for this maybe I could help out with it if you have time to share your thoughts, things are a little calmer in my day job at the moment so I should be able to spend some time if required...

Great! Happy to see you more around here :slightly_smiling_face:

pete-pjb, 2023-07-05:
Not Stale

pete-pjb, 2023-07-10:
Hi @fivekernels ,

I have pushed a fix to the latest version release/v8.3 can you test and see if it looks okay for you please?

Kind Regards,

Pete

fivekernels, 2023-07-10:
Hi @pete-pjb ,

I've updated my LVGL to commit 416b2677, but unfortunately the problem still seems to be the same.

Thank you.

fivekernels, 2023-07-10:
Hi @pete-pjb ,

I'll test it in detail and get back to you later.

Thank you.

pete-pjb, 2023-07-10:
Hi @fivekernels ,

Many thanks... whenever you can, no pressure 🙂 

If you can use the simulator to test then post your test code so I can use exactly the same code at my end.

It is very much appreciated!

Kind Regards,

Pete

fivekernels, 2023-07-11:
Hi @pete-pjb ,

I used Visual Studio's emulator again and the problem remained. The code and situation is the same as described here https://github.com/lvgl/lvgl/issues/4279#issuecomment-1595594200.

When I scroll the mouse wheel, the focus jumps to the hidden button, but since the button is hidden, the focus is not visible. At that time:

- If the scroll wheel is pressed, the focus goes inside the button matrix and is not visible; Only long press the wheel button again can jump out of focus.
- If I press "Enter" and "Left Arrow" on my keyboard, the program gets stuck in a while loop (line 586 of lv_btnmatrix.c); since I do not use a keyboard in my actual hardware product, this is just an unexpected discovery using the emulator.

Thank you.

pete-pjb, 2023-07-11:
Hi @fivekernels ,

Thank you for taking time to quantify this, I think the issues have been multiple and this one has gotten lost in the noise...
I can see now it's related to the ```lv_btnmatrix_set_btn_ctrl()``` call specifically, my fix addresses setting the visibility flag at an object level...  I will need to explore this further to find what is happening in this scenario...

I will let you know here, when I find a solution to the issue.

Kind Regards,

Pete

pete-pjb, 2023-07-12:
Hi @fivekernels ,

I hope you're well!

I have pushed another fix to the release/v8.3 branch [here](https://github.com/lvgl/lvgl/commit/29ed7c5717f6cfbca5ce888bf4497221525aae85)

When you have time please could you test it and report back your findings please?

Much appreciated!

Kind Regards,

Pete

fivekernels, 2023-07-13:
Hi @pete-pjb ,

Thank you for your attention. 

I just ran some tests and they look good but still have some minor issues:

If I have pressed the button on the encoder (mouse wheel middle button) to focus the btnmatrix before it is hidden, then when the btnmatrix of msgbox is hidden, although the focus appears to have been moved to other objects, turning the encoder (mouse wheel) focus at this point will not move between the remaining objects. I created a timer to simulate this behavior:

```c
static const char * btns[] = { "Apply", "Close", "" };

msgbox_1 = lv_msgbox_create(NULL, "Hello", "This is a message box with two buttons.", btns, true);
lv_obj_center(msgbox_1);

lv_obj_t * msgbut = lv_msgbox_get_btns(msgbox_1);
lv_group_focus_obj(msgbut);

lv_obj_t * btn1 = lv_btn_create(lv_scr_act());
lv_obj_align(btn1, LV_ALIGN_CENTER, 0, -200);
lv_obj_t * btn2 = lv_btn_create(lv_scr_act());
lv_obj_align(btn2, LV_ALIGN_CENTER, 0, -100);

lv_timer_create(lvtimer_gui_refresh_cb, 100, NULL);
```

```c
static void lvtimer_gui_refresh_cb(lv_timer_t * timer)
{
    static uint8_t cnt = 0;
    if (cnt < 50)
        cnt++;
    else if (cnt == 50)
    {
        cnt++;

        lv_obj_t * msgbut = lv_msgbox_get_btns(msgbox_1);
        lv_btnmatrix_set_btn_ctrl_all(msgbut, LV_BTNMATRIX_CTRL_HIDDEN); // do this to hidden btnmatrix

        // lv_obj_add_flag(msgbox_1, LV_OBJ_FLAG_HIDDEN); // or do this to hidden msgbox
    }
}
```

[image]

[image]

Also, if I use lv_obj_add_flag (msgbox_1, LV_OBJ_FLAG_HIDDEN); to hide the entire msgbox, then the focus does not seem to automatically focus on other objects.

[image]

Many thanks.

pete-pjb, 2023-07-13:
Hi @fivekernels ,

Thank you for your detailed breakdown.

I discussed the operation with @kisvegabor whilst looking into this and I had originally implemented it so it works the way you expect it too.

 Here is what he said to me about this:
> 
> I’d leave this manually adding the FOCUS_KEY state from this update because
> 
> - I think it’s unrelated
> - it’s not sure that the user always wants to add the outline in this case
> - this decision should be made in get_indev(), in a generic place
> 
> In v9, I’m planning to update the groups too, so I hope we will find a solution for it as well.

Perhaps you can comment here please Gabor so we can decide how to conclude this issue.

Kind Regards,

Pete

CC @kisvegabor

pete-pjb, 2023-07-14:
Hi @fivekernels ,

I have revisited this issue today as I have had a bit of time...

It seems the issues you are seeing here, although not probably currently entirely being handled correctly, are caused by the `model` message box.  If you set the first parameter to the active screen on creating the message box instead of ```NULL``` I think the behaviour is probably correct.  Can you try this your end please and let me know if you agree?

I will in the meantime try and work out how to prevent any access to the other objects on the screen whilst a modal message box is being displayed.

Kind Regards,

Pete

cc @kisvegabor

kisvegabor, 2023-07-14:
I've tried the code from [this comment](https://github.com/lvgl/lvgl/issues/4279#issuecomment-1633816139) and it works well for me. Turning the mousewheel changes focus in both cases.

A few month ago I've set up GitHub codespaces for LVGL. For easier code share we can try it out. Just fork LVGL, create and create a new code space here:
[image]

It will open an online VSCode editor where you can write any code and compile it by calling `./build_all.sh`. It will show the result in a simulator.

pete-pjb, 2023-07-17:
Hi @kisvegabor ,

As I see it, we still have some non-trivial changes to achieve the expected operation, we still need to address 2 issues:

1)  When the message box is hidden it does not remove the focus from the buttons if they are selected so when we check the object has focus when adding the LV_OBJ_FLAG_HIDDEN flag we somehow need to make sure it's children are also not in focus and if they are focused we need to shift focus to another object.  I am not sure how to go about it myself, if you can give me some pointers on how to implement I would be happy to sort it out.

2)  When the message box is 'MODAL' it should not be possible to focus on the buttons which are masked by the message boxes 'click masking backdrop' if that makes sense?  Currently those objects can be focused upon when moving the encoder, this however in my opinion is incorrect behaviour.  Again I can look at fixing the issue but I am not sure how to go about preventing the focus of the object which should be inaccessible, so if you can give me a clue about how to implement that it would be useful!

So the change needs to be as follows for hiding objects( 1) above...):
```c
void lv_obj_add_flag(lv_obj_t * obj, lv_obj_flag_t f)
{
    LV_ASSERT_OBJ(obj, MY_CLASS);

    bool was_on_layout = lv_obj_is_layout_positioned(obj);

    /* We must invalidate the area occupied by the object before we hide it as calls to invalidate hidden objects are ignored */
    if(f & LV_OBJ_FLAG_HIDDEN) lv_obj_invalidate(obj);

    obj->flags |= f;

    if(f & LV_OBJ_FLAG_HIDDEN) {
        if(lv_obj_has_state(obj, LV_STATE_FOCUSED) || some child of obj is focused ) {
            lv_group_t * group = lv_obj_get_group(obj);
            if(group != NULL) {
                lv_group_focus_next(group);
                lv_obj_t * next_obj = lv_group_get_focused(group);
                if(next_obj != NULL) {
                    lv_obj_invalidate(next_obj);
                }
            }
        }
    }

    if((was_on_layout != lv_obj_is_layout_positioned(obj)) || (f & (LV_OBJ_FLAG_LAYOUT_1 |  LV_OBJ_FLAG_LAYOUT_2))) {
        lv_obj_mark_layout_as_dirty(lv_obj_get_parent(obj));
        lv_obj_mark_layout_as_dirty(obj);
    }

    if(f & LV_OBJ_FLAG_SCROLLABLE) {
        lv_area_t hor_area, ver_area;
        lv_obj_get_scrollbar_area(obj, &hor_area, &ver_area);
        lv_obj_invalidate_area(obj, &hor_area);
        lv_obj_invalidate_area(obj, &ver_area);
    }
}
```

The code for shifting focus needs to be changed as follows:
```c
static bool focus_next_core(lv_group_t * group, void * (*begin)(const lv_ll_t *),
                            void * (*move)(const lv_ll_t *, const void *))
{
    bool focus_changed = false;
    if(group->frozen) return focus_changed;

    if( object is covered by a modal window don't select! ) return focus_changed;

    lv_obj_t ** obj_next     = group->obj_focus;
    lv_obj_t ** obj_sentinel = NULL;
    bool can_move            = true;
    bool can_begin           = true;

    for(;;) {
        if(obj_next == NULL) {
            if(group->wrap || obj_sentinel == NULL) {
                if(!can_begin) return focus_changed;
                obj_next  = begin(&group->obj_ll);
                can_move  = false;
                can_begin = false;
            }
            else {
                /*Currently focused object is the last/first in the group, keep it that way*/
                return focus_changed;
            }
        }

        if(obj_sentinel == NULL) {
            obj_sentinel = obj_next;
            if(obj_sentinel == NULL) return focus_changed; /*Group is empty*/
        }

        if(can_move) {
            obj_next = move(&group->obj_ll, obj_next);

            /*Give up if we walked the entire list and haven't found another visible object*/
            if(obj_next == obj_sentinel) return focus_changed;
        }

        can_move = true;

        if(obj_next == NULL) continue;
        if(lv_obj_get_state(*obj_next) & LV_STATE_DISABLED) continue;

        /*Hidden objects don't receive focus.
         *If any parent is hidden, the object is also hidden)*/
        lv_obj_t * parent = *obj_next;
        while(parent) {
            if(lv_obj_has_flag(parent, LV_OBJ_FLAG_HIDDEN)) break;
            parent = lv_obj_get_parent(parent);
        }

        if(parent && lv_obj_has_flag(parent, LV_OBJ_FLAG_HIDDEN)) continue;

        /*If we got her a good candidate is found*/
        break;
    }

    if(obj_next == group->obj_focus) return focus_changed; /*There's only one visible object and it's already focused*/

    if(group->obj_focus) {
        lv_res_t res = lv_event_send(*group->obj_focus, LV_EVENT_DEFOCUSED, get_indev(group));
        if(res != LV_RES_OK) return focus_changed;
        lv_obj_invalidate(*group->obj_focus);
    }

    group->obj_focus = obj_next;

    lv_res_t res = lv_event_send(*group->obj_focus, LV_EVENT_FOCUSED, get_indev(group));
    if(res != LV_RES_OK) return focus_changed;

    lv_obj_invalidate(*group->obj_focus);

    if(group->focus_cb) group->focus_cb(group);
    focus_changed = true;
    return focus_changed;
}
```

If this doesn't make sense please come back to me and I will try and explain it better!

Kind Regards,

Pete

cc @fivekernels

kisvegabor, 2023-07-19:
> When the message box is hidden it does not remove the focus from the buttons if they are selected so when we check the object has focus when adding the LV_OBJ_FLAG_HIDDEN flag we somehow need to [...]

It sounds really complicated. What if we leave it to the user instead of supporting this edge case in LVGL? It should be doable to call an `lv_group_focus_obj()` when needed.

> When the message box is 'MODAL' it should not be possible to focus on the buttons which are masked by the message boxes 'click masking backdrop' if that makes sense? Currently those objects can be [...]

Here I think the solution would be to allow "trapping" focus on any widget. If a parent widget has focus trap, only it's children can be focused. What do you think?

pete-pjb, 2023-07-26:
Hi @kisvegabor ,

My sincere apologies I seemed to have missed your reply!

>It sounds really complicated. What if we leave it to the user instead of supporting this edge case in LVGL? It should be doable to call an lv_group_focus_obj() when needed.

I agree it is a conundrum to solve, my only concern is I think it is possible to cause a crash.  I will see if I can create a test case to demonstrate.

>Here I think the solution would be to allow "trapping" focus on any widget. If a parent widget has focus trap, only it's children can be focused. What do you think?

I think this is a good solution, is it something you want to implement or if you specify how you want it to be implemented I may be able to take a look.

Let me know what you think.

Kind Regards,

Pete

pete-pjb, 2023-07-26:
Hi @kisvegabor ,

>I will see if I can create a test case to demonstrate.

It seems I can't cause a crash now it must have been a previous iteration I was thinking of....

So I will await you comment on the 'Trapping Focus'...

Kind Regards,

Pete