lvgl/lvgl #4286: buffer_copy callback for the STM32 DMA2D seems has some issue.

caacyzp, 2023-06-14:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version | I am sorry I did not try the latest but I checked changelog and seems no commit is ralated to what I am facing. I am using V8.3.7.
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

It seems that the buffer_copy callback of STM32DMA2d version and sw version produce different behavior. I have to change the parameter in order to create same ouput.

I' am using 2 screen sized drawing buffer with direct mode = 1 and full_refr = 0. Thus, per documentation, I have to perform sync job in flush_callback. I peeked several solutions and start to use the following code:

```c
                disp_drv.draw_ctx->buffer_copy(
                disp_drv.draw_ctx,                                //Draw context
                buff_off_screen, TFT_HOR_RES, &disp->inv_areas[i], //Dest
                buff_on_screen, TFT_HOR_RES, &disp->inv_areas[i]  //Src
                );
```
The application is fine when I set ``LV_USE_GPU_STM32_DMA2D`` as ``0``  in ``lv_conf.h``, while, when I enable the DMA2D support, the screen start to fill in the dirty area with the content at the front of the on_screen buffer. 

Take the following example:

This is the second round of the flush_callback and the frame rate is displayed at lower_right corner of the screen. In this round, the ```flush_callback``` was supposed to copy the framerate display area into the off_screen buffer, while, after calling the buffer_copy, the dumped memory in the off_screen buffer produces the below result.

[image]

That looks like the DMA2D copies the content of the on_screen buffer with the same area dimension but starts with 0 offset of the on_screen buffer into the offscreen buffer with the designated place.

If I stop using the buffer_copy callback from the draw_ctx but use ``lv_draw_sw_buffer_copy()``, the application works fine. 

I am sorry that I am new to the STM32 and LVGL development so I am not capable of tracing further deep into STM32 code. Hope this could clearly describe the issue.

### To Reproduce

### Expected behavior

### Screenshots or video

caacyzp, 2023-06-15:
Ok. I think I found some walkaround. 

To give some background, the DMA2D version of buffer_copy callback is located in `lv_gpu_stm32_dma2d.c` file. It is assigned as `lv_draw_stm32_dma2d_buffer_copy` function within `lv_draw_stm32_dma2d_ctx_init()`.  The `lv_draw_stm32_dma2d_buffer_copy()` further called  `_lv_draw_stm32_dma2d_blend_map` function.

The `_lv_draw_stm32_dma2d_blend_map` function, as far as my understanding, copy the data from the DMA2D_FGMAR register with the offset defined in DMA2D_FGOR register and then perform some other actions and put into dest memory with assigned offset. 

In `_lv_draw_stm32_dma2d_blend_map` function, the DMA2D_FGMAR register is assigned as following:
```c
    DMA2D->FGMAR = ((uint32_t)src_buf) + srcBpp * ((src_stride * src_offset->y) + src_offset->x);
    DMA2D->FGOR = src_stride - draw_width;
```
We can see that the parameter `src_offset` is used here to calculate the address of the pending copy pixel and this ``src_offset` parameter is passed in by the `lv_draw_stm32_dma2d_buffer_copy` function. 

I then look back to the `lv_draw_stm32_dma2d_buffer_copy` function and found out that the src_offset is calculated by
```c
lv_point_t src_offset = lv_area_get_offset(src_area, dest_area);
```
To be honest, I have no clue the purpose of calculating coordinate offset between two areas here, But by following what the `_lv_draw_stm32_dma2d_blend_map` function did with the `src_offset` variable. I made the following changes:
```c
\\lv_point_t src_offset = lv_area_get_offset(src_area, dest_area);
lv_point_t src_offset = {src_area->x1, src_area->y1};

```
Now, the buffer_copy function works as expected.

Again, I am not familiar with the LVGL framework and STM32 programming so this probably was just a lucky walkaround. 

Anyway, hope this could help.

caacyzp, 2023-06-15:
Ok. This seems already fixed in latest code of the repo. @kisvegabor Probably you can verify them and close the issue.

vahidajalluian, 2023-06-15:
> @vahidajalluian, it might be interesting for you too.

I'm primarily interested in using hal code instead of non-ST team code generated unless it has been working for long-run tests. Now I have some codes from past versions which embededt helped. and looking forward to have some fast UI without upgrading the chip from F4 to F7

kisvegabor, 2023-06-16:
>  @kisvegabor Probably you can verify them and close the issue.

Actually @tdjastrzebski knows more about the details of he new DAM2D implementation. Could you check it out, please?

tdjastrzebski, 2023-06-16:
@caacyzp I have to say that `buffer_copy()` was not well tested since I did not know how exactly how it was supposed to be used so it is entirely possible that it does not work the way it is expected, hence you may be right, using 'lv_area_get_offset()` in `buffer_copy()` may not be needed. Than you for finding the fix.

closed 2023-07-09 by lvgl-bot

tdjastrzebski, 2023-07-09:
Btw, this driver will require another fix for Cortex-M33 core, which uses a different SRAM cache management

NimaAgm, 2024-12-23:
This issue is not resolved in lvgl 8.4 branch

vahidajalluian, 2024-12-23:
> This issue is not resolved in lvgl 8.4 branch

You should do some coding, not all cakes are ready to eat, some should be baked.

NimaAgm, 2024-12-25:
> > This issue is not resolved in lvgl 8.4 branch
> 
> You should do some coding, not all cakes are ready to be served, some should be baked.

I have done but not every knows this the issue for their display.
Changing this line is easy but finding this issue is not. It takes a lot of time.
So when the issue is closed it should be fixed in the branches.

vahidajalluian, 2024-12-25:
Follow my questions on the LVGL forum and you'll find some answers from
embededt to my questions, those are of course for some previous versions
but still they work. Have you been able to run lvgl without dma2d(or
without any other graphic accelerator)? If yes your road is not much far.