lvgl/lvgl #4289: Why style opa should use draw layer?

FASTSHIFT, 2023-06-15:
### Introduce the problem
In versions after LVGL v8.3, lvgl introduces a layer function, which supports the rotation and scaling of controls. The layer is essentially a temporary buffer during the drawing process. lvgl will draw the control on the temporary buffer, and the transformation effects (opa, zoom, rotate) are realized using the `lv_draw_img` method.
But this will introduce a problem. Using layer is equivalent to introducing the overhead of a copy. For zoom and rotate effects, sw blend does not support it, so it can be realized by layer. But for the opa effect, why don't we just pre-multiply the transparency directly in the blend? This only needs to be rendered once, instead of using `lv_draw_img` to do one more color mixing.

### Examples and cases
For this scenario, The single-frame rendering time of LVGL v8.2 is only about half of the time-consuming of LVGL v9 single-frame rendering.
```c
void opa_test(void)
{
    lv_obj_t* obj = lv_obj_create(lv_scr_act());
    lv_obj_set_style_opa(obj, LV_OPA_50, 0);
    lv_obj_set_size(obj, LV_PCT(200), LV_PCT(100));
    lv_obj_create(obj);
}
```

### Suggested solution
Consider removing the layer generated by style opa, and adopt the v8.2 rendering method.

kisvegabor, 2023-06-15:
In v8.2 (where the opacity were just scaled) the colors are not correct. See this:
[image]

On the left there is a red rectangle with 50% opa, and a green rectangle with also 50% opa. As the red one is drawn first, the green one is blended to a redish-blueish color.

On the right, the red rectangle (with its green child) is on a new layer, so on green rectangle a pure green is blended to the blue background.

The scaling versin looks ugly, when the there are multiple objects behind each other. When opa != 100% all background objects are visible too. If we are using layers, only the final image is blended.

It really results in performance drop, so I think we should support both under a different name. E.g.
- `opa`: layered
- `opa_scale`: as v8.2

What do you think?

FASTSHIFT, 2023-06-19:
I am very much in favor of keeping the style opa behavior of v8.2, because it performs better, and in most cases, users may not pay much attention to the difference in color (including me), but the difference in frame rate is very easy to notice.

In addition, if the GPU supports global alpha and rotation and scaling of arbitrary content, we can consider designing an interface to replace the work of the layer, using a pure GPU implementation, because using a layer will introduce additional memory allocation and image drawing overhead, which affects the frame rate.

kisvegabor, 2023-06-19:
> I am very much in favor of keeping the style opa behavior of v8.2, because it performs better, and in most cases, users may not pay much attention to the difference in color (including me), but the difference in frame rate is very easy to notice.

What about using these names then:
1. `opa`: v8.2 scaling method
2. `opa_layered`: v8.3 method

> 
@kisvegabor
I am very much in favor of keeping the style opa behavior of v8.2, because it performs better, and in most cases, users may not pay much attention to the difference in color (including me), but the difference in frame rate is very easy to notice.

> In addition, if the GPU supports global alpha and rotation and scaling of arbitrary content, we can consider designing an interface to replace the work of the layer, using a pure GPU implementation, [...]

How it solve the opa issue? If the GPU draws the content after each other with 50% opa the colors from the previous drawing will be visible during blending.

FASTSHIFT, 2023-06-23:
@kisvegabor 
> What about using these names then:
opa: v8.2 scaling method
opa_layered: v8.3 method

I totally agree.

> How it solve the opa issue? If the GPU draws the content after each other with 50% opa the colors from the previous drawing will be visible during blending.

You're right, it just multiplies the alpha on each draw, the color of the previous draw is visible during blending, let's ignore that:)

kisvegabor, 2023-06-23:
Great! :slightly_smiling_face: 
Now I would like to focus on merging the parallel rendering branch and after that we can added these minor changes.

FASTSHIFT, 2023-07-11:
Not stale.

kisvegabor, 2023-07-12:
Now we have merged the parallel rendering update to master, are you interested in working on this update? In summary we need two opa style properties:
- `style_opa_layered`: create a new layer if it's not 0xff.
- `style_opa`: just scale down the opacity of `bg_opa`, `border_opa`, `text_opa`, etc. I think it shouldn't be a inherited style property, but we should have a helper function like `lv_obj_get_style_opa_recursive()` which get the `opa` of all parent and return the final scaled value. (`recursive` is not the best work to describe it)

What do you think?

kisvegabor, 2023-07-27:
Not stale.

closed 2023-08-17 by FASTSHIFT