lvgl/lvgl #4291: LV_FONT_MONTSERRAT_12_SUBPX not rendering properly

brentru, 2023-06-15:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2023-06-16:
Hi,

Unfortunately I couldn't reproduce the issue in the simulator either. :slightly_frowning_face:  

To fix it, it'd important to see the problem on my end too. To make the issue happen in the simulator, please be sure to use the same
- UI code
- color depth and LV_COLOR_16_SWAP settings
- draw buffer size

brentru, 2023-06-16:
> color depth and LV_COLOR_16_SWAP settings, draw buffer size

Where are these found - in the configuration file or in the port?

closed 2023-07-12 by lvgl-bot

daironxuan, 2023-07-15:
Yes, the PC is no problem, once on the hardware 'Chinese' will present � � � � Ҫ � ¨ squared � ѧ ϰ �...

The test screen is: TTFLCD, 240*320,

Whether it is defined using LV_FONT_DECLARE or not, it is still invalid!

kisvegabor, 2023-07-15:
I'm sorry but I have now idea. If it's working in simulator, I can't do much to figure it out. :slightly_frowning_face: