lvgl/lvgl #4293: Keil / CMSIS-Pack tries to pull non-existent pack file.

cinnamondev, 2023-06-16:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The version of LVGL on the 8.3 branch is newer (8.3.7) is newer than the version [arm KEIL](https://www.keil.arm.com/packs/lvgl-lvgl/versions/) attempts to provide (8.3.5). KEIL will try to download the following URL:

`https://github.com/lvgl/lvgl/raw/release/v8.3/env_support/cmsis-pack/LVGL.lvgl.8.3.5.pack`

While the latest in that release is `LVGL.lvgl.8.3.7.pack`. Keil should be updated to get 8.3.7 instead.

### To Reproduce

1. Go to [arm KEIL](https://www.keil.arm.com/packs/lvgl-lvgl/versions/), or tool for downloading CMSIS-Packs (ie `cpackget`, `pack-installer`)
2. Attempt to download version 8.3.5 of LVGL (`cpackget pack add LVGL::lvgl@8.3.5`, or `Download 8.3.5` on the web page.)
(404 Not Found)

### Expected behavior

1. Go to [arm KEIL](https://www.keil.arm.com/packs/lvgl-lvgl/versions/), or tool for downloading CMSIS-Packs (ie `cpackget`, `pack-installer`)
2. Attempt to download version 8.3.5 of LVGL (`cpackget pack add LVGL::lvgl@8.3.5`, or `Download 8.3.5` on the web page.)
3. Pack installs/downloads

### Screenshots or video

Command output: `cpackget pack add LVGL::lvgl@8.3.5 --verbose`

```
<snip...>
D: Specified packs [LVGL::lvgl@8.3.5]
D: Extracting pack info from "LVGL::lvgl@8.3.5"         ck="lvgl", Version="8.3.5", VersionModifier="0"
D: "LVGL::lvgl@8.3.5" is a packID with Vendor="LVGL", PappData\Local\Arm\Packs\.Web" that match "^.*\.pdsc$"ck="lvgl", Version="8.3.5", VersionModifier="0"         Local\Arm\Packs\.Web"
D: Listing files and directories in "C:\Users\cinnamon\AppData\Local\Arm\Packs\.Web" that match "^.*\.pdsc$"    
D: Found "LVGL.lvgl.pdsc" in "C:\Users\cinnamon\AppData\Local\Arm\Packs\.Web"                                   pData\Local\Arm\Packs\.Web\LVGL.lvgl.pdsc"
D: Checking if LVGL.lvgl.8.3.5 is installed             al\Arm\Packs\.Web\LVGL.lvgl.pdsc"
I: Adding pack "LVGL::lvgl@8.3.5"                       ng PDSC "C:\Users\cinnamon\AppData\Local\Arm\Packs\.Web\LVGL.lvgl.pdsc"
D: Finding URL for "LVGL::lvgl@8.3.5"
D: Initializing PdscXML object for "C:\Users\cinnamon\Aprelease/v8.3/env_support/cmsis-pack/LVGL.lvgl.8.3.5.pack" (or just making sure it exists locally)pData\Local\Arm\Packs\.Web\LVGL.lvgl.pdsc"              v8.3/env_support/cmsis-pack/LVGL.lvgl.8.3.5.pack to C:\Users\cinnamon\AppData\Local\Arm\Packs\.Download\LVGL.lvgl.8.3.5.pack
D: Reading pdsc from file "C:\Users\cinnamon\AppData\Local\Arm\Packs\.Web\LVGL.lvgl.pdsc"
D: Resolving version modifier for "LVGL::lvgl@8.3.5" using PDSC "C:\Users\cinnamon\AppData\Local\Arm\Packs\.Web\LVGL.lvgl.pdsc"
D: - resolved(@) as 8.3.5
D: Fetching pack file "https://github.com/lvgl/lvgl/raw/release/v8.3/env_support/cmsis-pack/LVGL.lvgl.8.3.5.pack" (or just making sure it exists locally)
D: Downloading https://github.com/lvgl/lvgl/raw/release/v8.3/env_support/cmsis-pack/LVGL.lvgl.8.3.5.pack to C:\Users\cinnamon\AppData\Local\Arm\Packs\.Download\LVGL.lvgl.8.3.5.pack
E: bad status: 404 Not Found
E: bad request
```

kisvegabor, 2023-06-19:
cc @GorgonMeducer

GorgonMeducer, 2023-06-20:
I will have a look

GorgonMeducer, 2023-06-20:
fixed in: https://github.com/lvgl/lvgl/pull/4303

@cinnamondev Due to the known issue of the KEIL website, unless we release v9.0.0, the [page](https://www.keil.arm.com/packs/lvgl-lvgl/versions/) won't refresh, i.e. you won't see the 5.3.7. Sorry.

Please download the cmsis-pack from the LVGL release page, or you can see the latest version from Pack Installer.

closed 2023-07-12 by lvgl-bot