lvgl/lvgl #4295: Keybaord repaint so slowly, is there a way to resolve it?

lanistor, 2023-06-18:
### Introduce the problem

The keybaord created by `lv_keyboard_create`, will render the whole keyboard on every button touch event.

This will cause repaint slowly, espcially on e-paper, such as on 6 inch size e-paper, every touch will cause nearly 3 seconds (and it repaint 2 times on each touch, for invert the color on pressed)

### Examples and cases

```cpp
auto kb = lv_keyboard_create(lv_layer_top());
lv_obj_set_size(kb, width, height);

auto kb2 = (lv_keyboard_t*)kb;
lv_obj_set_style_bg_color(&kb2->btnm.obj, color_make(0x00),
                            LV_PART_ITEMS | LV_STATE_PRESSED);
lv_obj_set_style_text_color(&kb2->btnm.obj, color_make(0xff),
                              LV_PART_ITEMS | LV_STATE_PRESSED);
```

### Suggested solution

Only repaint the pressed button area, instead of the whole keyboard area.

kisvegabor, 2023-06-19:
Hi,

For v9 I keep this issue in mind but until that I can recommend this workaround:
```c

static void invalidate_button_area(const lv_obj_t * obj, uint16_t btn_idx)
{
    if(btn_idx == LV_BTNMATRIX_BTN_NONE) return;

    lv_area_t btn_area;
    lv_area_t obj_area;

    lv_btnmatrix_t * btnm = (lv_btnmatrix_t *)obj;;
    if(btn_idx >= btnm->btn_cnt) return;

    lv_area_copy(&btn_area, &btnm->button_areas[btn_idx]);
    lv_obj_get_coords(obj, &obj_area);

    /*The buttons might have outline and shadow so make the invalidation larger with the gaps between the buttons.
     *It assumes that the outline or shadow is smaller than the gaps*/
    lv_coord_t row_gap = lv_obj_get_style_pad_row(obj, LV_PART_MAIN);
    lv_coord_t col_gap = lv_obj_get_style_pad_column(obj, LV_PART_MAIN);

    /*Be sure to have a minimal extra space if row/col_gap is small*/
    lv_coord_t dpi = lv_disp_get_dpi(lv_obj_get_disp(obj));
    row_gap = LV_MAX(row_gap, dpi / 10);
    col_gap = LV_MAX(col_gap, dpi / 10);

    /*Convert relative coordinates to absolute*/
    btn_area.x1 += obj_area.x1 - row_gap;
    btn_area.y1 += obj_area.y1 - col_gap;
    btn_area.x2 += obj_area.x1 + row_gap;
    btn_area.y2 += obj_area.y1 + col_gap;

    lv_obj_invalidate_area(obj, &btn_area);
}

void event_pre_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    if(code == LV_EVENT_PRESSED) {
        lv_disp_enable_invalidation(NULL, false); //Disable invalidation on press
    }
}

void event_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    lv_keyboard_t * keyboard = (lv_keyboard_t *) obj;
    if(code == LV_EVENT_PRESSED) {
        //Invalidate only the button
        lv_disp_enable_invalidation(NULL, true);
        invalidate_button_area(obj, keyboard->btnm.btn_id_sel);
        lv_disp_enable_invalidation(NULL, false);
    }
    if(code == LV_EVENT_RELEASED) {
        lv_disp_enable_invalidation(NULL, true);
        invalidate_button_area(obj, keyboard->btnm.btn_id_sel); //Enable invalidation on release
    }
}

```

lanistor, 2023-06-20:
> Hi,
> 
> For v9 I keep this issue in mind but until that I can recommend this workaround:
> 
> ```c
> static void invalidate_button_area(const lv_obj_t * obj, uint16_t btn_idx)
> {
>     if(btn_idx == LV_BTNMATRIX_BTN_NONE) return;
> 
>     lv_area_t btn_area;
>     lv_area_t obj_area;
> 
>     lv_btnmatrix_t * btnm = (lv_btnmatrix_t *)obj;;
>     if(btn_idx >= btnm->btn_cnt) return;
> 
>     lv_area_copy(&btn_area, &btnm->button_areas[btn_idx]);
>     lv_obj_get_coords(obj, &obj_area);
> 
>     /*The buttons might have outline and shadow so make the invalidation larger with the gaps between the buttons.
>      *It assumes that the outline or shadow is smaller than the gaps*/
>     lv_coord_t row_gap = lv_obj_get_style_pad_row(obj, LV_PART_MAIN);
>     lv_coord_t col_gap = lv_obj_get_style_pad_column(obj, LV_PART_MAIN);
> 
>     /*Be sure to have a minimal extra space if row/col_gap is small*/
>     lv_coord_t dpi = lv_disp_get_dpi(lv_obj_get_disp(obj));
>     row_gap = LV_MAX(row_gap, dpi / 10);
>     col_gap = LV_MAX(col_gap, dpi / 10);
> 
>     /*Convert relative coordinates to absolute*/
>     btn_area.x1 += obj_area.x1 - row_gap;
>     btn_area.y1 += obj_area.y1 - col_gap;
>     btn_area.x2 += obj_area.x1 + row_gap;
>     btn_area.y2 += obj_area.y1 + col_gap;
> 
>     lv_obj_invalidate_area(obj, &btn_area);
> }
> 
> 
> void event_pre_cb(lv_event_t * e)
> {
>     lv_event_code_t code = lv_event_get_code(e);
>     if(code == LV_EVENT_PRESSED) {
>         lv_disp_enable_invalidation(NULL, false); //Disable invalidation on press
>     }
> }
> 
> void event_cb(lv_event_t * e)
> {
>     lv_event_code_t code = lv_event_get_code(e);
>     lv_obj_t * obj = lv_event_get_target(e);
>     lv_keyboard_t * keyboard = (lv_keyboard_t *) obj;
>     if(code == LV_EVENT_PRESSED) {
>         //Invalidate only the button
>         lv_disp_enable_invalidation(NULL, true);
>         invalidate_button_area(obj, keyboard->btnm.btn_id_sel);
>         lv_disp_enable_invalidation(NULL, false);
>     }
>     if(code == LV_EVENT_RELEASED) {
>         lv_disp_enable_invalidation(NULL, true);
>         invalidate_button_area(obj, keyboard->btnm.btn_id_sel); //Enable invalidation on release
>     }
> }
> 
> ...
> 
> 
>     lv_obj_t * kb = lv_keyboard_create(lv_scr_act());
>     lv_obj_add_event(kb, event_pre_cb, LV_EVENT_ALL | LV_EVENT_PREPROCESS, NULL);
>     lv_obj_add_event(kb, event_cb, LV_EVENT_ALL, NULL);
> ```

Hi, i tested the code in `release/v8.1`, still have problem: After click the button, will repaint the whole keyboard first, then repaint the click area.

With the code changed like this:
```cpp
  lv_obj_t* kb = lv_keyboard_create(lv_scr_act());
  lv_obj_add_event_cb(kb, event_pre_cb, LV_EVENT_PRESSED, NULL);
  lv_obj_add_event_cb(kb, event_cb, LV_EVENT_ALL, NULL);
```

kisvegabor, 2023-06-20:
The `LV_EVENT_PREPROCESS` part is very important. Can you update to v8.3?

lanistor, 2023-06-21:
> ```c
> LV_EVENT_ALL
> ```

This works👍。

But i found another two problems:
1. If you touch `1#`, `ABC` to switch keyboard, it will not rerender the whole keyboard.
2. The textarea cannot receive the keyboard input, `lv_disp_enable_invalidation` in `event_pre_cb` effects this.

For the first problem, i had tried this:
```cpp
void event_cb(lv_event_t* e) {
  lv_event_code_t code     = lv_event_get_code(e);
  lv_obj_t*       obj      = lv_event_get_target(e);
  lv_keyboard_t*  keyboard = (lv_keyboard_t*)obj;
  if (code == LV_EVENT_PRESSED) {
    lv_disp_enable_invalidation(NULL, true);

    const char* txt =
      lv_btnmatrix_get_btn_text(obj, lv_btnmatrix_get_selected_btn(obj));
    bool is_switch_code =
      strcmp(txt, "abc") == 0 || strcmp(txt, "ABC") == 0 ||
      strcmp(txt, "1#") == 0 || strcmp(txt, LV_SYMBOL_CLOSE) == 0 ||
      strcmp(txt, LV_SYMBOL_KEYBOARD) == 0 || strcmp(txt, LV_SYMBOL_OK) == 0;

    if (is_switch_code) {
      // Invalidate the whole keyboard
      static lv_area_t kb_area1;
      lv_obj_get_coords(obj, &kb_area1);
      lv_obj_invalidate_area(obj, &kb_area1);
    } else {
      // Invalidate only the button
      invalidate_button_area(obj, keyboard->btnm.btn_id_sel);
    }

    lv_disp_enable_invalidation(NULL, false);
  }
  if (code == LV_EVENT_RELEASED) {
    const char* txt =
      lv_btnmatrix_get_btn_text(obj, lv_btnmatrix_get_selected_btn(obj));
    bool is_switch_code =
      strcmp(txt, "abc") == 0 || strcmp(txt, "ABC") == 0 ||
      strcmp(txt, "1#") == 0 || strcmp(txt, LV_SYMBOL_CLOSE) == 0 ||
      strcmp(txt, LV_SYMBOL_KEYBOARD) == 0 || strcmp(txt, LV_SYMBOL_OK) == 0;
    ESP_LOGW("####", "code: %s; is_switch_code: %d", txt, is_switch_code);
    lv_disp_enable_invalidation(NULL, true);
    if (is_switch_code) {
      // Invalidate the whole keyboard
      static lv_area_t kb_area2;
      lv_obj_get_coords(obj, &kb_area2);
      lv_obj_invalidate_area(obj, &kb_area2);
    } else {
      invalidate_button_area(
        obj, keyboard->btnm.btn_id_sel);  //Enable invalidation on release
    }
  }
}
```

kisvegabor, 2023-06-23:
I've addressed both issues here:
```c

static lv_keyboard_mode_t prev_mode;
void event_pre_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_PRESSED) {
        lv_disp_enable_invalidation(NULL, false); //Disable invalidation on press
        prev_mode = lv_keyboard_get_mode(obj);
    }
}

void event_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    lv_keyboard_t * keyboard = (lv_keyboard_t *) obj;
    if(code == LV_EVENT_PRESSED) {
        //Invalidate only the button
        lv_disp_enable_invalidation(NULL, true);
        invalidate_button_area(obj, keyboard->btnm.btn_id_sel);
        lv_disp_enable_invalidation(NULL, false);
    }
    if(code == LV_EVENT_RELEASED) {
        lv_disp_enable_invalidation(NULL, true);

        lv_keyboard_mode_t new_mode = lv_keyboard_get_mode(obj);
        if(prev_mode != new_mode) {
            lv_obj_invalidate(obj);
        } else {
            invalidate_button_area(obj, keyboard->btnm.btn_id_sel); //Enable invalidation on release
        }

        lv_obj_invalidate(lv_textarea_get_label(lv_keyboard_get_textarea(obj)));
    }
    if(code == LV_EVENT_READY || code == LV_EVENT_CANCEL) {
        lv_disp_enable_invalidation(NULL, true);
        lv_obj_del(obj);
    }
}
```

lanistor, 2023-06-23:
> I've addressed both issues here:
>
> ```c
> static lv_keyboard_mode_t prev_mode;
> void event_pre_cb(lv_event_t * e)
> {
>     lv_event_code_t code = lv_event_get_code(e);
>     lv_obj_t * obj = [...]

This works very well. Thank you so much.

closed 2023-06-23 by lanistor

bjfhskj, 2024-05-28:
Use lv_btnmatrix to create multiple buttons, can I don't redraw any button when I press it,so I can make a fast input and don't wait for the redraw ?

kisvegabor, 2024-05-28:
In the current implementation the whole keyboard is redrawn. If it takes 500 ms I think the system is very slow already. 

But I wonder why is is it low slow?
- What kind of MCU ds you use?
- Do you connect the display via SPI? If so what is the clock speed?
- What is resolution of the display?

bjfhskj, 2024-05-29:
> In the current implementation the whole keyboard is redrawn. If it takes 500 ms I think the system is very slow already.
> 
> But I wonder why is is it low slow?
> 
> * What kind of MCU ds you use?
> * Do you connect the display via SPI? If so what is the clock speed?
> * What is resolution of the display?

1. MCU: ESP32
2. I connect the display with SPI ,when the clock speed 40MHz,about 300ms redraw
3. resolution:480*320

I think if I can cancel redrawing, would the input speed be faster?But how to cancel it?

kisvegabor, 2024-05-30:
It redraws the whole keyboard because it would be hard to handle partial drawing in this case.

> MCU: ESP32
I connect the display with SPI ,when the clock speed 40MHz,about 200ms redraw
resolution:480*320

With these parameters redrawing the whole screen with 16 bit color depth should take 61 ms. As the keyboard is probably smaller than the whole screen, let's say it's only 30ms. 

So either the rendering is very slow or the SPI is not running at 40MHz for some reason.

bjfhskj, 2024-06-03:
Can I not repaint anything when I press the button, not just the entire keyboard, but also some parts. I think repaint is unnecessary, will it make the input much faster?

kisvegabor, 2024-06-05:
The problem is that when you press a widget the whole widget will be invalidated and the keyboard can not prevent it. So it would require a larger revamp on how area invalidation works but as this issue only rarely causes issues it's not a roadmap now.  
I'm sorry for that. :slightly_frowning_face: