lvgl/lvgl #4297: Monochrome, flush_cb output not correct?

epikao, 2023-06-19:
Hello,

I’m using LVGL version 8.3.5, with #define LV_COLOR_DEPTH 1, all other setup at default. (monochrome LCD)
Attached you can find the flush_cb record (Serial output).

### Describe the bug
Attached you can find the flush_cb record (Serial output).
[image]

First output looks ok, but second and next output are wrong to me. The x-Position of the text is not correct.
Why this? Do I miss maybe some additional setup on lv_conf ? Following my code to reproduce.

### To Reproduce
**Setup:**
```
    /*Initialize the display*/
    static lv_disp_drv_t disp_drv;
    lv_disp_drv_init( &disp_drv );
    /*Change the following line to your display resolution*/
    disp_drv.hor_res = TFT_WIDTH; //
    disp_drv.ver_res = TFT_HEIGHT; //
    disp_drv.flush_cb = lvgl_disp_flush;
    //disp_drv.rounder_cb = lvgl_rounder;
    //disp_drv.set_px_cb = lvgl_set_px;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register( &disp_drv );
...
...
    lv_obj_t *label = lv_label_create(lv_scr_act());
    lv_label_set_text(label, "Hello World!");
    lv_obj_set_pos(label, 2, 2);
```

**Flush:**
```
void lvgl_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
    uint8_t* lvglBuffer= (uint8_t*) color_p;

    uint8_t x1 = area->x1;
    uint8_t x2 = area->x2;
    uint8_t x1x2 = x2 - x1;
    uint8_t y1 = area->y1;
    uint8_t y2 = area->y2;
    uint8_t y1y2 = y2 - y1;

    uint8_t x = x1;
    uint8_t y = y1;

    Serial.print("x1: "); Serial.print(x1); Serial.print(" x2: "); Serial.print(x2);
    Serial.println("");
    while(y <= y2){
        Serial.print(lvglBuffer[(y*x1x2) + x]); 
        if(x > x2){
          y++;
          x = x1;
         Serial.println(" ");
        }
    }

    Serial.println(" ");
    Serial.println(" ");
    Serial.println(" ");

    lv_disp_flush_ready( disp );
}
```

kisvegabor, 2023-06-19:
What is the expected value of X?

> uint8_t x1x2 = x2 - x1;

I should be `uint8_t x1x2 = x2 - x1 + 1;`
It leads to error in `lvglBuffer[(y*x1x2) + x]);`

epikao, 2023-06-19:
Ok thanks. I did a new output with comments to the expected values, see attachment.

[image]

kisvegabor, 2023-06-19:
> It should be uint8_t x1x2 = x2 - x1 + 1;

Have you tried this modification?

epikao, 2023-06-19:
> Have you tried this modification?
Yes, as you can see in the pictures the text looks not anymore italic..

kisvegabor, 2023-06-19:
Ah, okay. I'm trying to figure out the error `y` first. Does the text look incorrect on the display too or only in the log?

How can I reproduce the error in 2nd case? What is the exact screen resolution, label position, and invalidated area?

epikao, 2023-06-19:
label position you can see above in setup:
lv_obj_set_pos(label, 2, 2);

Resolution of the display is 132 x 44, and it's the same on the display as on my log.
Attached a longer record of the output.

[flush_cb_output_2_long.txt](https://github.com/lvgl/lvgl/files/11788017/flush_cb_output_2_long.txt)

kisvegabor, 2023-06-19:
Which font do you use?

epikao, 2023-06-19:
> Which font do you use?

mm, didn't defined, so I would say it is the default font &lv_font_montserrat_14.
I have only 3 lines GUI code:
```
    lv_obj_t *label = lv_label_create(lv_scr_act());
    lv_label_set_text(label, "Hello World!");
    lv_obj_set_pos(label, 2, 2);
```

kisvegabor, 2023-06-20:
No matter how I try I can't reproduce it. 

Could you create an [Eclipse simulator](https://github.com/lvgl/lv_port_pc_eclipse/tree/release/v8.3) project and reproduce the issue there? Don't forget to update LVGL to the latest version form `release/v8.3` branch.

epikao, 2023-06-20:
> Could you create an [Eclipse simulator](https://github.com/lvgl/lv_port_pc_eclipse/tree/release/v8.3) project and reproduce the issue there? Don't forget to update LVGL to the latest version form `release/v8.3` branch.

With 8.3.7 it works. Great! Thank you.

closed 2023-06-20 by epikao

kisvegabor, 2023-06-20:
Oh, I haven't expected that, but great :slightly_smiling_face: