lvgl/lvgl #4298: odd code in indev.c

kdschlosser, 2023-06-19:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Not really a bug, It's code that doesn't make any sense why it is done the way it is. 

This is the code.

```c
    if(indev_obj_act == NULL) {
        if(indev_obj_act == NULL) {
            indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_bottom(disp), &indev->pointer.act_point);
        }
        indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_sys(disp), &indev->pointer.act_point);
        if(indev_obj_act == NULL) {
            indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_top(disp), &indev->pointer.act_point);
        }
        if(indev_obj_act == NULL) {
            indev_obj_act = lv_indev_search_obj(lv_disp_get_scr_act(disp), &indev->pointer.act_point);
        }
        new_obj_searched = true;
    }
```

I am thinking there  was a boo boo made and the code should read
```c
    if(indev_obj_act == NULL) {
        indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_bottom(disp), &indev->pointer.act_point);

        if(indev_obj_act == NULL) {
            indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_sys(disp), &indev->pointer.act_point);
        }
        if(indev_obj_act == NULL) {
            indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_top(disp), &indev->pointer.act_point);
        }
        if(indev_obj_act == NULL) {
            indev_obj_act = lv_indev_search_obj(lv_disp_get_scr_act(disp), &indev->pointer.act_point);
        }
        new_obj_searched = true;
    }
```

Now that I have looked at it, it could cause a bug because the lowest layer possible would be the sys layer. where as the lowest layer possible is the bottom layer with the correction.

Here is a link to the spot in question
https://github.com/lvgl/lvgl/blob/1bfdecf585bb50fbb22ed1c38450b5c9eb600196/src/core/lv_indev.c#L1005

kisvegabor, 2023-06-19:
Thank you for pointing it out. It was really incorrect.

The correct order should be: sys, top, normal, bottom.

closed 2023-06-19 by kisvegabor

kdschlosser, 2023-06-19:
knew it didn't look right. didn't know how it needed to be tho. 🤣 
There are benefits that come from me reading the code and asking questions when something doesn't make sense to me 😉

kdschlosser, 2023-06-19:
I see how the mechanism works and I am wondering if the code can be written in this manner? would there be any benefit to it being done in this manner? IDK if c99 even supports this syntax. This is a learning question.

```c
if(indev_obj_act == NULL) {
    if((indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_sys(disp), &indev->pointer.act_point)) == NULL) {
       if((indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_top(disp), &indev->pointer.act_point))== NULL) {
            if((indev_obj_act = lv_indev_search_obj(lv_disp_get_scr_act(disp), &indev->pointer.act_point)) == NULL) {
                indev_obj_act = lv_indev_search_obj(lv_disp_get_layer_bottom(disp), &indev->pointer.act_point);
            }
        }
    }
    new_obj_searched = true;
}
```

kisvegabor, 2023-06-20:
> There are benefits that come from me reading the code and asking questions when something doesn't make sense to me

Definitely, keep it up please :slightly_smiling_face: 

> Would there be any benefit to it being done in this manner? IDK if c99 even supports this syntax.

It also work and a good solution. It's just a personal preference that I'd like to avoid deep nesting where possible.

kdschlosser, 2023-06-20:
I didn't know if the code compiled any differently that would make one faster than the other

kisvegabor, 2023-06-21:
This is not a performance bottle neck, so here it's better to optimize for readability :smile:

commit 5d84ac7a (2023-06-19): fix(indev): fix getting the active object from bottom/sys layer

fixes: #4298