lvgl/lvgl #4299: lv_img_buf_get_px_color() not able to read the color of the image

krupis, 2023-06-19:
### Introduce the problem
For my particular project, I would like to display an image that contains many different colors. The display that I use supports touch.
When the user touches the display, the interrupt is triggered and I get the X/Y coordinates of where the touch has been recored.

With the X/Y coordinates I would like to extract the image color code.

I have created a thread for the LVGL forum but I did not manage to get any help. I am hoping to get some help here as I think this would be relevant for others as well.

### Examples and cases

I have already tried to use ```lv_img_buf_set_px_color``` and ```lv_img_buf_get_px_color``` but with no luck. For details please see the forum post:
https://forum.lvgl.io/t/reading-rgb-color-values-of-the-image-specific-coordinates/12035/2

I would appreciate any help regarding how to use lv_img_buf_get_px_color or any different method to extract the color from the image. Thanks in advance.

kisvegabor, 2023-06-19:
Which LVGL version are you using?

krupis, 2023-06-19:
> Which LVGL version are you using?
Sorry, I forgot to mention that in my initial post.
I am using version 8.3.6

kisvegabor, 2023-06-20:
Ah, I see the problem. In `lv_color_t pixel_color = lv_img_buf_get_px_color(ui_redsquare, 50, 50,lv_color_make(0, 0, 0));` you should pass the image, and not  an `lv_obj_t *`.

So 
```c
lv_color_t pixel_color = lv_img_buf_get_px_color(&ui_img_red_square_png, 50, 50,lv_color_make(0, 0, 0));
```

krupis, 2023-06-20:
> Ah, I see the problem. In `lv_color_t pixel_color = lv_img_buf_get_px_color(ui_redsquare, 50, 50,lv_color_make(0, 0, 0));` you should pass the image, and not an `lv_obj_t *`.
> 
> So
> 
> ```c
> lv_color_t pixel_color = lv_img_buf_get_px_color(&ui_img_red_square_png, 50, 50,lv_color_make(0, 0, 0));
> ```

Ah that makes sense.
I have tried the following:

```

LV_IMG_DECLARE(red_square)

void get_img_color(){
    lv_color_t pixel_color = lv_img_buf_get_px_color(&red_square, 50, 50,lv_color_make(0, 0, 0));
    uint8_t red = LV_COLOR_GET_R(pixel_color);
    uint8_t green = LV_COLOR_GET_G(pixel_color);
    uint8_t blue = LV_COLOR_GET_B(pixel_color);
    printf("Pixel color: R:%d, G:%d, B:%d \n", red, green, blue);
}

```

red_square is declared in my red_square.c component

The result is:
```
Pixel color: R:31, G:0, B:0
```

It seems to work fine but not quite as expected. The color of my red square is R:255 G:0 B:0

Is there any particular reason why the pixel color returned is not fully red?

kisvegabor, 2023-06-20:
> Is there any particular reason why the pixel color returned is not fully red?

Yes, you are using RGB565 color format, where Red has 5 bits (max 31). You can use this color in any styles to set a pure red color for something as all APIs uses the same color format.

krupis, 2023-06-20:
> > Is there any particular reason why the pixel color returned is not fully red?
> 
> Yes, you are using RGB565 color format, where Red has 5 bits (max 31). You can use this color in any styles to set a pure red color for something as all APIs uses the same color format.

Thanks for the response but can you help me understand a little more.

What makes you think that I use RGB565 color format? Is there any way to change it?

For this particular example, I use red square but my final application is going to use color gradient so it would be ideal for me to be able to read whole 8 bits of color information.

How am I going to distinguish red 100 from red 255 if its only 5 bits?

kisvegabor, 2023-06-20:
> What makes you think that I use RGB565 color format?

I though because the read value was 32 for a fully red image. :slightly_smiling_face: 

>  Is there any way to change it?

You can change `LV_COLOR_DEPTH` to 32 in `lv_conf.h`. But note that it should be set according to color depth of your display.

> How am I going to distinguish red 100 from red 255 if its only 5 bits?

For the sake of simlicity you can use `uint32_t c32 = lv_color_to32(color)`. In this the first byte is always `0xff` (alpha channel) and the other bytes contain the R G B bytes.

krupis, 2023-06-20:
> 

Thanks for explaining. I have been reading up about my display and more about the LVGL. It is getting more and more clear for me.

I use 1.9 inch ST7789 Display. The color depth is 16 bit (RGB556).

I am still not fully understanding how does that affect the color of my display. Does that mean that displays with 16 bit color depth are not capable of displaying complex colors accurately?

kisvegabor, 2023-06-21:
> I am still not fully understanding how does that affect the color of my display. Does that mean that displays with 16 bit color depth are not capable of displaying complex colors accurately?

That true. On 16 bit (RGB565) your have only 5 bits (32 shades) for red. It's usually ok but might not be enough for some very soft gradients.

closed 2023-06-23 by krupis