lvgl/lvgl #4301: Changing style to btnmatrix when a button is pressed

HaoboGu, 2023-06-20:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I want to change the btnmatrix pressed style when one of the buttons in the btnmatrix is pressed. 
Use `lv_obj_add_style(btn_matrix, &style_btn_pressed,
                     LV_PART_ITEMS | LV_STATE_CHECKED)` cannot do this
### To Reproduce

### Expected behavior

The pressed button's style is changed to style_btn_pressed

### Screenshots or video

kisvegabor, 2023-06-20:
It looks like this for me, which seems correct based on the styles you add.

[image]

Do you see the same result?

HaoboGu, 2023-06-20:
Could you try `lv_style_set_transform_height(&style_btn_pressed, 10);` or `lv_style_set_height(&style_btn_pressed, 10);`? The button size is not changed

kisvegabor, 2023-06-21:
Yes, it's because the coordinates of the buttons are set automatically. See here: https://docs.lvgl.io/8.3/widgets/core/btnmatrix.html#control-buttons

kisvegabor, 2023-06-23:
How did you make it scrollable? Please attach a short codesnippet.

HaoboGu, 2023-06-23:
I just set the height of the btnmatrix larger than the background and screen size. Here is the code:
```c
btn_matrix_screen = lv_obj_create(NULL);
lv_obj_set_size(btn_matrix_screen, 300, 200);

btn_matrix = lv_btnmatrix_create(btn_matrix_screen);
lv_btnmatrix_set_map(btn_matrix, btn_map);
lv_obj_set_size(btn_matrix, 320, 300);
lv_obj_scroll_to_x(btn_matrix, 0, LV_ANIM_OFF);
lv_obj_align(btn_matrix, LV_ALIGN_TOP_MID, 0, 0);
```

kisvegabor, 2023-06-23:
In this case really the background (i.e. the screen) will be scrollable. However you shouldn't change the size of the screen. The screen is always display sized.

HaoboGu, 2023-06-25:
@kisvegabor thanks for the explanation! If I want to achieve btnmatrix scrolling, and keep background unchanged, what's the best way?

kisvegabor, 2023-06-26:
You can wrap the button matrix into an `lv_obj` and make the background of the btnmatrix transparent:
```c
static const char * btnm_map[] = {"1", "2", "3", "4", "5", "\n",
                                  "6", "7", "8", "9", "0", "\n",
                                  "Action1", "Action2", ""
                                 };

void lv_example_btnmatrix_1(void)
{
    lv_obj_t * wrapper = lv_obj_create(lv_scr_act());
    lv_obj_set_size(wrapper, 300, 200);

    lv_obj_t * btnm1 = lv_btnmatrix_create(wrapper);
    lv_obj_set_size(btnm1, 400, 150);
    lv_obj_set_style_bg_opa(btnm1, 0, 0);
    lv_obj_set_style_border_opa(btnm1, 0, 0);

    lv_btnmatrix_set_map(btnm1, btnm_map);
```

[image]

I've never tried a scrolling button matrix and as you can see there is an issue while scroll it the buttons are activated. I've just pushed a fixed for it, so it should work normally now.

HaoboGu, 2023-06-27:
Thanks very much! It worked as expected.

HaoboGu, 2023-06-27:
> I've never tried a scrolling button matrix and as you can see there is an issue while scroll it the buttons are activated. I've just pushed a fixed for it, so it should work normally now.

I found this issue also exists when using gesture: 

https://github.com/lvgl/lvgl/assets/8640918/395208b2-d371-4c73-99ae-52131bcfec48

HaoboGu, 2023-07-12:
@kisvegabor any updates?

HaoboGu, 2023-07-14:
@kisvegabor Yes, I agree. 

I also tried on several devices, including my iPhone. No button is activated if my finger slides out of the button, the gesture/scroll can be triggered in this case.

kisvegabor, 2023-07-14:
Great, I've pushed a fix to `master`.

commit 1f58504e (2023-07-14): fix(btnmatrix): if a slid to a new button, discard the current button and don't press any buttons

fixes #4301"
closed 2023-07-14 by kisvegabor