lvgl/lvgl #4302: LVGL 8.3 Image decoding interface requirements

htk719809837, 2023-06-20:
hello, I use an RTOS device, which has no file system, but I need to load a dynamically transformed image. The w, h, depth and other information of this image have been obtained through the protocol, and the binary data of jpg image has been read and stored in the buf array. Is there an interface that can decode this binary array and call it to LVGL? I look forward to hearing from you

kisvegabor, 2023-06-20:
If you have a JPEG image as an array you can do this:
```c
  static lv_img_dsc_t img_dsc;
  img_dsc.header.w = w;
  img_dsc.header.h = h;
  img_dsc.header.cf = LV_IMG_CF_RAW;
  img_dsc.header.always_zero = 0;
  img_dsc.data = jpg_data;
  img_dsc.data_size = byte_count_of_jpg_data;

  lv_img_set_src(img1, &img_dsc);
```

You should also enable `LV_USE_SJPG 1` in `lv_conf.h`.

htk719809837, 2023-06-20:
jpg_data is the jpg image data that has been decoded;
At present, the buf I have here is obtained by reading the jpg image in linux (without decoding), passing it to the rots system using a protocol. The binary data of the image is stored inside the buf array. Does this buf have a way to parse it into jpg_datane by calling the internal function of lvlg

htk719809837, 2023-06-20:
For example, I take a picture from the camera using the linux standard open("xxx/xx.jpg", O_RDONLY, 0); The function saves the image binary data in the buf array, and then uses the protocol to pass it to the rtos system without a file system. How do I decode this image on the rtos system and use it

closed 2023-06-20 by htk719809837