lvgl/lvgl #4304: [v8.3][Docs] Display Software Rotation: Mention Incompatibility with `full_refresh` and `direct_mode`

kolliemo, 2023-06-21:
As far as I understand, software rotation is incompatible with using
either `full_refresh` or `direct_mode`. Is that correct or am I
misunderstanding something here?

It that is correct, what about adding a small mention of that to the
[display porting docs](https://docs.lvgl.io/8.3/porting/display.html)? 
Personally, this has caused me quite a bit of confusion why my display
would glitch when trying to rotate, until I found this mentioned in some
github issues and some few forum threads, e.g.:

https://github.com/lvgl/lvgl/issues/4060#issuecomment-1486538873

https://github.com/lvgl/lvgl/issues/3829

https://forum.lvgl.io/t/framebuffer-refresh-problem/10043/6

kisvegabor, 2023-06-23:
Hi,

I agree that it should be mentioned there. Could you send a pull request with this update?

kolliemo, 2023-06-23:
Will do, just wanted to double check I didn't misunderstand, thanks!

kisvegabor, 2023-06-23:
Merged, thank you!

PR #4308 (2023-06-23): V8.3 Docs (Display Porting): Mention incompatibility between software rotation and `direct_mode` or `full_refresh`
Although software rotation is not compatible with either of these driver settings, this is not mentioned in the display porting documentation yet. (I only found it mentioned -- for full refresh -- in a lv_refr source comment, and -- for direct mode -- buried in some github issues / forum threads.)

Fixes: #4304

I added a little note aswell, that one should rotate the pixels manually e.g. in the `flush_cb` -- I suspect one will have to also adjust the `lv_disp_drv_t` width/height  [... cut]
closed 2023-06-23 by kisvegabor