lvgl/lvgl #4312: font parameter has no effect in function lv_theme_default_init()

hyansuper, 2023-06-24:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

the `font` parameter has no effect in function [`lv_theme_default_init()`](https://github.com/lvgl/lvgl/blob/master/src/themes/default/lv_theme_default.c#L652).
except for changing the ✔ in checkbox(https://github.com/lvgl/lvgl/blob/master/src/themes/default/lv_theme_default.c#L430).

you can only change default font by `#define LV_FONT_DEFAULT ...`
### Expected behavior
able to set default large/normal/small fonts like described in the [doc](https://docs.lvgl.io/master/overview/style.html#themes)

### Screenshots or video

kisvegabor, 2023-06-26:
Hi,

I've just pushed a fix. Does it work on your end too?

hyansuper, 2023-06-26:
Very fast fix, Thank you