lvgl/lvgl #4316: Cannot set label text dynamically

edword01, 2023-06-26:
### Describe the bug
#### hardware
ESP32S3 + 0.96`  SPI screen with st7735s

i already create a label with a initial value=0,  and i need a function that can dynamically. so i try several way follow:
(1) using function: ```lv_obj_add_flag(ui_Label4Oidcode, LV_OBJ_FLAG_CLICKABLE); ``` to enable click action, add label obj to group, and binding with a custom input device. cannot work.
(1) adding another buttion with hidden flag to receive the event. this time i use :  ```lv_event_send(ui_Button1, LV_EVENT_PRESSED, NULL); ``` to send event,  and my event_cb look like: 

### To Reproduce

```c
void ui_event_Button1(lv_event_t * e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t * target = lv_event_get_target(e);
    if(event_code == LV_EVENT_PRESSED) {
        ESP_LOGE("UI", "test=%d", event_code);
        _ui_label_set_property(ui_Label4Oidcode, _UI_LABEL_PROPERTY_TEXT, "200");
    }
}
```

### Expected behavior

bug1:  now, the text of the label i created should be changed,  but it does't happend.
bug2:  i get some distorted pixels at the bottom of the screen.
### Screenshots or video

distorted pixels:
[image]

correct effect:
[image]

kisvegabor, 2023-06-26:
Hi,

1. Are you sure your touchpad is working correctly? Try logging something from the PRESSED event. 
2. Are you sure your MCU is running (not crashed). Try logging something from the the main loop
3. It seems there is an offset on the display as the "Demo" text is clipped. Probably there is a bug in your `flush_cb`

edword01, 2023-06-26:
> Hi,
>
> 1. Are you sure your touchpad is working correctly? Try logging something from the PRESSED event.
> 2. Are you sure your MCU is running (not crashed). Try logging something from the the main loop [...]

hello， i already fix the error pixels, and i want to know how to binding buttion input device with lvgl component?? 
and generate LV_EVENT_PRESSED ??? 
now, i use ```lv_event_send(ui_Button1, LV_EVENT_PRESSED, &tmp_code);``` can change label correctly. but i don't want to use this method,  i want to use event_cb way, my event_cb like this: 
```
_read(lv_indev_drv_t *indev_drv, lv_indev_data_t *data)
{
    if (oid_data_flag)
    {
        data->state = LV_INDEV_STATE_PRESSED;
        // ESP_LOGI(TAG, "Pressed");
    }
    else
    {
        data->state = LV_INDEV_STATE_RELEASED;
        // ESP_LOGE(TAG, "Released");
    }
    //ESP_LOGI(TAG, "enc_diff %d, status %d", data->enc_diff, data->state);

}
```
this can be runned ok.  but cannot enter this function:
```
void ui_event_Button1(lv_event_t * e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t * target = lv_event_get_target(e);
    int* code;
    code = lv_event_get_param(e);
    if(event_code == LV_EVENT_PRESSED) {
        ESP_LOGE(TAG, "test");
        char buf[10];
        intToCharArray(*code, buf);
        _ui_label_set_property(ui_Label4Oidcode, _UI_LABEL_PROPERTY_TEXT, buf);
    }
}
```
if i use ```lv_event_send(ui_Button1, LV_EVENT_PRESSED, &tmp_code);``` , ```ui_event_Button1```function can run ok. 
how to use normal lvgl input device ?????

kisvegabor, 2023-06-26:
Is the label clickable? Note that the labels are not clickable by default, so if you want to make it catch the press events, you need to add `lv_obj_add_flag(label, LV_OBJ_FLAG_CLICKABLE);`.

closed 2023-07-18 by lvgl-bot