lvgl/lvgl #4324: 8.3.6 - ButtonMatrix While loops can crash

Satnet, 2023-06-28:
In some cases the buttonMatrix while loops can crash. I have spent a week trying to find the root cause without success.

In my case I am using encryption to encrypt wireless messages. This somehow collides with lvgl buttonMatrix, I know it sounds absolutely insane but this is the case. So I used a completely different encrypt library, same thing happens.

I get messages every 5ms so it's super fast. No idea if that has something to do with it.
Not using the button matrix makes everything work fine. I made my own button matrix with ordinary buttons, using for loops and if statement inside the loop to break the loop if \n or \0. That works fine.

I understand this is very little info. I wish I could give more. I will continue to try but hopefully Kisvegabor find the issue quicker.

pete-pjb, 2023-06-29:
Hi @Satnet,

I have added some fixes for this type of behaviour [here](https://github.com/lvgl/lvgl/commit/07bce7466d471a15ba241a9db3239f7a116b0b46) can you try updated to the latest 8.3.7 as it may well address your issues.

Let me know how it goes.

Kind Regards,

Pete

Satnet, 2023-06-29:
> Hi @Satnet,
> 
> I have added some fixes for this type of behaviour [here](https://github.com/lvgl/lvgl/commit/07bce7466d471a15ba241a9db3239f7a116b0b46) can you try updating to the latest 8.3.7 as it may well address your issues.
> 
> Let me know how it goes.
> 
> Kind Regards,
> 
> Pete

Thanks for your reply!
Unfortunately this doesn't help. CrashReport blames line 842 but that doesn't mean that this line is the root cause.

pete-pjb, 2023-06-29:
That's a shame... 

Are you able to create a minimal code snippet to reproduce the problem? 

Also you say:
> In some cases the buttonMatrix while loops can crash.

Can you give a bit more information about the crash?  Is it a memory access violation or something like that? Can you post a screen shot?

Can you share your button map declaration?

Also as you say the crash report may be incorrect, if you have memory corruption it could be caused by anything in your system... Are you running an OS of any description... are there multiple threads of execution?

Kind Regards,

Pete

Satnet, 2023-06-29:
It's basically impossible to make a code snippet since this is very precise. Seems like some sort of memory issue. If I changed other parts of the code, which is not connected the matrix, it can make it work, alltho the text on the buttons are gibberish. I can share a printscreen privately with you over email if you wish. Or even a short videoclip.

Screenshoot of Crash report
[image]

Button matrix declaration
`static const char * MATRIX[] = {"1", "2", "3", "4", "5", "6", ""};`

No OS running, this is running on a Teensy 4.1.

Thanks!

pete-pjb, 2023-06-29:
Hi @Satnet ,

My email is: petebone00@gmail.com

Kind Regards,

Pete

pete-pjb, 2023-06-29:
Hi @Satnet ,

Also, have you tried enabling the asserts in lv_conf.h?

```c
/*-------------
 * Asserts
 *-----------*/

/*Enable asserts if an operation is failed or an invalid data is found.
 *If LV_USE_LOG is enabled an error message will be printed on failure*/
#define LV_USE_ASSERT_NULL          1   /*Check if the parameter is NULL. (Very fast, recommended)*/
#define LV_USE_ASSERT_MALLOC        1   /*Checks is the memory is successfully allocated or no. (Very fast, recommended)*/
#define LV_USE_ASSERT_STYLE         1   /*Check if the styles are properly initialized. (Very fast, recommended)*/
#define LV_USE_ASSERT_MEM_INTEGRITY 0   /*Check the integrity of `lv_mem` after critical operations. (Slow)*/
#define LV_USE_ASSERT_OBJ           0   /*Check the object's type and existence (e.g. not deleted). (Slow)*/
```
Personally I have caught a number of bugs with those...

Certainly worth a try!

Kind Regards,

Pete

Satnet, 2023-06-29:
I use that sometimes but it have not helped this time.

kisvegabor, 2023-06-29:
Can you reproduce the issue in a simulator on PC?

pete-pjb, 2023-07-14:
Hi @Satnet ,

I am going to assume you have solved your problem and close this issue, if you need to reopen please feel free and we will respond.

Kind Regards,

Pete

closed 2023-07-14 by pete-pjb