lvgl/lvgl #4328: Press an object obj1 and slide to another object obj1, obj1 received LV_EVENT_CLICKED

MiltonWuSZ, 2023-06-30:
### About the question
My input device is touchscreen，lvgl version 8.3.7
I create two object `obj1` and `obj2`. I pressed  `obj1` and slide to `obj2`, `obj1` received  event `LV_EVENT_CLICKED`.
My expectation is that no click event occurs, in this scenario, the user cancels the click.
###  Examples and cases
```c
static void event_cb(lv_event_t * e)
{
    if (e->code == LV_EVENT_CLICKED) {
        printf("obj1 clicked\n");
    }
}
static void event2_cb(lv_event_t * e)
{

    if (e->code == LV_EVENT_CLICKED) {
        printf("obj2 clicked\n");
    }
}
void obj_event_test(void)
    lv_obj_t * content = lv_obj_create(lv_scr_act());
    lv_obj_remove_style_all(content);
    lv_obj_set_flex_flow(content, LV_FLEX_FLOW_COLUMN);
    lv_obj_set_size(content, 300,600);
    lv_obj_set_align(content, LV_ALIGN_CENTER);
    lv_obj_set_flex_align(content, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);
    
    lv_obj_t * obj1 = lv_obj_create(content);
    lv_obj_add_flag(obj1, LV_OBJ_FLAG_CLICKABLE);  
    lv_obj_set_style_bg_color(obj1,lv_color_hex(0xff0000), 0);
    lv_obj_set_style_bg_opa(obj1,255, 0);
    lv_obj_set_size(obj1, 100,100);
    lv_obj_t * txt = lv_label_create(obj1);
    lv_label_set_text(txt, "obj1");  
    lv_obj_add_event_cb(obj1, event_cb, LV_EVENT_CLICKED, NULL);
    
    lv_obj_t * obj2 = lv_obj_create(content);
    lv_obj_add_flag(obj2, LV_OBJ_FLAG_CLICKABLE);  
    lv_obj_set_style_bg_opa(obj2,255, 0);
    lv_obj_set_size(obj2, 100,100);
    lv_obj_t * txt2 = lv_label_create(obj2);
    lv_label_set_text(txt2, "obj2");  
    lv_obj_add_event_cb(obj2, event2_cb, LV_EVENT_CLICKED, NULL);
}
```

kisvegabor, 2023-07-03:
You can use `lv_obj_clear_flag(obj1, LV_OBJ_FLAG_PRESS_LOCK);`.

MiltonWuSZ, 2023-07-03:
> You can use `lv_obj_clear_flag(obj1, LV_OBJ_FLAG_PRESS_LOCK);`.

When I add `lv_obj_clear_flag(obj1, LV_OBJ_FLAG_PRESS_LOCK);` and  `lv_obj_clear_flag(obj2, LV_OBJ_FLAG_PRESS_LOCK);` 
I press `obj2` and moved to `obj1`, the `obj1` also recived `LV_EVENT_CLICKED`. My expectation is that both `obj1` and `obj2` have no click event occurs, in this scenario, the user cancels the click.

kisvegabor, 2023-07-03:
It's also possible but needs some extra work. `LV_OBJ_FLAG_PRESS_LOCK` needs to be cleared and you need to add an event to the objects like this:
```
static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    if(code == LV_EVENT_PRESS_LOST) {
        lv_indev_wait_release(lv_indev_get_act());
    }
}
```

MiltonWuSZ, 2023-07-04:
dd

MiltonWuSZ, 2023-07-04:
> It's also possible but needs some extra work. `LV_OBJ_FLAG_PRESS_LOCK` needs to be cleared and you shoud to add an event to the objects like this:
> 
I add events  `LV_EVENT_PRESS_LOST`  in `obj1` and `obj2`,  it works，thank you very much. But if every button has to be handled like this, I think it's a bit bad smell. 
Why not check whether `indev_obj_act` is the object pressed for the first time before the `indev_proc_release` function sends `LV_EVENT_CLICKED` or `LV_EVENT_RELEASED` events, if not, do not send.

kisvegabor, 2023-07-05:
The behavior could be easily changed in `lv_indev.c`, however we can't support all the scenarios by default. :slightly_frowning_face: 

In similar cases I suggest creating a functions like:
```c
lv_obj_t * ui_my_btn_create(lv_obj_t * parent)
{
    lv_obj_t * btn = lv_btn_create(parent);
    ...custom things...
   return btn;
}
```

MiltonWuSZ, 2023-07-06:
On Android or iPhone, when tapping a btn and then swiping to another btn, neither btn is tapped. I think it's a more common scenario？

kisvegabor, 2023-07-06:
Actually we can do it when `PRESS_LOCK` is disabled. I've just tested on my phone that if you move your finger back to the button, it will not send an event. 

So if you slide your finger off the button, it's over, you can't do anything until you release and press the screen again. 

Can you confirm it?

MiltonWuSZ, 2023-07-07:
Yes, I agree with the test on the phone:  if you slide your finger off the button, it's over, you can't do anything until you release and press the screen again.
But on the lvgl, if the screen has two buttons `btn1` and `btn2`, and `LV_OBJ_FLAG_PRESS_LOCK ` are disabled, there are the following differences ：
1. I press `btn1` and then  slider my finger off the `btn1`, and back to the `btn1`,  the `btn1` will receive `LV_EVENT_CLICKED`.
2. I press `btn1` and then  slider my finger to the `btn2`,  the `btn2` will receive `LV_EVENT_CLICKED`.

kisvegabor, 2023-07-09:
Added in `master`.

Note that [a large update](https://github.com/lvgl/lvgl/issues/4011) was `master` to master a few days ago.

closed 2023-07-09 by kisvegabor

incity, 2023-08-18:
@kisvegabor I pull the latest code, but it seems not work with the current lv_drivers?

kisvegabor, 2023-08-18:
True, in v9 we will have the driver in the lvgl repo. So lv_driver will be obsolete for v9.

commit ecf1ea4d (2023-07-09): fix(indev): do not search for new obj if slid out of the obj  and no PRESS_LOCK

fixes #4328