lvgl/lvgl #4333: tiny_ttf display very slow

Leadrive, 2023-07-02:
Hi, I copy the master branch tiny_ttf to v8.3.7
the ttf file about 800K, include Chinese, place in /spiffs
and add the lvgl file system stdio, and set 16K cache.

````
[*] File system on top of stdio API                                                                                                                                                                                                
(47)    Set an upper cased letter on which the drive will accessible (e.g. 'A' i.e. 65 )                                                                                                                                           
(/)     Set the working directory                                                                                                                                                                                                  
(16384) >0 to cache this number of bytes in lv_fs_read() 
````

After run, can found the ttf file and open and show text ok.
But the screen display very slow, about one more minute to show.

The same ttf file, if use freetype to display, about one second to show.

How to check the problem?
Thank you.
@codewitch-honey-crisis

codewitch-honey-crisis, 2023-07-02:
If you are loading it from an SD card or other external IO it will be slow because it has to stream off of the storage. Freetype loads the font into RAM first. You could load it into RAM assuming you have enough, or you can embed it in your flash as a header. Either way will yield a dramatic performance improvement over using external storage.

Leadrive, 2023-07-02:
@codewitch-honey-crisis 
Thanks your reply.
ESP32S3
I want to try it as header.
How to make ttf=>.h file?

codewitch-honey-crisis, 2023-07-02:
You can use this tool [https://honeythecodewitch.com/gfx/converter](https://honeythecodewitch.com/gfx/converter)

Just choose C++ and uncheck Arduino. You will get a byte array you can pass to tiny TTF.

Leadrive, 2023-07-03:
Thank you very much.@codewitch-honey-crisis
Now display ok, I will test it...

When use the freetype, the program will handup random. lv_timer_handler() did not return.
I guess maybe the freetype have some bugs.
So I want to try tiny_ttf.

closed 2023-07-03 by Leadrive