lvgl/lvgl #4337: CPython binding updates

kdschlosser, 2023-07-03:
Thought it would be best to create a separate issue for progress reports with the CPython Binding.

While the current version of it does work for the most part there are several hiccups in it that I am not too fond of. One of them specifically needing to set things as globals and not being able to pass the returned object of a function call directly to another function without having to set it into a variable first. It all has to deal with how CFFI doesn't internally manage references. To write code to handle that kind of a thing would really make the binding more complex and because it would be handled in Python it would be slower. 

Now I don't want to have to generate all of that in C code either. It would make the generation code really complex and difficult to maintain. Python has the ability to access dynamic libraries and that ability is built into it. The code to handle the reference counting is already done and it is done in C code. This ability uses FFI just like CFFI does but it does not allow for accessing any "types" directly like CFFI does. So things like structures and unions have to have a python representation of them written. The binding automatically generates all of the code needed so there is no concern there.

The point I am at now is the generation script works and I am able to import the generated module without any errors. This tells me that the compiled shared library is accessible and the generated code is correct and contains no syntax errors at all. Through code inspection I am able to tell if there are any references to objects that don't exist and there are no reports of issues there.

kdschlosser, 2023-07-03:
I still have to test it on linux and maybe someone would be able to help me out in this arena as I am not fully versed on gcc and how to deal with exporting functions from aa shared library. when compiling with MSVC there are special compiler switches that have to be used and a definition file needs to be created to instruct the linker what the exported functions are.

kdschlosser, 2023-07-03:
On a side note. I have altered how the python build system (setuptools) works so now the build on my 32 logical processor ThreadRipper CPU will use all 32 logical processors when compiling. Now it only takes 8.7 seconds from start to finish to generate the python code and to compile LVGL. It does this without creating multiple processes of Python so the RAM overhead is a whole lot smaller. The folks that wrote setuptools/distutils either didn't think about or don't know how an OS manages multiples of the same process running and that it would automatically round robin the core used for a new process. So hat I did was instead of spawning additional python processes to handle the compiling I create "n" threads where n is the number of logical processors the machine has and each thread is handed a lost of files to compile and each thread runs the compiler on each file consecutively. what core the compiler runs on is managed by the OS. The main thread is stalled using a mutex until the "worker" threads have all finished and the main thread then calls the linker to put all of the compiled objects together.

In my use case because I have 32 logical processors the build process is somewhere in the neighborhood of 20 times faster. On a machine that has 4 logical processors I would expect it to be about 3 times faster and on a machine using 2 cores there might be a boost of about 1.75 times.

kisvegabor, 2023-07-03:
Thank you for opening this issue. It will be really simpler to follow it here.

> One of them specifically needing to set things as globals and not being able to pass the returned object of a function call directly to another function without having to set it into a variable first. 

How complicated would be to manage these?

kdschlosser, 2023-07-03:
That was the issue when I was using CFFI. It wasn't keeping track of references which is why I was not able to do that. Trying to do it on the python side of things would have been really hard to do because how the garbage collection works. Python has built into it the ability to access C libraries and it keeps track of references. It requires a lot more code on the python side of things to work properly because of having to create python representations of structures and unions where using CFFI didn't.

The reference issue would be solved using ctypes (built into Python) which is the reason why I have removed using CFFI in favor of ctypes. I could not come up with a solution to the reference issue on the python side of things that would keep references to things that went out of scope. I had to maintain a reference to the object in Python code which ultimately ends up as a memory leak because of not knowing when the reference could be deleted so it could get garbage collected.

From my past experience using ctypes I believe it will solve the issue of the references because it keeps track of them in C code and not in Python code.

kdschlosser, 2023-07-03:
Because of the large amount of code > 50000 lines there is a lot of memory consumption by Python. about 70mb of memory used to load the module and library. while this isn't a problem for a desktop it might be a problem for an MCU that is able to run Linux and has a reduced amount of memory.

That being said. I can break it up into different pieces. something that would follow the same file structure as the header files. This way a person could import a specific "header" for the pieces they use. It all depends on how interwoven the header files are to each other that would dictate how much memory ends up being used.

kdschlosser, 2023-07-04:
HA!!!! YES!!! It works!!!

I am super excited about this thing working. There is only a single hitch to it tho and once again it's because of reference handling. The one aspect of ctypes where the references are not handled is the function pointers, specifically the function that gets passed is not being held in the object. so I am not able to pass a callback directly to a function... kind of. It's a bit of a goofy thing that has to be done to make it work but it's really not a big deal to do.

here is a code example.

```python

import lvgl as lv
import time

lv.init()
disp = lv.sdl_window_create(480, 320)
group = lv.group_create()
lv.group_set_default(group)
mouse = lv.sdl_mouse_create()
keyboard = lv.sdl_keyboard_create()
lv.indev_set_group(keyboard, group)

@lv.anim_custom_exec_cb_t
def anim_x(a, v):
    print('lv.obj_set_x(label, v)')
    print(a, label, v, v.value)
    lv.obj_set_x(label, v.value)

@lv.event_cb_t
def sw_event_cb(e):
    sw = lv.event_get_target_obj(e)

    a = lv.anim_t()
    lv.anim_init(a)
    lv.anim_set_var(a, label)
    lv.anim_set_time(a, 500)

    if lv.obj_has_state(sw, lv.STATE_CHECKED):
        lv.anim_set_values(a, lv.obj_get_x(label), 100)
        lv.anim_set_path_cb(a, lv.anim_path_overshoot)
    else:
        lv.anim_set_values(a, lv.obj_get_x(label), - lv.obj_get_width(label))
        lv.anim_set_path_cb(a, lv.anim_path_ease_in)

    lv.anim_set_custom_exec_cb(a, anim_x)
    lv.anim_start(a)

label = lv.label_create(lv.scr_act())

lv.label_set_text(label, "Hello animations!")
lv.obj_set_pos(label, 100, 10)
lv.obj_set_x(label, 20)

sw = lv.switch_create(lv.scr_act())
lv.obj_center(sw)
lv.obj_add_state(sw, lv.STATE_CHECKED)
lv.obj_add_event(sw, sw_event_cb, lv.EVENT_VALUE_CHANGED, None)

start = time.time()

while True:
    stop = time.time()
    diff = int((stop * 1000) - (start * 1000))
    if diff >= 1:
        start = stop
        lv.tick_inc(lv.uint32_t(diff))
        lv.task_handler()
```

now while it may look like I am passing a callback to a function I am really not. Python has what are called decorators. decorators are able to be used before any callable object declaration. so a class, function, method, property, etc....

This is what a decorator would look like

```python

def some_decorator(func):
    
    def wrapper(*args, **kwargs):
        print('DECORATOR CALLED!!!')
        return func(*args, **kwargs)

    return wrapper
```

and that decorator would be used like so.

```python

@some_decorator
def some_function():
    print('FUNCTION CALLED')
```

and if I run this code

```python
some_function()
print(some_function)
```

the output is
```shell
DECORATOR CALLED!!!
FUNCTION CALLED
<function some_decorator.<locals>.wrapper at 0x0000028A186196C0>
```

but look at the location of `some_function` on the last line. 
```some_decorator.<locals>.wrapper```

and if I remove the decorator this is the output
```shell
FUNCTION CALLED
<function some_function at 0x000002BA96CB8A40>
```

the function is reported as `some_function` in the last line.

A decorator can modify input data to the function to be called, or it can modify output data of the function being called or it can place the function into a container/object to be used in some other fashion which is what is being done in out use case. Because we used the decorator before the callback function the function is no longer callable in python code and has been converted to run when it is called from C code. so it is a python representation of a function pointer and that is why we are able to pass it to a function call.

Now. I do have the ability to handle this specific problem and not need to use a decorator. I can create a weak reference of the passed in function and store the function inside the c object and attach the weak reference to the python function to the c object and store it someplace like that. so when all references to the python function no longer exist the weak reference will loose it's hold on the c object causing both to get garbage collected. that is the way it should be!

I am stoked that it works tho.

kdschlosser, 2023-07-04:
OK so I got rid of the decorators. so no need to use those anymore. 

Everything else looks like it is working well. I have to tinker about with arrays but those shouldn't be too difficult to iron out. I still have to get the type converters set in place for the structures. I am not 100% sure on how I am going to go about doing that because of how ctypes is made there is no real way to override the get/set functions for the structure fields. I might be able to access then after the class is built and override them that way. I have to dink about with it to see what I am able to do.

Good news is no more decorators needed for callbacks.

kdschlosser, 2023-07-05:
For the most part everything works. I am going to be testing some advanced code to see if it works properly. Basic use everything seems to be working as it should. I have one last animal to tackle and that's putting objects into structures. I have to figure out a way to do it so the user doesn't have to worry about pointers and creating a ctypes array.

It just dawned on me how to go about doing it. I can use the magic methods `__getattribute__` and `__setattribute__` to step in the middle and modify data types as needed.

kdschlosser, 2023-07-06:
@kisvegabor 

I have brought in the most recent changes to the CPython Binding. I am working on getting everything to work with as little changes as possible to LVGL. So far there are only 2 things that need to be changed and that is some structures that are defined in source files and these structures are used in public functions. I need them to be exposed in the header files so I can parse the files to create python representations of them.

I am going to attempt creating an empty structure for them to see if they will be able to be passed to functions without causing any issues. The information contained within the structures would not be accessible from the binding if it does work.

What I am having an issue with now is all of the lv_disp_* functions I am not able to export them. I don't understand why I am having the problem because the functions are not static. I am not sure what may have changed surrounding this functions or maybe the header file in general that would cause an issue like this.

Here is the error I am getting.

```shell
exports.def : error LNK2001: unresolved external symbol _lv_disp_get_refr_timer
exports.def : error LNK2001: unresolved external symbol _lv_font_clean_up_fmt_txt
exports.def : error LNK2001: unresolved external symbol lv_disp_add_event
exports.def : error LNK2001: unresolved external symbol lv_disp_create
exports.def : error LNK2001: unresolved external symbol lv_disp_enable_invalidation
exports.def : error LNK2001: unresolved external symbol lv_disp_flush_is_last
exports.def : error LNK2001: unresolved external symbol lv_disp_flush_ready
exports.def : error LNK2001: unresolved external symbol lv_disp_get_antialiasing
exports.def : error LNK2001: unresolved external symbol lv_disp_get_color_format
exports.def : error LNK2001: unresolved external symbol lv_disp_get_default
exports.def : error LNK2001: unresolved external symbol lv_disp_get_dpi
exports.def : error LNK2001: unresolved external symbol lv_disp_get_driver_data
exports.def : error LNK2001: unresolved external symbol lv_disp_get_event_count
exports.def : error LNK2001: unresolved external symbol lv_disp_get_event_dsc
exports.def : error LNK2001: unresolved external symbol lv_disp_get_hor_res
exports.def : error LNK2001: unresolved external symbol lv_disp_get_inactive_time
exports.def : error LNK2001: unresolved external symbol lv_disp_get_layer_bottom
exports.def : error LNK2001: unresolved external symbol lv_disp_get_layer_sys
exports.def : error LNK2001: unresolved external symbol lv_disp_get_layer_top
exports.def : error LNK2001: unresolved external symbol lv_disp_get_next
exports.def : error LNK2001: unresolved external symbol lv_disp_get_offset_x
exports.def : error LNK2001: unresolved external symbol lv_disp_get_offset_y
exports.def : error LNK2001: unresolved external symbol lv_disp_get_physical_hor_res
exports.def : error LNK2001: unresolved external symbol lv_disp_get_physical_ver_res
exports.def : error LNK2001: unresolved external symbol lv_disp_get_rotation
exports.def : error LNK2001: unresolved external symbol lv_disp_get_scr_act
exports.def : error LNK2001: unresolved external symbol lv_disp_get_scr_prev
exports.def : error LNK2001: unresolved external symbol lv_disp_get_theme
exports.def : error LNK2001: unresolved external symbol lv_disp_get_user_data
exports.def : error LNK2001: unresolved external symbol lv_disp_get_ver_res
exports.def : error LNK2001: unresolved external symbol lv_disp_is_double_buffered
exports.def : error LNK2001: unresolved external symbol lv_disp_is_invalidation_enabled
exports.def : error LNK2001: unresolved external symbol lv_disp_load_scr
exports.def : error LNK2001: unresolved external symbol lv_disp_remove
exports.def : error LNK2001: unresolved external symbol lv_disp_remove_event
exports.def : error LNK2001: unresolved external symbol lv_disp_send_event
exports.def : error LNK2001: unresolved external symbol lv_disp_set_antialiasing
exports.def : error LNK2001: unresolved external symbol lv_disp_set_color_format
exports.def : error LNK2001: unresolved external symbol lv_disp_set_default
exports.def : error LNK2001: unresolved external symbol lv_disp_set_dpi
exports.def : error LNK2001: unresolved external symbol lv_disp_set_draw_buffers
exports.def : error LNK2001: unresolved external symbol lv_disp_set_driver_data
exports.def : error LNK2001: unresolved external symbol lv_disp_set_flush_cb
exports.def : error LNK2001: unresolved external symbol lv_disp_set_offset
exports.def : error LNK2001: unresolved external symbol lv_disp_set_physical_res
exports.def : error LNK2001: unresolved external symbol lv_disp_set_res
exports.def : error LNK2001: unresolved external symbol lv_disp_set_rotation
exports.def : error LNK2001: unresolved external symbol lv_disp_set_theme
exports.def : error LNK2001: unresolved external symbol lv_disp_set_user_data
exports.def : error LNK2001: unresolved external symbol lv_disp_trig_activity
exports.def : error LNK2001: unresolved external symbol lv_scr_load_anim
```

This function declaration needs to be removed as it is not used and the function is not defined

_lv_font_clean_up_fmt_txt

kisvegabor, 2023-07-06:
Hi,

Sorry for my late reply. So happy to hear that it's working well know! :blush: 

Regarding the `lv_disp` error. So it seems `lv_disp.c` is not compiled and/or linked.  
- How do you get the list of source files? It's CMake, right? Do you use LVGL's CMakeList.txt file?
- What is you add this to `lv_disp.c`? `#error "!!! It was compiled !!!"`

kdschlosser, 2023-07-06:
I iterate over the entire directory tree for LVGL and pick up all of the files that end in ".c"

Your response allowed me to locate the problem so that is fixed. One last problem is that declared function that is not defined.
`_lv_font_clean_up_fmt_txt`

That is the last linking error that I have.

OH!!!

did you guys place lv_disp into it's own directory? Maybe that is why I am having a problem. That could introduce another issue for me that I have to mess about with.

kdschlosser, 2023-07-07:
YAY. It compiles under Windows. and it works too!! lol
I have to test on Linux and see how it goes. So the use of CFFI is gone in favor of using ctypes. That means that LVGL on Python has no runtime dependencies outside of what is in the standard library that is shipped with Python. It is also portable meaning it can be moved between machines of the same OS without needing to compile. Why is this a good thing? Well if someone wanted to use LVGL as the GUI framework for the application. the code can be written in Python, compiled using Cython and then packaged into an executable using cxfreeze or py2exe. Use innosetup to create the installer and away you go.

I think that it might be more version specific on Linux and OSX so not as portable.

kisvegabor, 2023-07-07:
> did you guys place lv_disp into it's own directory? Maybe that is why I am having a problem. That could introduce another issue for me that I have to mess about with.

This week I merged a huge update. See [here](https://github.com/lvgl/lvgl/issues/4011).
As part of it the directory structure changed too and `lv_conf.h` needs to be updates as well.

kdschlosser, 2023-07-07:
got it done. Yeah I know you made a massive update to LVGL. However what I didn't know what that the directory structure changed. when iterating over the source files I had that process ignore a directory named "disp" which there are now 2 named that.

kdschlosser, 2023-07-22:
not stale

kdschlosser, 2023-07-22:
ok so here is an update. with help from @kisvegabor we finally got the images to open up properly YAY!!. Got the file system driver working properly. Fonts are now working as well. 

I still have a little bit of tweaking to do with the callback functions. Currently ctype objects are being passed to the callback functions. The hard part here is if buffer objects that need to be filled are being passed. I am working on how to deal with that. I have to look more closely at the callbacks to see if buffers needing to be filled are all passed as a void pointer. If they are, this will make it a little bit easier to deal with as I can add a method to the pointer that will hide the ctypes code. 

It's really not all that bad. Here is the code for filling a void pointer buffer with byte data to a void pointer. structures are handled differently, a pointer to the structure would need to be placed into the void pointer.
```python
_ctypes.memmove(_ctypes.addressof(buf.contents), data, len(data))
```

I am going to have to spend some time thinking about how I want to handle this issue. I really don't want to use things like `.value` and `.contents` to access the underlying data in callbacks.

kdschlosser, 2023-07-22:
This is the CPython binding running a fairly complex demo.

[image]

kisvegabor, 2023-07-24:
Looks amazing! :heart_eyes:

kdschlosser, 2023-07-24:
it's getting there. working out all the little edge case oddities.  more complex things like how to handle arrays. Specifically from structures where there is no way to identify if a pointer is pointing to an array of variable length or if the field is a pointer to a single object. When there is a field that is an array of variable length there is also another field that specifies the length of that array but there is no consistent naming convention for that field to be able to marry it to the array field. If there was I could pair the 2 fields together. Here is an example.

```c
typedef struct {
    struct _lv_obj_t ** children;       /**< Store the pointer of the children in an array.*/
    uint32_t child_cnt;                 /**< Number of children*/
    lv_group_t * group_p;
    lv_event_list_t event_list;

    lv_point_t scroll;                  /**< The current X/Y scroll offset*/

    lv_coord_t ext_click_pad;           /**< Extra click padding in all direction*/
    lv_coord_t ext_draw_size;           /**< EXTend the size in every direction for drawing.*/

    lv_scrollbar_mode_t scrollbar_mode : 2; /**< How to display scrollbars*/
    lv_scroll_snap_t scroll_snap_x : 2;     /**< Where to align the snappable children horizontally*/
    lv_scroll_snap_t scroll_snap_y : 2;     /**< Where to align the snappable children vertically*/
    lv_dir_t scroll_dir : 4;                /**< The allowed scroll direction(s)*/
    uint8_t layer_type : 2;    /**< Cache the layer type here. Element of @lv_intermediate_layer_type_t */
} _lv_obj_spec_attr_t;

```

In the example above I am able to determine that `children` would be an array because it is a pointer to pointers. If that was consistently done that way. I would still need to marry `child_cnt` so I would know the length of the array. 

```c
typedef struct {
    lv_obj_t obj;
    const lv_point_t * point_array;     /**< Pointer to an array with the points of the line*/
    uint16_t point_num;                 /**< Number of points in 'point_array'*/
    uint8_t y_inv : 1;                  /**< 1: y == 0 will be on the bottom*/
} lv_line_t;
```

Now this one `point_array` is also an array but it is not an array of pointers. So I am not able to use the `**` seen in the first example as the way of knowing what is and is not an array. the field name that is used for the length of the array is not consistent. in the example above the name is `point_num`.

We still need to go over the code in LVGL and change all of the functions that accepts arrays as a parameter. A lot of them are typed as `{type} * param` and they need to be changed to `{type} param[]`. That simple change would not mess anything up in LVGL but it allows me to know if a parameter is an array.

I am stuck at this point as there is no way for me to deal with this issue other than hard coding those functions. It's not one or 2 functions either. It would take me a really long time to do because I would have to read over each and every single line of code in LVGL to know what is an array and what is not. The documentation is not always clear on what is an array and what isn't.

kdschlosser, 2023-07-24:
maybe the best way of handling this having a structure for an array type It would handle the issue for both functions as well as structures.

here is an example.

```c
typedef struct {
    uint8_t * array;
    uint32_t count;
} uint8_array_t;

typedef struct {
    lv_obj_t * array;
    uint16_t count;
} lv_obj_array_t;

typedef struct {
    lv_point_t * array;
    uint16_t count;
} lv_point_array_t;
```

This is easily identifiable as to what it is exactly there is a consistent naming convention

It can also be used when passing a buffer to a callback that needs to be filled.

Here is an example of such a use case
```c
lv_fs_res_t (*read_cb)(struct _lv_fs_drv_t * drv, void * file_p, void * buf, uint32_t btr, uint32_t * br);
```

This has to be made into a function pointer prototype still so it would actually be

```c
typedef lv_fs_res_t (*lv_fs_read_cb_t)(struct _lv_fs_drv_t * drv, void * file_p, void * buf, uint32_t btr, uint32_t * br);
```

```c
typedef struct {
    void * array;
    uint32_t * count
} lv_fs_read_array_t;

typedef lv_fs_res_t (*lv_fs_read_cb_t)(struct _lv_fs_drv_t * drv, void * file_p, lv_fs_read_array_t * data);
```

In this use case the `count` field is a dual purpose field. It is used to hold the number of bytes to read and upon successful completion it will hold the number of bytes that have been read.

I am not 100% sure why that specific callback function takes a void * parameter for the buffer. You would think that `uint8_t *` would be a more fitting type, that is another topic.

kdschlosser, 2023-07-25:
You can hide whatever you want in private headers But if the structure is used in a function argument then whatever header the structure is defined in is going to have to be imported for the binding. There has to be something that matches the type give for the function parameter.

If there is a function to handle the array that is in the structure as in placing and retrieving the array Then there is no need for me to do anything with it. But if the only way to access the items in the array is by dealing with it directly in the structure then something will need to be done so I am able to know that it is an array and not a pointer to a single object. The problem is not going from Python to ctype it going from a ctype to Python.

I am trying to keep the ctypes backend code hidden.

as an example.

in a function you have the ability to do this.

`uint8_t some_array[]`

but in a structure you have to specify the size of the array or declare the filed as `uint8_t * some_array`

`some_array` can be an array of uint8_t or it can be a pointer to a single uint8_t. unless there is some kind of a naming convention for the field name or there is a structure  that holds both the array and the count I have no way of knowing if it's a pointer to a single item or if it is an actual array with more than one item.

In Python an array gets represented as a list so when something gets passed to be set as a field I am able to create a proper c array. But when going the other way around it becomes harder to do.

I could add a method to all of the ctypes so a user would be able to convert a single python type into an array. The user is going to have to ultimately know to do that.

I think I will make a wrapper class around the ctypes classes that will handle this.

kdschlosser, 2023-07-26:
a function like `lv_line_get_points` would not be returning an array, a pointer to an array would get passed as a parameter and ideally the function would return the number of elements in the array. How I have seen this kind of a thing handled in the past is passing NULL for the array parameter and the function returns the number of elements in the array and then an array of proper size can be created and a second call to the same function passing the array this time would fill the array.

kdschlosser, 2023-07-26:
any buffer object that gets passed to LVGL. that is an array that gets passed.

kdschlosser, 2023-07-26:
any multi character text that gets passed is also an array.

kisvegabor, 2023-07-27:
> a function like lv_line_get_points would not be returning an array, a pointer to an array would get passed as a parameter and ideally the function would return the number of elements in the array. [...]

I don't get it. We have `lv_line_set_points(line_widget, point_array, point_cnt_in_array);`

`lv_line_get_points()` returns `point_array` and there can be a sparate `lv_line_get_point_cnt()` function too.

kdschlosser, 2023-07-27:
There is no way to know how long the array is that is returned from `lv_line_get_points` and without having that knowledge the API is not going to be able to handle converting the returned pointer to a Python array.

Unless the binding is hard coded there would be no way of making the ctypes binding layers transparent to the user.

kisvegabor, 2023-07-27:
How would this line points example work in Python? E.g. I want to get the points of a line and change 
`p[3].x=25;`

kdschlosser, 2023-07-27:
it would work exactly the same if there is a way to know how many elements are in the array. If the binding has no forward knowledge of that then the user would have to do this.

```python

array_len = lv.line_get_point_cnt(line_obj)
p_pointer = lv.line_get_points(line_obj)
p = ctypes.cast(ctypes.addressof(p_pointer), ctypes.POINTER(lv_point_t * array_length)).contents
p[3].x=25;
```

It can be handled this way if the function took some additional parameters. This is a little bit easier to deal with from a user standpoint.

```python
array_len = lv_line_get_point_cnt(line_obj)
p = (lv.point_t * array_len)()
lv.line_get_points(line_obj, array_len, p)

p[3].x=25;
```

and then we have the best way it can be done.
```
array_len = lv.uint16_t()
p_pointer = lv.line_get_points(line_obj, array_len.byref)
p = p_pointer * array_len
p[3].x=25;
```

in the last example byref is a pointer to uint16_t which gets filled with the length of the array. The returned type from `line_get_points` would be an `lv_point_t` structure that has pinning to the underlying array. from that point I can handle it one of 2 ways. I can extend the c objects so they can be indexed which would move to a specific element or we can use right hand operands to move to a different element in the array but to convert the array to a python list (array) a call like `p = p_pointer * array_len` would need to be done.

The downside of this approach is all returned pointers would have to function in this manner including pointers that are not arrays. so a user could cause a memory access violation if they attempt to access something that isn't an array as if it was an array.

kisvegabor, 2023-07-28:
What if we provide a function like `lv_line_set_point_x(line, 3, 25)`? We can do it general when the user needs to interact with arrays.

kdschlosser, 2023-07-28:
that can be added But I would add it so the function would only be available if a macro is set , something that would one be used for bindings. There would be no purpose to having the function for the general API.

Now.... There is going to be one issue. The use of functions like this would ideally be transparent to the user. So by naming convention there would need to be a way to marry it to a specific array type.  using the lv_point_t array as an example. name the function something along the lines of `lv_binding_point_array_get` and `lv_binding_point_array_set`. These functions would need to accept a pointer to an lv_point_t structure. The functions cannot error so checks need to be made to ensure the array is of proper size when getting or setting and if it is not then it would return a non zero error that can be handled in a more graceful manner by the binding program. One problem we still have is how to identify a pointer to a single lv_point_t and an array of them because they are going to be declared the same for a return value. The only way that can be properly achieved is if there is a core function and then separate functions that contain the name of the common API function it is going to be married to.

I don't think we need to be specific to the contents of the structures inside of the array. Only need to be able to get and set the index of the array. The rest will work properly.

This is what I am seeing.

```c
lv_res_t lv_binding_point_array_get(lv_obj_t * obj, uint16_t index, lv_point_t * ret);

lv_res_t lv_binding_point_array_set(lv_obj_t * obj, uint16_t index, lv_point_t point);

lv_res_t lv_binding_line_get_points_get(lv_obj_t * obj, uint16_t index, lv_point_t * ret) {
    return lv_binding_point_array_get(obj->point_array, index, ret);
}

lv_res_t lv_binding_line_get_points_set(lv_obj_t * obj, uint16_t index, lv_point_t point) {
    return lv_binding_point_array_set(obj->point_array, index, point);
}
lv_res_t lv_binding_line_get_points_set(lv_obj_t * obj, uint16_t index, lv_point_t point) {
lv_res_t lv_binding_line_get_points_get(lv_obj_t * obj, uint16_t index, lv_point_t * ret) {

lv_res_t lv_binding_triangle_get_points_get(lv_draw_triangle_dsc_t * dsc, uint16_t index, lv_point_t * ret) {
    return lv_binding_point_array_get(dsc->p, index, ret);
}

lv_res_t lv_binding_triangle_get_points_set(lv_draw_triangle_dsc_t * dsc, uint16_t index, lv_point_t point) {
    return lv_binding_point_array_set(dsc->p, index, point);
}
```

Because of the names having "binding" in them I know the functions that the above functions get married to are `lv_triangle_get_points` and `lv_line_get_points`. So when the binding code gets generated This is what it would look like.

```python

class _Array(object):
    
    def __init__(self, get_func, set_func, remove_func, c_struct, c_type):
        self._get_func_ = get_func
        self._set_func_ = set_func
        self._remove_func_ = remove_func
        self._c_struct_ = c_struct
        self._c_type_ = c_type
        self._index_ = 0
        self._c_obj_ = c_type()
        
        get_func(c_struct, 0, ctypes.byref(self._c_obj_))

    def append(self, obj):
        self._set_func_(self._c_struct_, len(self), obj)

    def extend(self, obj):
        index = len(self)
        
        for item in obj:
            self._set_func_(self._c_struct_, item, obj)
            index += 1
            
    def remove(self, obj):
        objs = list(self)
        objs.remove(obj)

        for i, item in enumerate(objs):
            self._set_func_(self._c_struct_, i, item)

        self._remove_func_(self._c_struct_)

    def pop(self, index=0):
        objs = list(self)
[... 134 lines cut ...]
    ('y', coord_t)
]

_lib_lvgl.lv_line_get_points.argtypes = __CArgList(
    obj=_ctypes.POINTER(obj_t)
)
_lib_lvgl.py_lv_line_set_points.restype = ctypes.POINTER(point_t)

def line_get_points(
    obj: obj_t,
) -> _Array[point_t]:

    obj = _convert_to_ctype(
        obj, 
        _lib_lvgl.py_lv_line_set_points.argtypes.obj
    )
    
    points_pointer = _lib_lvgl.py_lv_line_get_points(obj)

lv_binding_line_get_points_get,
lv_binding_line_get_points_set, 
lv_binding_line_get_points_remove, 
obj, 
point_t
```

kdschlosser, 2023-07-28:
with the above I am able to know what types are expected and leave the accessing of the actual array in C. That makes is a HUGE amount easier to deal with that's for sure. Having the macro makes it so it would not be apart of the normal API. This also keeps the API in tact through the binding so if a user wanted to port the code to C they wouldn't have to deal with API changes that would need to be made.

kisvegabor, 2023-07-31:
> that can be added But I would add it so the function would only be available if a macro is set , something that would one be used for bindings. There would be no purpose to having the function for the general API.

It can be useful in general too. We have only a few widgets dealing with array, so it's not really of issue of having these API function.

kisvegabor, 2023-07-31:
>  One problem we still have is how to identify a pointer to a single lv_point_t and an array of them because they are going to be declared the same for a return value. 

In function prototypes we can use `lv_point_t x[]` and `lv_point_t * x`.

> lv_binding_triangle_get_points_get

Triangle is a tricky one, because `draw_dsc`s  doesn't have API. I think we can figure out something custom for it instead of something generic.

kdschlosser, 2023-07-31:
I am still messing about with some ideas on how to handle it without changing the API at all. Since I changed over to using ctypes I am handling the callback functions being defined in structures and also as function parameters without them needing to explicitly declared as a type.

There is quite a bit more flexibility but I am going to have to leave some of the work for the user. Any way I go about trying to keep track of references ultimately ends up in a memory leak. No way around it from the python side of things.

kisvegabor, 2023-08-01:
>  Any way I go about trying to keep track of references ultimately ends up in a memory leak. 

So there is a memory leak now which can't be avoided?

kdschlosser, 2023-08-02:
Ya lost me on that question.

I thought that `lv.line_set_points` takes an array of `lv.point_t` structures

kisvegabor, 2023-08-02:
Ah really :smile: 

How to declare an array of points in CPython? :slightly_smiling_face: 

But regardless of it, this is what you meant by the out of scope memory leak/issue?

kdschlosser, 2023-08-03:
Yes because how it works is this

```python
def some_lvgl_function(python_array):
    c_array = (uint8_t * len)(python_array)(*python_array)
    lib_lvgl.some_lvgl_function(c_array)
```

OK so I have a wrapper function that would take care of converting the python list into a c array. that c array then gets passed as a pointer to the lvgl function. Now remember it's only a memory address that gets passed so as soon as the wrapper function exits the c array then get marked for garbage collection because it is out of scope.. But if I hold a reference to it I have no way of knowing when it is no longer in use in order to release that reference so guess what happens? Memory leak.

kdschlosser, 2023-08-03:
I still think I can solve this issue by making a container that is in the python structure to hold the c types that get set into the structure by the LVGL function. If the structure that gets passed to the function goes out of scope that would be the users doing because they no longer need to use it.

I do have a question about it tho. If I initialize say an lv_draw_line_dsc_t structure in a local namespace and I populate that structure with whatever it needs to be populated with if I call the draw function using that structure does LVGL use it right away or does the pointer to it get stored somewhere for use later when refreshing is done?

kisvegabor, 2023-08-03:
> I still think I can solve this issue by making a container that is in the python structure to hold the c types that get set into the structure by the LVGL function. If the structure that gets passed [...]

I think it makes sense to decouple the life cycle of the arrays and widgets. The widgets might be deleted but user still might want to keep the data in the array. 

> I do have a question about it tho. If I initialize say an lv_draw_line_dsc_t structure in a local namespace and I populate that structure with whatever it needs to be populated with if I call the [...]

`lv_draw_line_dsc_t` is saved by value and not by pointer.  The only tricky thing is is `label_dsc.text` where `label_dsc.text_local` tells if LVGL needs to malloc a buffer for the text or it will ive while the drawing is executed.

closed 2023-08-25 by lvgl-bot