lvgl/lvgl #4339: PNG (lodepng) forces you to supply an entire out buffer

codewitch-honey-crisis, 2023-07-04:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [ X] (n/a) Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) 

### Describe the bug

It's not really a bug per se, but a performance/resource usage issue, wherein you must - unless I dramatically misunderstand the API - supply an in memory output buffer to hold the decompressed PNG image. This is what I would consider to be undesirable on IoT devices. 

### To Reproduce
Use this code
[https://github.com/lvgl/lvgl/blob/master/examples/libs/png/lv_example_png_1.c](https://github.com/lvgl/lvgl/blob/master/examples/libs/png/lv_example_png_1.c)

### Expected behavior

Well, it would be nice if it would call you back with pixel information in order to draw, but I'm not sure how compatible that would be with the way LVGL works. Pngle would be a good option for this instead of lodepng

kisvegabor, 2023-07-05:
It would be really great, but our current PNG decoder ([lodepng](https://github.com/lvandeve/lodepng)) doesn't support it. :slightly_frowning_face:

codewitch-honey-crisis, 2023-07-05:
Pngle does, and would be easy to integrate if I could figure out a way to get LVGL to draw on demand without a backing bitmap. I'm a bit of a noob with LVGL.

kisvegabor, 2023-07-06:
I'm afraid that it will be very slow as Pngle [calls a callback](https://github.com/kikuchan/pngle#generic-c) on each pixel to draw. It should draw at least a line in a callback call.

codewitch-honey-crisis, 2023-07-06:
That could be easily done. I just don't know how to coerce LVGL to call me back to do a draw. When doing a canvas, it draws to a bitmap. In my own UI lib (which i was sort of forced to write for my graphics lib to support ESP32s) i get an on_paint() callback when a control needs to redraw. I would need something similar in LVGL.

kisvegabor, 2023-07-09:
In LVGL have [image decoder](https://docs.lvgl.io/8.3/overview/image.html#image-decoder)s. I'm planning to change the `read_line_cb` to `read_area_cb` in `v9` to make it more flexible, but it doesn't really change the concept.

The next section about [image caching](https://docs.lvgl.io/8.3/overview/image.html#image-caching) could be also interesting.

kisvegabor, 2023-07-13:
> The name "read" is weird to me. It implies that the caller can dictate what to draw, rather than the callback defining the location and size, for example. Maybe that's just me.

What about `get_area`?

codewitch-honey-crisis, 2023-07-13:
To my mind that seems better, but again, it could just be me. It's just read implies (to me) control over what it is being read. I don't know.

codewitch-honey-crisis, 2023-07-28:
I'm leaving this open but stale, because I'd like to revisit it once get_area is in place.

kisvegabor, 2023-07-28:
I've pinned this issue.

kisvegabor, 2023-10-27:
The get_area feature is added. You can see it action here: https://github.com/lvgl/lvgl/blob/ef1885f670e7162ce0c4c595bd35316c6a6e24da/src/libs/tjpgd/lv_tjpgd.c#L202

xlbao123, 2023-11-03:
From a performance perspective, should we consider using [Wuffs](https://github.com/google/wuffs-mirror-release-c)? Based on my testing on a mcu, lodepng appears to be the slowest. Here are some test data:
|     | decode time |
|  ----  | ----  |
| wuffs | 86ms |
| libspng+zlib | 113ms |
| libpng+zlib | 140ms |
| stb image | 151ms |
| lodepng | 208ms |

codewitch-honey-crisis, 2023-11-03:
That's a really interesting repo. I had to dig around at the link you provided to get to the meat before I even understood what i was looking at. My concern is flash size and segregatability (i know it's not a word) by which i mean Wuffs seems to be an all-in-one for a bunch of different formats, and while compelling can we just take what we need from it, or is it so tied together with cross dependencies that we're even duplicating work already in LVGL? it's a rather substantial codebase from the looks of it.

xlbao123, 2023-11-03:
> That's a really interesting repo. I had to dig around at the link you provided to get to the meat before I even understood what i was looking at. My concern is flash size and segregatability (i know [...]

It's a single-file C library that only needs to include 'wuffs-v0.X.c,' for example:
`#define WUFFS_IMPLEMENTATION
#define WUFFS_CONFIG__STATIC_FUNCTIONS
#include "wuffs-v0.4.c"`

It includes various functions, such as PNG, GIF, BMP, TGA, and more. It seems like we can use macros like 'WUFFS_CONFIG__MODULES_XXX' to customize. Since this code is auto-generated, it's not very readable, and I'm just starting  
 to study it. However, using it to decode image is both simple and efficient.

kisvegabor, 2023-11-03:
Thanks for sharing the banchmark! I'm not big fan of files with 64k lines either. And it seems it's no stable version yet as it's on v0.3.

closed 2023-11-26 by lvgl-bot