lvgl/lvgl #4342: [Parallel rendering] Optimization

kisvegabor, 2023-07-05:
Continue #4016

The parallel rendering update is merged to `master` but now its performance is worse than v8. 
We need to identify the bottlenecks and optimize them.

kisvegabor, 2023-07-05:
I've tried out the current master on an STM32F769 board and found these:
- I needed to [add a `volatile`](https://github.com/lvgl/lvgl/commit/388a5d7f975a91eb964e419e8d3ab81fd8a306c5) somewhere to make rendering work well with O3 optimization. I don't understand why.
- I tested with `lv_demo_benchmark(LV_DEMO_BENCHMARK_MODE_RENDER_AND_DRIVER);` and got very high FPS. Better than v8
- Using the widget's demo the perfomance felt good too (not worse than v8)
- In the benchmark of v8 there is no MODEs for `lv_demo_benchmark` so probably it's not a fair comparision
- We need the same benchmark for both v8 and v9 (I think v9's benchmark can be easily backported to v8, but I haven't tried it yet)

This the profiler log from v9. There is nothing extraordinary in it. 
[image]

When you test v9, be sure to disable OS (`LV_USE_OS  == LV_OS_NONE`) and enable O3 optimization for both v8 an v9.

nicusorcitu, 2023-07-06:
Just run the V9 release, CPU only, O3, on RT1170. The CPU is a single core 1GHz. GPU is disabled for now.
The results are attached without and with the FreeRTOS (e.g. with two threads lvgl_main and lvgl_draw SW).
[v9_cpu_os_freertos.txt](https://github.com/lvgl/lvgl/files/11969417/v9_cpu_os_freertos.txt)
[v9_cpu_os_none.txt](https://github.com/lvgl/lvgl/files/11969418/v9_cpu_os_none.txt)
I do not think the task switching takes so much, I will rework the task sync mechanism and see if any difference. But till then, 
I am just curious, on pthread (under linux) is the same big difference?

nicusorcitu, 2023-07-06:
How do you get the profiler logs?

kisvegabor, 2023-07-06:
On Linux there is some difference but not that much. Sometimes it was faster without OS and something it was faster with pthread.

This is my way of thinking:
- If we have an OS there are can be a lot of draw tasks and it takes more time to dispatch them. That's why we can see `lv_draw_dispatch_layer` with 13% in case of pthread. We need to optimize it.
- With OS there will be a new thread for drawing, so rendering can run in parallel with with the creation of draw tasks. That's why it's faster in some cases.

[linux_no_opt.txt](https://github.com/lvgl/lvgl/files/11969594/linux_no_opt.txt)

Linux, -O0, OS none
[image]

Linux -O0, pthread

[image]

> How do you get the profiler logs?

I have this button in Eclipse:  [image] I needed to set up `gprof` to make it work.

On STM it's called SWV statistic profiling
[image]

However the great news is that now LVGL has it's own built-in platform independent profiler: https://docs.lvgl.io/master/overview/profiler.html
We don't cover the all places with `LV_PROFILER_BEGIN/END` so feel free to add them anywhere it feels appropriate.

GorgonMeducer, 2023-07-06:
Just one quick question, where should we put the new GPU drivers?
Any instructions about which interfaces we should implement?

kisvegabor, 2023-07-06:
We can add the GPUs [here](https://github.com/lvgl/lvgl/tree/master/src/draw), next to the `sw` folder. 

This is the [dispatcher of the SW renderer](https://github.com/lvgl/lvgl/blob/508180505fce904820caf056b02822b246a9af36/src/draw/sw/lv_draw_sw.c#L104-L159). It finds an indepenent draw task, and if the GPU can draw it, it can mark it as `LV_DRAW_TASK_STATE_IN_PROGRESS`.

The interface itself  is quite abstract and the GPU can be implemented in a flexible way. The most important part of the interface in the SW renderer is [here](https://github.com/lvgl/lvgl/blob/508180505fce904820caf056b02822b246a9af36/src/draw/sw/lv_draw_sw.c#L302-L342). So it checks the type of the draw task and calls its own functions with the draw  descriptors.

GorgonMeducer, 2023-07-07:
What's the meaning of the return value for interface:
```c
int32_t (*dispatch)(struct _lv_draw_unit_t * draw_unit, struct _lv_layer_t * layer);
```

There is no comment. Based on the code segment here:

https://github.com/lvgl/lvgl/blob/508180505fce904820caf056b02822b246a9af36/src/draw/lv_draw.c#L226-L234

I guess:

- <0 means error?
- ==0 means not taken?
- >0 means taken?

kisvegabor, 2023-07-09:
I've added comment here: 9a97ffa24

> If my memory is correct, we planned to use this return value as a "score" for degree-of-matching or degree-of-performance etc.
Current dispatcher algorithm is just a skeleton to let everything work, am I right?

I left this complex things for the "optimization" milestone, so what we have now is a very basic dispatcher. 

However I'm not sure we need the degree concept in the end. The problem is that probably we need to map certain draw tasks to given draw units. Imagine that a GPU draws a gradient slightly differently than the SW render,  the gradient is redrawn a lot of times, and sometimes the GPU, sometimes the SW render draws it. In the end, it will result in slight flickering.  My idea is this:
- Create the draw units in a specific order. Dumbest first, smartest last
- Let the dumbest take the simple tasks and let the smart deal with the complex tasks
- Also make possible to not take, just mark certain tasks as "I will take it later". This way e.g. all gradients can be taken by a given draw unit to be sure they are always drawn the same way.
What do you think?

nicusorcitu, 2023-07-10:
> On Linux there is some difference but not that much. Sometimes it was faster without OS and something it was faster with pthread.
>
Have you tested with a single core CPU?

GorgonMeducer, 2023-07-10:
For me, the first key question is:

Who makes the decision? The dispatcher or the draw unit?

- If the dispatcher decides, then the next question is, "How to pass sufficient information from draw units to the dispatcher?"
- If the draw units decide, then the next question is, "how to arrange the order of the draw units so the relatively correct one can take a task?"

kisvegabor, 2023-07-10:
It's the draw unit who makes the decision

> If the draw units decide, then the next question is, "How to arrange the order of the draw units so the relatively correct one can take a task?"

Good question. For the built-in draw units (SW and supported GPU) we can add their init in `lv_init` in the correct order. For not support (user added) GPUs, they can init and create them manually after `lv_init.`

GorgonMeducer, 2023-07-10:
@kisvegabor Well, I think letting the dispatcher decide might be a better solution. But I reserve this to myself.

kisvegabor, 2023-07-10:
> But I reserve this to myself.

Please don't. :slightly_smiling_face: It wouldn't be the first time that my idea is not the best. :smile: 
So please, tell me why you think "letting the dispatcher decide" to which draw unit to shell execute a draw task is better?

My way of thinking is that the draw units already know their capabilities and can decide about themselves is they can handle a draw task or not. So we can interface to get the capabilities of the draw units.  

Besides that it's simple. If we want to introduce something more complex and more sophisticated we need to be sure that the additional complexity really pays off.

nicusorcitu, 2023-07-11:
Currently the dispatcher search for draw units (stack order) and if a draw unit support a task - it will take it.

IMO the task dependency can be taken in consideration to optimize things.

E.G. The first task in double buffer scenario is the FILL SCREEN with white color. As this task covers all the screen, any other task will depend of the completion of FILL SCREEN. If one unit takes the FILL SCREEN task, then the same unit can take also any other supporting tasks without waiting for FILL task completion.
Or, if we keep the same behavior as now, we shall split the FILL SCREEN into multiple tasks to be processed in parallel by multiple units - if my memory is correct, this was Gabor's idea when we discussed about tasks.

GorgonMeducer, 2023-07-11:
> So please, tell me why you think "letting the dispatcher decide" to which draw unit to shell execute a draw task is better?

a. draw unit only knows itself but lacks a global view, so in terms of the best decision-making, only the one who has the global view can make the right choice, i.e. the dispatcher
b. even if we let the dispatcher responsible for decision making, it has the freedom to:
    * arrange the order of the draw unit
    * visit the draw unit list in its desired way, and
    * let the first draw unit that responds with a non-zero positive value (i.e. OK) take the task truly. 
        * **This is, in fact the "the draw units decide" solution**

As a conclusion, by letting dispatcher to make the decision, we have the freedom to choose any policies which including the currently used one: "the draw unit decide"

I hope this helps.

kisvegabor, 2023-07-13:
@nicusorcitu 
The current `lv_draw_get_next_available_task` is really limited and slow. I waas already thinking about how to make it more flexible and faster. My idea is to build up an array in which we store the draw tasks in the order as they needs to be drawn (same as now). However we also assign a `level` to the draw tasks. All `level=1` tasks are independent and can be picked for drawing. Each `level=1` task is followed by `level = 2`s which depend on them, and so on. For example the levels could look like this:
```
1       2          3             4            3            2             2       3            2
screen  slider bg  slider indic  slider knob  slider text  screen title  button  button text  some label
```

When the screen is ready the level of all tasks after it decremented by one. So the level list looks like:
```
X 1 2 3 2 1 1 2 1
```

So we have 4 independent tasks. When the first `1` is ready we decrement the task only until we find a new `1` as these tasks belong to the same branch of the dependency tree.
```
X X 1 2 1 1 1 2 1
```

Adding the tasks with the correct initial level is a challenge but doable. We should check that on which tasks it depends from the back of the list and set the level value accordingly. 

This method has the advantages of:
- not a complex tree with many pointer which is hard to update
- easy to traverse
- fast to see the independent tasks

If provided the draw task list to the GPU with this `level` notion it would be simple to pick a task with `level=1` and direct dependency as well with `level=2` right after the first task.

What do you think?

---
@GorgonMeducer 
I see your point, but in the practice I can't imagine how the the dispatcher can pick the right draw unit effectively. Let's there is a simple small blue rectangle. How can is decide? 
1. Store a capability list in the draw units with speed score? It seems too complex as there are many combinations. 
2. Ask the draw units for a score on the draw tasks first? Actually it can work.
3. Anything else?

GorgonMeducer, 2023-07-14:
@kisvegabor I am about to take vacations, so here is my quick answer:

By letting the dispatcher make the decision, we reserve all the possibilities. The solutions/answers are always more than the questions.

kisvegabor, 2023-07-14:
Okay, let's continue it after your vacation. Have a great holiday!

kisvegabor, 2023-07-31:
Not stale.

nicusorcitu, 2023-08-24:
> The current lv_draw_get_next_available_task is really limited and slow. I waas already thinking about how to make it more flexible and faster.
>
Ok, so the screen size will be the first blocker one. We need to assure we can do it faster with the units available. If there is one unit, leave it so. If there are multiple units, split it to the number of the units. But this would be another story, something similar we might have with the screen rotation task.

nicusorcitu, 2023-08-25:
You are the really expert here, so I can only point my opinion.

> So the questions is how to force some draw task types will be always taken by given draw units. (it's required to be sure complex things like rounded rectangles are always drawn in the very same way)
>
Only that we don;t have a special task for rounded rectangle, circle, simple rectangle with solid color or gradient.
Instead there is a single task covering all of those operations: LV_DRAW_TASK_TYPE_FILL.

> This decision making of draw units needs to be customizable by the user, so it should be a callback, which contains a some default logic, but can be replaced by the user.
>
Yes, it shall be customizable by the user.
I believe we can have a default `preference-level` on the units for every task. If the user wants a specific unit to do certain things, it shall increase that `preference-level`. Once a task is being taken by the unit, we set on task type the unit unique ID so that all other similar tasks are processed by the same unit. I see this preference match of unit-task being more like a performance improvement - similar to the unit scoring for each supported task, that was suggested at the beginning of parallel rendering.

**Ideally it shall not be too much differences on draw units result. E.G. the GPU should draw the same gradient as the CPU. If there is a discrepancy I believe it shall be fixed (to one side or another).**

Next, we really need to test the performance while doing those extra checks. We don't want to increase the complexity of the dispatcher too much.

kisvegabor, 2023-08-25:
> Only that we don;t have a special task for rounded rectangle, circle, simple rectangle with solid color or gradient.
Instead there is a single task covering all of those operations: LV_DRAW_TASK_TYPE_FILL.

When the draw unit decides whether it wants to assign a task to itself it can look into the draw descriptors as well.

> Ideally it shall not be too much differences on draw units result. E.G. the GPU should draw the same gradient as the CPU. If there is a discrepancy I believe it shall be fixed (to one side or another).

The keyword is the "ideally". :slightly_smiling_face: I can't imagine VG-Lite, and ST's NeoChrome will rotate the image pixel perfectly the same way. Besides in SW image rotation I a lot of trade offs to make it fast and not look too bad. If we made it's result the same as GPU, SW render would be much slower. It's the same for arc or line drawing as well.

> Next, we really need to test the performance while doing those extra checks. We don't want to increase the complexity of the dispatcher too much.

We can highly optimize it, as only the new tasks needs to be checked. So each tasks needs to be examined only once, which should not be too slow.  I think it should be a different callback and not part of the dispatching.  This way it would be easier to replace this part by user as well.

GorgonMeducer, 2023-08-30:
> Let's there is a simple small blue rectangle. How can is decide?

> - Store a capability list in the draw units with speed score? It seems too complex as there are many combinations.
> - Ask the draw units for a score on the draw tasks first? Actually it can work.
> - Anything else?

I prefer to divide this question into two levels:

- Multi-work-capability evaluation
     for a drawing task (combining a set of sub tasks, SubTask0, SubTask1, SubTask2...), PE can return a capability description about the number of immediate-continued sub-tasks that the PE can handle. For example:
    > If a PE can handle SubTask0, SubTask1, SubTask2, SubTask4 but not SubTask3, it should return 3 (0~2) as its multi-work capability

- Performance evaluation
     Each PE should return a score for the performance evaluation. For example, If a PE can handle SubTask0, SubTask1, SubTask2, SubTask4 but not SubTask3, then it should return a score as the sum of SubTask0, SubTask1 and SubTask2

The dispatcher should give the Multi-work-capability score a high weight, for example times 0x10000 and give the performance evaluation score a low weight, for example (store it in the BIT0~BIT15).

In the early stage, the dispatcher can ignore the performance evaluation part and only use the multi-work-capability evaluation score to choose PE. In later stage, after a collecting sufficient feedback from community, we can complete the performance evaluation system and use its result as a way of fine-tune. 

I hope this helps.

kisvegabor, 2023-08-30:
> Multi-work-capability evaluation

What if there are 10 buttons with a border and a label on them. So we have these tasks:
```
1. rectangle
2. border
3. label

4. rectangle
5. border
6. label

7. rectangle
8. border
9. label
...
``` 

Let's say the GPU can draw on rectangles. But it can draw batch of them.  So it can take 1, 4, 7, .... What I mean by that is that it's not necessary that subsequent draw tasks will be drawn together. Therefore I think the PE should be able to pick any draw tasks it wants.

> Performance evaluation

I think don't have as much room for optimization here as we assumed at the beginning. It's because except a simple rectangle, we need to decide which PE will handle a given task type. In v8 I had many issues because ST's DMA2D calculates opacity a little bit differently. I guess some rounding issues and and off by one issue is very visible on RGB565.  The problem was that sometimes a larger semi transparent area was blended by SW and sometimes by DMA2D and it was visible. 

So we need to decide something like:
- simple solid rectangle: any draw unit
- rectangle with opa only: Arm-2D
- rectangle with any opa + radius or  gradient:  VG-Lite

else we will see flickering on the screen if a task is taken by different GPUs on each frame.

For [this ](https://github.com/lvgl/lvgl/issues/4342#issuecomment-1691636624) I had this idea: 
- when a draw task is added we ask all draw unit what they think about it and ask them to assign a preference level (i.e. score). 
- if a draw unit wants a task more (its preference level is greater then the currently set value) save it's ID in the draw task
- draw units can take tasks only with their ID or a special "ANY ID" value

In summary this solves the "Performance evaluation"  problem, but as "Multi-work-capability evaluation" can be very complicated and seemingly unrelated tasks can be drawn together too (button background + a scrollbar of an other rectangle + border of an other rectangle) I think we should leave it to the GPUs what draw tasks they want to pick.

GorgonMeducer, 2023-08-30:
> So it can take 1, 4, 7, .... What I mean by that is that it's not necessary that subsequent draw tasks will be drawn together. 

For this case, how dispatcher know the PE takes which tasks? 

So you know, how multi-work-capability is evaluated is based on how the dispatcher providing the tasks. If the dispatcher provides the tasks in the way you described, and PE think it is capable to do multiple work, it will return the number. 

In conclusion, PE returns a score describing how many sub-tasks it can handle, this doesn't mean the way PE decide the capability only in one way but not others.

kisvegabor, 2023-08-30:
I think it should be decided per draw task, not in batch. 
We can figure out anything here E.g. the preference level for SW rendering is 100. Let's say 70 means 30% faster than SW, 130 means 30% slower than SW.

I understand that drawing 2 rectangle at once is faster then drawing then one by one which is not considered by preference level. 

However, as  I described above we have no room for such optimization. E.g. it's not possible to draw a semi transparent rectangle  with Arm-2D but draw 2 semi transparent rectangles with VG-Lite as blending might look slight different. So we need to assign all semi transparent rectangles to either Arm-2D or VG-Lite (to the faster one).
 
I also admit that there are tricky cases: I found that e.g. ST's DMA2D is much slower than SW render. Let's say it's preference level is 130 (30% slower than SW), so we should always choose SW. But, if DMA2D can work on a rectangle while SW is drawing a label in the end we might be faster. I have no idea how to decide when we shall use a slower than SW draw_unit. Especially if we consider that all draw tasks with the same type should be handled by that slower draw_unit.

GorgonMeducer, 2023-08-30:
@kisvegabor  
Thank you for your explanation. 
I reserve my idea to myself, and I agree with you now. I have no further question.

kisvegabor, 2023-08-30:
I added `preferred_draw_unit_id` and `preference_score` to `draw_task`s. `evaluate_cb` is also added to `draw_unit`s, which is called for all `draw_task`s when they are created to set the score and the ID.

See da876e4a2

nicusorcitu, 2023-08-31:
I think we do not have to split a task into subtasks, we can have directly the subtasks (see the rectangle have separate task for border, bg, outline) so the units can take them or group them.

About grouping multiple tasks in VGLite GPU I did in this way:
1. In GPU dispatcher (SW thread) I check if the task is supported. If yes, then I mark the state as in progress and I send it to the GPU thread.
2. In GPU thread, I put that task in a queue and I start the execution (preparing the GPU command_buffer1)
3. Before submitting the command buffer to the GPU, I am getting the GPU status.
    3.1. If GPU is BUSY then I let the task queued (in command_buffer_1) and return to dispatcher to get one more task.
    3.2. If GPU is IDLE, then submit the command_buffer1 and mark all previously flushed tasks as complete.
4. Return to dispatcher to get more tasks - this time it is used the command_buffer2 for queuing.
5. Goto 1.

kisvegabor, 2023-08-31:
>  If GPU is BUSY then I let the task queued (in command_buffer_1) and return to dispatcher to get one more task.

Why do you need to return to the dispatcher to get one more task?  Can't you just call `lv_draw_get_next_available_task` again to get a new available task?

nicusorcitu, 2023-09-05:
> I added preferred_draw_unit_id and preference_score to draw_tasks. evaluate_cb is also added to draw_units, which is called for all draw_tasks when they are created to set the score and the ID.
> 
> See https://github.com/lvgl/lvgl/commit/da876e4a213df345d55fb7fe4bc114211ee676d3
>
I see few issues here:
1. It is difficult to set a simple score for a task.
For example: `LV_DRAW_TASK_TYPE_IMAGE` is used for:
- blit all color formats w/ or w/o opa
- rotate
- recolor
- zoom
Each of above operations have different performance depending also of the color format and the size of the image.
2. Let's say we agree that `LV_DRAW_TASK_TYPE_IMAGE` blit RGB565 with GPU have a better score. So I want that all blits to be done by GPU, so in `evaluate_cb` I set the GPU Unit ID to `LV_DRAW_TASK_TYPE_IMAGE `. How the other operations (like rotation, recolor) will be differentiated to be executed by other units? Recolor might not even be supported by the GPU.

kisvegabor, 2023-09-05:
> It is difficult to set a simple score for a task.

The score doesn't have to be set by `LV_DRAW_TASK_TYPE_*` only but you can look into the `draw_task->draw_dsc` as well. You need to cover only the major cases: opacity and/or transformation and/or recolor

Besides to score doesn't have to be very precise. We will never compare the score of VG-Lite with ST's ChromaArt as these two GPU's will never exist in the same system. So the scores only purpose  is to decide whether a draw task should be handled by SW, VG-Lite, or PXP. You can say:
- 100: means worst case, only if there were no better candidate
- 60: I'm somewhat better than SW
- 30: I really want to take that draw task but live some room to be taken by an customer added draw_unit
- 0: Always give it to me 

Maybe we can forget the idea of performance score and do only the above mentioned pure preference value. What do you think?

> Let's say we agree that LV_DRAW_TASK_TYPE_IMAGE blit RGB565 with GPU have a better score. So I want that all blits to be done by GPU, so in evaluate_cb I set the GPU Unit ID to LV_DRAW_TASK_TYPE_IMAGE 

You set the  GPU Unit ID on a given draw_task and your preference score might be affected by a lot of factors besides the `LV_DRAW_TASK_TYPE_`. For example:
- the other parameters in `draw_task->draw_dsc`
- the size of the image
- the format (raw, png, jpg, etc) of the particular image source
- any run time config to alter the behavior of the GPU 
- your current mood. Just joking, but you can really consider anything about the current draw task including any global states as well.

nicusorcitu, 2023-09-07:
Sorry, I still do not get it. Here you check if the task has a preferred unit.
https://github.com/lvgl/lvgl/blob/4068c079a241a06bd934e9acacf314d4516fa1fa/src/draw/lv_draw.c#L287
For example, in case of `LV_DRAW_TASK_TYPE_IMAGE` how do you distinguish the recolor by transformation? The above code will run for both and will wrongly think the preferred unit is the GPU when in fact the GPU does not even support recolor.

kisvegabor, 2023-09-08:
I imagined it like this:
```c
void my_evaluate_cb(lv_draw_unit_t * u, lv_draw_task_t * t)
{
    if(t->type == LV_DRAW_TASK_TYPE_IMAGE) {
        lv_draw_img_dsc_t * dsc = t->draw_dsc;

        /*It can't handle recolor and non-normal blend modes so don't take this task*/
        if(dsc->recolor_opa > 0 || dsc->blend_mode != LV_BLEND_MODE_NORMAL) return;

        /*It's very good at image transformation*/
        if(dsc->zoom != LV_ZOOM_NONE || dsc->angle != 0) {
            if(t->preference_score > 30) {
                t->preference_score = 30;
                t->preferred_draw_unit_id = MY_ID;
            }
            return;
        }

        /*Image with opacity or ARGB is images are much faster than SW*/
        if(dsc->opa != LV_OPA_COVER || lv_color_format_has_alpha(dsc->header.cf)) {
            if(t->preference_score > 50) {
                t->preference_score = 50;
                t->preferred_draw_unit_id = MY_ID;
            }
            return;
        }

        /*In case of simple RGB image copy we are not faster than SW
         *but at least it runs in the background */
        if(t->preference_score > 90) {
            t->preference_score = 90;
            t->preferred_draw_unit_id = MY_ID;
        }
    }
}

```

nicusorcitu, 2023-09-09:
I  understood that, but how do you make the distinguish in "lv_draw_get_next_available_task"?

Will evaluate_cb be done for each an every task? I thought that once a task type is set as prefere by a unit, it will never be evaluted again. This is why i made the connection between a task type and score and not between every single task and a score.
________________________________
De la: Gabor Kiss-Vamosi ***@***.***>
Trimis: Friday, September 8, 2023 9:55:04 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] [Parallel rendering] Optimization (Issue #4342)

Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

I imagined it like this:

void my_evaluate_cb(lv_draw_unit_t * u, lv_draw_task_t * t)
{
    if(t->type == LV_DRAW_TASK_TYPE_IMAGE) {
        lv_draw_img_dsc_t * dsc = t->draw_dsc;

        /*It can't handle recolor and non-normal blend modes so don't take this task*/
        if(dsc->recolor_opa > 0 || dsc->blend_mode != LV_BLEND_MODE_NORMAL) return;

        /*It's very good at image transformation*/
        if(dsc->zoom != LV_ZOOM_NONE || dsc->angle != 0) {
            if(t->preference_score > 30) {
                t->preference_score = 30;
                t->preferred_draw_unit_id = MY_ID;
            }
            return;
        }

        /*Image with opacity or ARGB is images are much faster than SW*/
        if(dsc->opa != LV_OPA_COVER || lv_color_format_has_alpha(dsc->header.cf)) {
            if(t->preference_score > 50) {
                t->preference_score = 50;
                t->preferred_draw_unit_id = MY_ID;
            }
            return;
        }

        /*In case of simple RGB image copy we are not faster than SW
         *but at least it runs in the background */
        if(t->preference_score > 90) {
            t->preference_score = 90;
            t->preferred_draw_unit_id = MY_ID;
        }
    }
}

kisvegabor, 2023-09-11:
See here:
https://github.com/lvgl/lvgl/blob/4415f4a766440f1d75f39e63d746a5cdd78d60af/src/draw/lv_draw.c#L97-L105

`draw_unit-yevalute_cb` is called for each created `draw_task`s. (`lv_draw_finalize_task_creation` is called at the end of `lv_draw_rect/label/img/etc()`.)

So you can store a unique score for each single `draw_task`.

nicusorcitu, 2023-09-11:
Ok, it make sense now!

kisvegabor, 2023-09-27:
Not stale

kisvegabor, 2023-10-13:
We will know more about what to optimize when we will have somethings from #4624.

kisvegabor, 2023-10-29:
Not stale

kisvegabor, 2023-11-13:
Not stale.

kisvegabor, 2023-11-28:
We have dedicated issues for optimizations so I close this one.