lvgl/lvgl #4344: consult about the list obj

htk719809837, 2023-07-06:
Hello, lvgl7.11 is there a way to make the list object column can display any number of images and text method at the same time？

htk719809837, 2023-07-06:
For example, in the list of wifi, the front of the list shows whether the connection pattern, the middle shows the name of wifi, and the back shows the icon of signal strength and whether there is a password. lv_list_add_btn(list, &icon_img, "Text") This interface can only add one text and one icon

kisvegabor, 2023-07-06:
Hi,

We use GitHub Issues for development-related discussions.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

closed 2023-07-06 by kisvegabor