lvgl/lvgl #4347: Reconsider the Micropython an other binding examples

kisvegabor, 2023-07-06:
Now the Micropython examples are one-to-one copy of the C examples. As a few years have been passed now I see a few issues with this approach:
- If anyone changes or adds a new C example they need to add and test the related Micropython and all the other binding  example too. It requires a lot of background knowledge from all the bindings and their environment setup. 
- Now we have 100+ examples. If someone wants to add a new binding  with all the examples it's an unreasonably huge work
- What is possible and straight forward in C is not necessary possible and typical in a binding. Therefore if the C example is the base the binding examples might feel unnatural. Or if we follow the C way of thinking there might be binding features that might not be showcased.

My suggestions is to decouple the binding examples form the C examples. For example we can have 5 C examples for a `lv_btn` to show all the things what LVGL can do with button. Next to that we can have 3 MicroPyhon examples show casing e.g. a simple button; a styled button; and some ther Pythonic magic which has no C counterpart. And we can also have only 1 JavaScript example.

What do you think?

cc @amirgon @uraich

pete-pjb, 2023-07-06:
I agree with this way of thinking @kisvegabor .

kdschlosser, 2023-07-07:
If the API of the bindings followed as close as possible to the C API this would actually be not so much of an issue. Provided that the examples are not overly complex, Writing a script that would convert the examples to their specific language would not be a big thing to do.

I personally feel that LVGL should only have examples in C and the MicroPython binding should have examples with it and so on and so forth. The mixing of LVGL C and LVGL MicroPython in documentation and also in code should not be done. It makes things more difficult to maintain because You are always having to think about a binding and what it is going to effect. Make whatever changes to LVGL needs to be made without any thought for any bindings this includes examples and let the maintainers of the binding change what needs to be changed in the binding to make it work or submit a PR to LVGL to make a change to fix an issue in a binding so long as the change isn't specific to a single binding. What I mean by that is if a change is made and that change breaks 3 bindings but fixes one and the only purpose to the change was to fix the one, probably not a good change to make.

kdschlosser, 2023-07-07:
same thing goes for tests. and compilation. no binding should get compiled at the same time as LVGL. the test cases should not be mixed together.

I am a fan of visual help when trying to explain things.

[image]

See how LVGL is not connected to the tests or examples of any of the bindings. I feel that is the best way to go about it. It keeps code specific to the binding with the binding.

kdschlosser, 2023-07-07:
The only information about a binding in the LVGL documentation should be a link to the bindings documentation. There should not be any information on how to use it, or how to compile it. Only a link..

LVGL will get unmanageable if you try to lump everything together.

kisvegabor, 2023-07-07:
Besides the MP binding we will have a Python3 binding, a C++ binding, and I hope the JS binding will be activated too. We also have a Rust binding which is quite active.

So in summary, I agree to have the examples in the binding itself and let the maintainers of the binding write examples freely.

kdschlosser, 2023-07-07:
It really makes more sense to do it that way. I am going to work on a way to extract the documentation from LVGL that can be used for the bindings to build their API documentation easier. The information will be able to be pulled into sphinx this way the general format can be kept across the board for them. It would just be a matter of passing the LVGL object name into it and it will return the information that is contained as JSON data that would be simple to use. doxygen xml is not the easiest thing to read so there is really no sense in having a different way of doing it for each binding. The build system will function much in the same way except the reST files will have some changes to them so they would be specific to the binding. A lot of the information actually remains the same, just the roles need to be swapped out.

kdschlosser, 2023-07-07:
@uraich 

I know they are intertwined. To what extent I do not know. They really should be separate things. It has got to make development more difficult having to deal with when wanting to make changes to LVGL. Any change can potentially break LVGL all together when that change might work for LVGL but would require changes made to the MicroPython binding in order to get it to work. which is a separate thing. LVGL is not dependent on MicroPython there for it should not be apart of the LVGL build system or even directly apart of LVGL development.

Think about it this way. If for whatever reason the developer of MicroPython stops maintaining it. what then? That could have a very negative impact on developing LVGL when it shouldn't. LVGL's development should not be directly effected by a binding no longer being maintained, or some portion of the binding no longer being maintained because LVGL should not be dependent on any portion of a binding.

MicroPython is a very involved system. A person has to spend a HUGE amount of time to learn it. If something happens to the maintainer of it locating someone that is able to and also willing to step in to update it is not going to be an easy thing to locate.

I am just voicing my opinion and as everyone knows. "Opinions are like #$& holes, every one has one and everyone things everyone else's stinks.".

I am along for the ride so whatever decision is made is not  mine to make. Make a list of pros and cons about it and apply a weight to each thing and add those up and see if it has more pros than cons. 

Typically when there is porting or a binding you typically see the binding get updated after the library. In some cases the authors of a library will wait to make a release so porting or libraries can get updated so all releases can be done at the same time.

amirgon, 2023-07-07:
Sorry for the long post, but I really have a lot to say about this:

> My suggestions is to decouple the binding examples form the C examples.

I feel, same as @uraich, that the coupling of the Micropython examples and C examples is very important for multiple reasons.  

The Micropython examples are useful both as a reference and as automated tests. Today all these examples are ran automatically on every LVGL commit so we can immediately catch issues either with the Micropython examples themselves (when LVGL API changed and the example was not fixed), with LVGL API or with the binding logic.

Since the Micropython binding is automatically generated from the C API, it makes more sense that the Micropython examples are part of LVGL repo and not of the binding repo. The binding repo is not limited to a single LVGL API. LVGL version is determined according to where the LVGL git submodule points. So when LVGL API changes we would also want the examples (C, Micropython etc.) be aligned with these API changes. If the examples were part of the bindings repo, that would lock the binding repo with a specific LVGL version.

Another thing to lose when decoupling the binding examples from the C examples is LVGL API testing for coding conventions.  
Very early in the development of the Micropython bindings we found out that LVGL API must follow a [set of conventions](https://docs.lvgl.io/master/get-started/bindings/micropython.html#lvgl-c-api-coding-conventions) for the binding to work. That covers callback conventions but not only.  
We also found out that when naively writing/changing the LVGL API, it's very easy to make mistakes and not fully follow these conventions.  
When the Micropython examples are automatically tested on every LVGL commit, that also verifies that LVGL API follows these conventions and that there are no new issues with the bindings that we are unaware of.  
If the examples are decoupled from LVGL, we could no longer run them as automated tests and very soon LVGL API will deviate from these conventions and break the Micropython bindings.

> * If anyone changes or adds a new C example they need to add and test the related Micropython and all the other binding  example too. It requires a lot of background knowledge from all the bindings and their environment setup.

Right, and I actually see this as an advantage.  
An example is useless if it's broken, it cannot be used as a reference and cannot be used for automated testing.  
So if someone changes or adds a new C example and cannot or don't want to change the Micropython example, that someone is in fact breaking the Micropython example. That Micropython example can no longer be used as a reference or for testing. What you are suggesting is just to ignore this situation.  
We can think of ways to make it easier for people to change and test Micropython examples, but if someone really doesn't want to touch them, it's better to disable the example or comment it out rather than keep a broken example.  
I would also expect that someone to at least open a new GitHub issue saying that he disabled that example, so that example could be fixed in the future.

But if what you are suggesting is to just ignore this situation, let people break examples without anyone knowing about this, very soon we'll have lots of irrelevant examples that are wrong as reference and useless as tests.

> * Now we have 100+ examples. If someone wants to add a new binding  with all the examples it's an unreasonably huge work

No one says that every C example has to have a corresponding non-C example.  
@uraich did a tremendous work of translating the C examples to Micropython.  
If it becomes too hard to translate C examples to other languages, it's ok to translate only some of the examples. Or even create a widget with only C examples, In which case that widget will probably only be used with C because it will most probably not follow the conventions required for the Micropython bindings.

> * What is possible and straight forward in C is not necessary possible and typical in a binding. Therefore if the C example is the base, the binding examples might feel unnatural. Or if we follow the C way of thinking there might be binding features that might not be showcased.

In 90% of the cases the translation from C to Micropython is straightforward, and that's because the binding is automatically generated from the C API. There is a set of rules that map from C to Micropython and we theoretically can think of automatic scripts for this translation (although this is probably a hard task and involves a lot of work to do it right).

In the past the translation work itself revealed issues with the binding and drove fixes and new features to the core Micropython bindings.  
I believe the Micropython examples, in most cases, can follow closely the C examples. In those rare cases they don't - it would be interesting enough to let us know (me,Uli) so we could look into it.

> My suggestions is to decouple the binding examples form the C examples.

I strongly vote against decoupling, for any automatically generated bindings.

> For example we can have 5 C examples for a `lv_btn` to show all the things LVGL can do with button. Next to that we can have 3 MicroPyhon examples show casing e.g. a simple button; a styled button; 

[... cut ...]
- When LVGL didn't follow the coding conventions - we found about it too late.
- It did not allow using the same lv_binding_micropython with different versions of LVGL

Taking all the Micropython examples out from the LVGL repo is therefore, in my opinion, a big step backwards.

> As far as I know, the C binding and the MicroPython binding are strongly linked. The MicroPython binding is automatically generated from the C binding and it is therefore not easy to decouple the two.

As mentioned above - I totally agree.

kdschlosser, 2023-07-08:
> Since the Micropython binding is automatically generated from the C API, it makes more sense that the Micropython examples are part of LVGL repo and not of the binding repo. The binding repo is not [...]

The CPython binding is also going to get automatically generated, should the examples for the CPython Binding also be included into LVGL along with the the documentation on how to use it? 

>So when LVGL API changes we would also want the examples (C, Micropython etc.) be aligned with these API changes. If the examples were part of the bindings repo, that would lock the binding repo with a specific LVGL version.

That doesn't make any sense. the Binding would have releases just like LVGL does. The releases would contain any updates that are needed to allow it work with a specific LVGL version. 

> In 90% of the cases the translation from C to Micropython is straightforward, and that's because the binding is automatically generated from the C API. There is a set of rules that map from C to [...]

This is what SquarlineStudio attempts to do and doesn't get it right and the code often will not run, That is because it is not as straight forward as you believe it to be. I have said it and so have other people is there is a lot a confusion as to what the exact mapping from CAPI to MicroPython API actually is. I have asked many times about it only to be left more confused as to how the mapping is and how it is being done.

> Another thing to lose when decoupling the binding examples from the C examples is LVGL API testing for coding conventions.
Very early in the development of the Micropython bindings we found out that LVGL API must follow a [set of conventions](https://docs.lvgl.io/master/get-started/bindings/micropython.html#lvgl-c-api-coding-conventions) for the binding to work. That covers callback conventions but not only.
We also found out that when naively writing/changing the LVGL API, it's very easy to make mistakes and not fully follow these conventions.
When the Micropython examples are automatically tested on every LVGL commit, that also verifies that LVGL API follows these conventions and that there are no new issues with the bindings that we are unaware of.
If the examples are decoupled from LVGL, we could no longer run them as automated tests and very soon LVGL API will deviate from these conventions and break the Micropython bindings.

I am confused now. You had told me in the past that the script that generates the NicroPython code is not specific to LVGL. You have told me that it was written so it could handle writing the code to allow modules other than LVGL to work in Micropython. That generator script also takes care of writing the binding code for parts of the ESPIDF. So what exactly is it that would make it specific to LVGL in such a way that would make it a necessity to have the 2 be coupled together like they are?

If LVGL has changes made and those changes cause an issue in the MicroPython binding 2 choices get made. The same 2 choices Every other binding has to make. Either code around the problem at the binding level (if it can be done) or submit a PR to LVGL to make a change at the LVGL level that corrects the problem. see #4348. This is a fantastic example of how to handle a problem. I am not able to code around those problems in the CPython Binding so I made some changes at the LVGL level and submitted a PR. If those changes get brought into LVGL and they end up causing a widespread issue they can be reverted. and if not they can be kept.

kdschlosser, 2023-07-08:
Right now because the 2 are coupled together and there is a change that gets made that would fix an issue with say the Rust binding and also with the CPython binding and the change would not effect how LVGL runs the build will fail if the change causes a minor hiccup in the MicroPython binding. There is a requirement for any changes being made that they MUST allow the MicroPython binding to function. In reality it is the MicroPython binding that needs to have changes made to it in order to allow it to work. There is a forcing of developers to have to learn how to fix problems in a binding they have no clue about. They do not want to spend the time to have to learn it and there for they end up not fixing their binding issue because of the requirement of keeping someone else's project maintained for them.

The MicroPython binding is dependent on LVGL not the other way around. Right now it is almost like the entire system revolves around the MicroPython binding. There seems to be some confusion about this because it is not supposed to be the responsibility of anyone other than the author of a binding to keep that binding operational. I believe this to be a reason why there is not as many people contributing as there should be. Contributing to LVGL is one thing but then also needing to make changes to keep the MicroPython binding running as well... I can see this not wanting to be done due to the complexity of it and if there is a change that specifically needs to be made to the MicroPython Binding there is no gurantee that the change will get accepted. The MicroPython binding has felt the ill effects of this with the requirement of having to run a special modified version of MicroPython because of PR's that are made upstream not getting accepted. This is the same thing. There should be no requirement of changes made in LVGL needing to be made in such a way to keep a binding running properly. That hinders the development of LVGL. If a change gets made that breaks a binding then either fix the binding or submit a change to LVGL that corrects the problem afterwards. Bindings do not need to be updated at the same pace as LVGL. When LVGL is about to have a release the binding authors are informed that there is going to be a release made and time is given so any changes that need to be made to the binding can be made and during that same time PR's are done to LVGL to make changes that help to fix issues the bindings are having. 

CI's are what test LVGL to make sure the code runs properly. examples are not supposed to be unittests or test cases as they have no way of validating if something is actually functioning properly. This is the CPython binding, there are no errors at all and the code seemingly executes properly. But there is an obvious issue.

[image]

If I had test cases where I could check to make sure the image data is loaded properly then it would fail. Using an example only ensures the code runs without error It does not ensure it actually works properly. Using MicroPython or any other language to test C code is not the best way to test. C code needs to be tested using C code that is written to specifically validate the data that is supposed to be there. If I used Python code to test the validity of the data for LVGL I wouldn't know if the problem existed on the Python side of things or on the C code side of things. This is a problem on the Python side not on the C side and the only way I would know that is if the C Code has been tested using C code and it tests out as working properly.

uraich, 2023-07-10:
The tests done on the MP examples consist simply of running the program and checking that is does not crash. Therefore the style in which the program is written does not really influence the test. I believe that my MP examples are readable by a human!
Testing this way will not show if the program is running correctly, it will only exhibit problems when the coding conventions are not respected and therefore the automatic binding generation does not work or there is some other coding problem resulting in a crash of the example.
@amirgon : Is this correct?
Up to now, weekly interventions were not necessary and there was just a major update necessary when we switched versions. 
When I was alerted that a number of programs did not work any longer or were simply not written yet (e.g. because the C version was written after my last intervention) I tried to respond as quickly as I could. However, I cannot make any promise for the future as I am also involved in other projects (physics teaching in Africa) and I may not be reachable on a weekly basis.

amirgon, 2023-07-11:
> It wasn't obvious to me so far, but now the MP examples are used as tests. I'm not sure it's a good idea. Tests can be boring, harder to read, generated, half for human half for machines, etc. Examples should be clean, simple, for humans.

@kisvegabor -  
Uli created a large number of simple, clean and readable examples and their first and most important goal is to be useful for humans as a reference.   
I added to the CI some automation that runs them, to check that they are valid and not broken due to some LVGL change. It is the equivalent of compiling the C tests in the CI.  
Since Micropython is an interpreted language, the simplest way to check the validity of an example is to run it.

> So it's ok to run the MP examples just to see if they can run (we compile the C examples too) but for testing there should be more effective and targeted options too.

They are not meant to be tests in the sense that you are thinking - they don't check LVGL logic, rendering or preformance.  
Running the Micropython examples can catch only the following problems:
- LVGL headers changed and the API the test uses is not valid any more (For example a function name changed, the number of parameters or their types changed, etc.)
- Something is wrong with the [LVGL coding convention](https://docs.lvgl.io/master/get-started/bindings/micropython.html#lvgl-c-api-coding-conventions). For example, `user_data` is not passed correctly in callbacks.
- [Garbage Collection problem](https://docs.lvgl.io/master/get-started/bindings/micropython.html#memory-management) - missing root pointers.
- Some bug in the Micropython Bindings scripts.

In fact, we found and fixed many of the above issues by running the Micropython examples.  
I think it is very useful to run them as part of LVGL CI as it ensures that:
- LVGL repo does not contain defected Micropython examples.
- LVGL API always follow the conventions we agreed upon.

> 4\. will be espesially important in the future if we want to maintain LVGL effectively. We should clearly see who are responsible for the binding, who can we ask to update the examples, give feedback, etc.

Maintaining and updating examples is a different task than giving feedback or explaining.  
The question is **who holds the responsibility** for creating and fixing examples as LVGL grows or changes.  
In that point - it seems we don't agree.   

In your view, the person who adds a new widget or changes one isn't responsible for creating/fixing the examples in the different languages (maybe only for C examples).   
In my view, if a person created a new widget he should be the owner of everything related to that widget, including any examples in any language that are part of the LVGL repo.

So, if you need my help for supporting by giving feedback or advice (regarding examples) - I'm here to help.  
But I don't think it should be my responsibility to fix an example every time someone else changes something in LVGL (nor do I have the capacity for that). I doubt you find someone who would want to volunteer to such a task on a daily basis as you ask...

> The tests done on the MP examples consist simply of running the program and checking that it does not crash. Therefore the style in which the program is written does not really influence the test. I believe that my MP examples are readable by a human!
> Testing this way will not show if the program is running correctly, it will only exhibit problems when the coding conventions are not respected and therefore the automatic binding generation does not work or there is some other coding problem resulting in a crash of the example.

@uraich Correct. You explained that very clearly and accurately, thank you.

s-hadinger, 2023-07-11:
Interesting discussion. I started a while ago some LVGL examples for Berry (in Tasmota) but I struggled because many of the C examples are quite advanced and use special features of the binding. 

I would have preferred to show some basic examples first and optional examples for advanced features.

XuNeo, 2023-07-11:
> and use special features of the binding.

I faced the similar problem for lua binding. The binding for lua is a simple wrapper focused on how to build UI, not to develop feature rich widgets which should be done in C given the porting between C and lua is pretty straight forward.

Actually I think it's the rule for all script language bindings if the UI needs higher refresh rate.

kisvegabor, 2023-07-13:
@amirgon @uraich 
These are valid statements. Let's try to find the root of the problem. I see these things:
1. In order to say a binding is officially supported, we need to specify the requirements. Something like what I described above: have some examples, docs, a CI, and at least 1 maintainer. But, I suggest not requesting to follow the C examples (can be done optionally, but not mandatory).
2. What are the responsibilities of the binding maintainers? 
    - **In case of large scale LVGL API changes, typically on new major versions:**  Before releasing a new major version we need to be sure that LVGL is still compatible with all the supported bindings. However, I don't think all API changes during the development needs to be followed in the binding too, but as we get closer to the release, we should see that the bindings are working too. It means we can have broken binding test during major version development. I think developing a new major version should be a joint effort and the binding maintainers should contribute by updating the tests. We have a new major version with 1.5-2 year period, so I think updating the examples at this rate is acceptable.   
    - **In case of changes in minor versions:** Minor version should be backward compatible, so ideally the bindings shouldn't be broken. I think related issues can be easily fixed by anyone by looking into the error of the CI without setting up a the binding's environment. If not, I believe we can get help from the maintainers. 
    - **Adding new minor features:** Automatically generated bindings shouldn't be broken due to the new feature. But I think it's acceptable if the new feature are not immediately added to the binding. I mean, what if there is no example for a new widget in some bindings? It's not the end of the world. We can leave it for later when the maintainers or other contributors will have time/interest to add examples. 

What do you think?

s-hadinger, 2023-07-14:
I agree with the criterias. I'm struggling for now in providing a CI as standalone compile. But this should be doable.

kisvegabor, 2023-07-14:
> I'm struggling for now in providing a CI as standalone compile.

For this anything simple would work, just to see that the binding still works with the given LVGL commit. If we say that the examples really can be used as tests, you can also just run the examples.

amirgon, 2023-07-14:
> 1. have some examples, docs, a CI, and at least 1 maintainer. But, I suggest not requesting to follow the C examples (can be done optionally, but not mandatory).

Sure, we don't have to follow the C examples. But the C examples are very good as a reference so it's usually easy to follow them when creating examples in other languages.   

Once an example is created and is part of the LVGL repo - there should be a way to validate it's not broken.
My point was - when someone changes LVGL API and breaks an example due to that change, it's that someone's responsibility to fix it.

> * In case of changes in minor versions ...

Minor versions are not a big concern since the public API doesn't change and new features are usually added to the main branch and not to the version branch. I don't think we had many issues with a version after it was released (for example v8).  
But it's still important to have tests in place to check that nothing is getting broken.

> * Adding new minor features ...

When a new widget is added, for example, if a Micropython example is missing we can almost guarantee this widget will not work on Micropython because the widget author surely missed some coding convention, didn't pass `user_data` correctly, forgot GC root pointers etc.  
The problem is that fixing this often means changing the widget public API, so if we notice this only after a major version is released it means it's usually too late to fix the public LVGL API and therefore the widget will be permanently broken on Micropython for that major version (possibly fixed only on the next major version)

> * In case of large scale LVGL API changes, typically on new major versions:

> * However, I don't think all API changes during the development needs to be followed in the binding too, but as we get closer to the release, we should see that the bindings are working too.

This is **not** how we did things until now.   
So far we had tests in place which detected problems on any PR or commit, no matter what the development stage was.

What you suggest goes against the continues-integration methodology: Instead of making sure that each small change doesn't break anything, you are suggesting now to wait and not run the Micropython tests on LVGL until the final stages of the development of a major version, at which point lots of things are probably already broken for months.  

In my opinion this will not work well:
- Major LVGL version development takes very long, 1.5-2 years as you mentioned. It's a lot of time and lots of things will surely go wrong.
- Fixing all the problems accumulated over a year or so before releasing a major version is a very heavy burden on the shoulders of the maintainers. It's much easier to handle small problems and keep LVGL and the bindings aligned over time than fix all the problems at once many months after things went wrong.
- Remember that many (most?) of the fixes are to the LVGL API (due to coding conventions issues), not fixes to the binding logic. If we only start to look at these LVGL API problems before a major release, most chances are that we will not have enough time to fix the LVGL API before the version is released - which means that this release will not work with the binding because of problems in the LVGL API in that release. Fixes will need to be postponed to next big release etc.
- The work of fixing the LVGL API due to coding conventions issue would be exclusively on the shoulders of the binding maintainers - instead of on the people who actually changed/developed that API in the first place.

> * I think developing a new major version should be a joint effort and the binding maintainers should contribute by updating the tests.

So, say you change a widget API and you also fix the C example. You fixed the C example because the widget uses a different API now. And suppose that widget also has a Micropython example.   
What you are saying is that it's my responsibility as the Micropython binding maintainer to fix that example although I didn't change the LVGL C API of that widget. The example was broken because you changed the LVGL API, not because anything was wrong with the example or with the Micropython binding scripts.  

I'm sorry, but I don't agree to this. I think that the person who changes the API should be responsible to all the fixes related to that API change in all examples including languages other than C.  
As bindings maintainers we can help, advice etc. but I don't think it should be our responsibility to fix them in such case.

That's why I'm pushing towards Continues Integration where the LVGL API is automatically tested (by running the examples) on every PR and commit during the all development stages. Continues Integration detects problems immediately so the author who changed the API could fix his PR on time before pushing it with defective API (defective in the sense of the agreed coding conventions).

And what about a totally new widget with a new LVGL API? That's a good question.  
My preference in such case is that the author who writes the C example would also write the Micropython example, but I understand this can be a bit hard for someone who knows little about Micropython. In that case the widget author can tag and ask for help (from Uli or me) and also ask us to create that example for him. I still think it should be the widget author goal to drive this and achieve a PR with as much examples as possible, and not only C examples.

[... cut ...]
I strongly feel that it's better **not** to do it that way.
Instead, in my opinion, it's better to keep the Continues Integration, where Micropython examples verify the LVGL API on every PR and commit.  
This guarantees that problems are detected and fixed as early as possible. It worked well so far! We found and fixed a lot of problems on time.

Neglecting LVGL API problems instead of fixing them immediately will make them accumulate and grow and overall will be much harder to properly fix many months after they were introduced.

kisvegabor, 2023-07-15:
> What you are saying is that it's my responsibility as the Micropython binding maintainer to fix that example although I didn't change the LVGL C API of that widget.  The example was broken because [...]

It'd like to highlight a few things: 
-  I don't change the API because I have free time and don't what to do. These changes are made to make LVGL better. 
- Not only I change the API of LVGL
- It's not only your responsibility to maintain the MicroPython binding. Actually nothing is mandatory. You are free to do as much as still feels good. However, of course, I'm really happy that you add your expertise to LVGL.

---

Anyway, I still see your points, but still can't see how it can work if we will have 5 bindings. Let's approach is from an other angle. IMO the C people has the following responsibilities regarding the bindings:
- Follow the coding standard
- Fix trivial things in the binding examples (I mean search and replace complexity things)
- Ask for help from the binding maintainers if needed
- Update the C code to make it compatible with all bindings. (based on feedback)

What are the responsibilities binding maintainers?
- When a test fails with no obvious reason?  Look into that given issue if cc-ed, as these random fails are usually require expertise to track.  I think it's ok and not a big burden. 
- When a new widget/feature is added?  Add at least 1 example. It will already ensure basic compatibility and would serve as a test as well. I think it's also ok. In the practice it means  max 1-2 example per month
- When there are huge major changes? This is the hard question. 
    - No responsibility, a C developers shall update all bindings? I think it won't work especially if we will have more bindings. I'm not speak about myself, but the contributors in general.
    - Updating only before the release? We are right, we were driving blind this way. We need continuous feedback.
    - What I was doing now, is that I was updating at least a few example to tests all features/widgets, but disabled the examples which don't test new things and would have been complicated to update. So it might be also ok to go with a smaller number of examples/tests during major version development. BTW, It's an other point where the examples and test could be separated. Now we have 10+ examples with draw events. These are really complex and needs some time to implement, especially if someone is not familiar with the syntax, debugging, and the environment in general. However one single test could check if the binding is able to perform custom draw things.   

To elaborate a little on examples vs tests: By "tests" I mean something like
- 1 test per widget: Create and delete a few instances, set some properties. That's it. It doesn't have to test the styles, events, and other things, because they will have they own tests and they are tested in C as well. That is if a slider's indicator can be styled in C, and we know that styling in general works in MP then we can say with a great confidence that the slider indicator can be styles in MP as well. An other example, we don't have to test all `LV_OBJ_FLAG_`s in MP. If we test them in C and we know that we can pass these flags to a widget, it should work in MP too. 
- 1 test for styles in general: global styles, local styles, parts, states, transitions.
- 1 test for events: e.g. attach a 2-3 events to a button and see if they can be triggered.
- 1 test for animations: just to see if we can start an animation in MP too
- 1 test for images: to if we can display images
- 1 test for fonts: load a font and use it on a widget
- 1 test for colors
- 1 test for the file system API
- probably 2-3 other things that I've missed now
 
I think it'd be ~1-2 day of work for a maintainer to add  these tests to a binding. And if see these passing, we would know that the binding is working quite well. 

What do you think?

kisvegabor, 2023-07-31:
Could you give some feedback about this?
> To elaborate a little on examples vs tests: By "tests" I mean something like
>
> - 1 test per widget: Create and delete a few instances, set some properties. That's it. It doesn't have to test the [...]

s-hadinger, 2023-07-31:
I agree with the list of tests, but I don't see yet how to automate them. But it should be doable to have a simple script going trough all the tests.

amirgon, 2023-08-18:
> Do you have any objection about the tests I have mentioned above?

I think I've already expressed my opinion above.
I believe that removing the MP examples and not running them as part of the CI is a step backwards.
The result will be that eventually many MP examples will become broken and it will be a big effort to fix them, and to fix LVGL itself as it becomes incompatible with the MP bindings.

Replacing these examples by tests, as you suggested, does not solve this problem. These tests will not cover what we cover today with the examples. We will then have to maintain both examples and tests, instead of only the examples. I don't see the benefit of maintaining tests separately from examples for the MP bindings.

> Real life case: We have a [larger update](https://github.com/lvgl/lvgl/issues/4358) which works in general, except with 2 MP examples. It's clear that, the new feature is working, but only the two examples needs to be fixed. We cannot delay this (or other features in the future) as fixing those examples is not really blocking.

Why are these MP examples failing in this update? Is that a problem with the examples or with LVGL?  
You are saying "It's clear that, the new feature is working, but only the two examples needs to be fixed"  
If it's clear that the problem is in the examples and not in LVGL then let's fix the examples or disable them until the problem is fixed. But - how are you so sure the problem is in the examples and not in LVGL?  

And how did you decide that "the new feature is working"? If it's not working with MP and only works in C, do you still consider this feature "working"?  
Specifically, we saw `lv_example_ffmpeg_2.py` crash with `corrupted size vs. prev_size`. This only happens after the LVGL change - nothing changed in MP. Don't you think we should suspect that something is wrong with the new feature and it is not really "working"? Or at least maybe this feature is not compatible with MP? (for example, missing gc roots)

Usually if examples ran before and started failing only after an LVGL change - it indicates that something is wrong with LVGL itself, not with the examples, and that LVGL is no longer compatible with the MP bindings in some way.  
If we stop running MP examples in LVGL it means we no longer make an effort to keep LVGL and the MP bindings compatible.

Suppose we didn't run MP examples in the CI, as you suggest. (and only run a few tests, if someone wrote them)  
Then a big LVGL update breaks these examples, but we will not know about this. The person who added this change and broke MP will not be aware of this.
Until the next big release another 10 big updates are merged into LVGL and break many more MP examples.
Broken MP examples are not the source of the problem - they are a symptom of incompatibility between the MP bindings and LVGL, and really indicate that the MP bindings cannot be used for certain use cases that the examples cover. So it's not just "examples are not working", but instead "we can't use MP bindings with that LVGL feature". 

Instead of fixing LVGL at the moment it breaks MP, we will face these issues only months later. Fixes may require going back to each of these big updates and identifying the problem on each (coding conventions that weren't followed, for example). This could be difficult to do after these updates are already merged into LVGL. Fixes may be API breaking so possibly unfixable on LVGL until next major release.

But anyway, I've already explained all that in this issue. I'm repeating myself and getting tired of trying to make a point. I feel that I've probably failed to explain the problem.   

So @kisvegabor just do whatever you think is best for LVGL and we'll wait and see what are the consequences in a few months or a year from now.

kisvegabor, 2023-08-22:
I cc here the core people from each binding and summarize what I feel like a sustainable direction. 
MicroPython: @amirgon @uraich 
JavaScript: @derekstavis
Berry: @s-hadinger 
PikaScript: @neuschaefer
Rust:  @nia-e
C++: still unclear

---

So this the the approach that I would like to follow with the bindings:
1. The binding example should be hosted in the binding's repository (it's already done this way for all the bindings except MicroPython)
2. For each binding we should have a few simple tests in the LVGL repo which run in our CI on each commit. LVGL should have hundreds of tests to the be sure LVGL itself is working, but the binding tests should tests only if the binding itself (mapping the given language to C) is working, and not deeply testing the LVGL features. 
3. If we see that the tests are passing, but the binding is broken, we update the LVGL or binding tests to find these kind of issues too in the future. 
4. We ensure that all tests are passing when a new feature is merged, and if needed tag the related maintainer to get help.
5. Following the changes of LVGL in the binding and examples is managed by the maintainers of binding. If an incompatibility is found, open an issue in the LVGL repo, so that we can discuss and fix it.
6. (Bonus): Setup a GitHub workspace for the binding so people can try it out with on click in their browser. (Already working for the LVGL repo). 

Just to briefly express my motivation behind these changes: we should unify how the bindings are managed to be sure that the bindings the we officially recommend are stable enough.  If we had the all the binding examples in the LVGL repo, it wouldn't be reasonable to ask any random contributor to update potentially hundreds of examples in several technologies/platforms.  As we know that the bindings are maintained by volunteers (which is amazing) , but no one guarantees that  they can fix the binding examples in 1-2 days. Due to this it can happen that a PR can not be merged for weeks because a binding maintainer is not available. We just can't work with such uncertainty and larger companies who are relying on new LVGL features can not afford this risk either. 

I'm deeply sorry that we couldn't find a solution which makes everyone happy. :slightly_frowning_face: However, I think this approach will provide more flexibility and a better workflow in the future.  

---

The first step is to create the mentioned basic tests for each binding. For this we need to create a simple CI/CD which sets up an environment for the binding and runs at least on test. Please let me know, if you can send a PR to get started with it.

neuschaefer, 2023-08-23:
@kisvegabor to be clear, I've only made the tiniest contribution to the PikaScript binding documentation, so I don't think I really qualify as "core person" there.

kisvegabor, 2023-08-23:
Sorry, I meant cc @pikasTech :slightly_smiling_face:

pikasTech, 2023-08-23:
> I cc here the core people from each binding and summarize what I feel like a sustainable direction. MicroPython: @amirgon @uraich JavaScript: @derekstavis Berry: @s-hadinger PikaScript: @pikasTech Rust: @nia-e C++: still unclear
> 
> So this the the approach that I would like to follow with the bindings:
> 
> 1. The binding example should be hosted in the binding's repository (it's already done this way for all the bindings except MicroPython)
> 2. For each binding we should have a few simple tests in the LVGL repo which run in our CI on each commit. LVGL should have hundreds of tests to the be sure LVGL itself is working, but the binding tests should tests only if the binding itself (mapping the given language to C) is working, and not deeply testing the LVGL features.
> 3. If we see that the tests are passing, but the binding is broken, we update the LVGL or binding tests to find these kind of issues too in the future.
> 4. We ensure that all tests are passing when a new feature is merged, and if needed tag the related maintainer to get help.
> 5. Following the changes of LVGL in the binding and examples is managed by the maintainers of binding. If an incompatibility is found, open an issue in the LVGL repo, so that we can discuss and fix it.
> 6. (Bonus): Setup a GitHub workspace for the binding so people can try it out with on click in their browser. (Already working for the LVGL repo).
> 
> Just to briefly express my motivation behind these changes: we should unify how the bindings are managed to be sure that the bindings the we officially recommend are stable enough. If we had the all the binding examples in the LVGL repo, it wouldn't be reasonable to ask any random contributor to update potentially hundreds of examples in several technologies/platforms. As we know that the bindings are maintained by volunteers (which is amazing) , but no one guarantees that they can fix the binding examples in 1-2 days. Due to this it can happen that a PR can not be merged for weeks because a binding maintainer is not available. We just can't work with such uncertainty and larger companies who are relying on new LVGL features can not afford this risk either.
> 
> I'm deeply sorry that we couldn't find a solution which makes everyone happy. 🙁 However, I think this approach will provide more flexibility and a better workflow in the future.
> 
> The first step is to create the mentioned basic tests for each binding. For this we need to create a simple CI/CD which sets up an environment for the binding and runs at least on test. Please let me know, if you can send a PR to get started with it.

I am quite supportive of this new arrangement, as it will make the organizational structure of LVGL's bindings more coherent. I am more than willing to continue maintaining the LVGL bindings for Pikascript.

pikasTech, 2023-08-23:
> 

While you may consider your contribution to be modest, it holds significance for PikaScript. Thank you for your support!

kisvegabor, 2023-08-25:
> I am quite supportive of this new arrangement, as it will make the organizational structure of LVGL's bindings more coherent. I am more than willing to continue maintaining the LVGL bindings for Pikascript.

It's amazing, thank you for the positive feedback! Do you plan to send a pull request with the mentioned CI update for PikaScript and a basic test?

pikasTech, 2023-08-29:
> It's amazing, thank you for the positive feedback! Do you plan to send a pull request with the mentioned CI update for PikaScript and a basic test?

I think I will complete this submission as soon as possible.

kisvegabor, 2023-08-29:
Thank you @pikasTech. Could you send me the  official icon of PikaScript? I would like to add it to our website.

pikasTech, 2023-08-31:
> Thank you @pikasTech. Could you send me the official icon of PikaScript? I would like to add it to our website.

You can use this image if you find it suitable.

[image]

kisvegabor, 2023-09-17:
Not stale

hermitsoft, 2023-10-02:
Hello LVGL community.

I'd like to join the project by taking part in the further development and modifications of the LVGL MicroPython binding and tests/examples.

kisvegabor, 2023-10-02:
Hi @hermitsoft,

So happy to hear that!  Please open a Pull request if you have something to show so that we can discuss the details.

kisvegabor, 2023-10-03:
For reference I add here what tests would be critical to have:
1. `test_widgets`: Create a few widgets (e.g. label, slider, chart, drop down, roller) and set some of their properties and flags. It will test if the create and property set/get API of the widgets is successfully exposed to by the binding.
2. `test_styles`: Create a slider and add some styles to various parts and states, and also add some local style properties. Use the style get API to see if the styles properties are added successfully. Change state as well.
3. `test_events`: Create a button, add a click event, and see if it can be triggered by `lv_obj_send_event()` 
4. `test_animation`: Create an animation to change the position of widget. Also set the start and ready callback to see if the animations was really running.
5. `test_image`: Load an image from file and array too.
6. `test_font`: Change the font of a label and see if its size has been changed accordingly.
7. `test_color`: Check if you can create a colors with `lv_color_*()` and `lv_palette_*()`
8. `test_layout`: Simple grid and flex tests with a few widgets
9. `test_file_system`: Check if file can be created, written and read back using `lv_fs_*()`

hermitsoft, 2023-10-03:
Thanks for the list. It will come handy soon as a base to build upon.

hermitsoft, 2023-10-18:
The above mentioned critical tests are made, pull-request was made, meanwhile @kisvegabor mentioned a few more ideas for the widget-tests to be implemented:
"various data types can be passed. For this is recommend setting and getting:
        label text (test string)
        an external array for a chart (integer array)
        points for a line widget (point array)
        the map of a button matrix (string array)
        options of the dropdown (multi line string)
        passing flags and states (enum)
 I think having test only for these widgets with a lot of gc.collect() calls should be suitable."

kisvegabor, 2023-11-03:
Not stale

kisvegabor, 2024-01-14:
The MicroPython test is added for LVGL v9. This [C code](https://github.com/lvgl/lvgl/blob/master/tests/src/test_cases/test_bindings.c) is [implemented in MicroPython](https://github.com/lvgl/lvgl/blob/master/tests/micropy_test/micropy.py) resulting in [this reference image](https://github.com/lvgl/lvgl/blob/master/tests/ref_imgs/binding.png).

In order to ensure that the all bindings are working well with LVGL supporting all the major features it would be important to have this single test added for each binding in the LVGL repo. This way we can say with great confidence that LVGL really supports a binding.  

LVGL v9 will be released tomorrow, and I'm sure it will take some time for the bindings to follow the changes. Please comment about your plans regarding 
- updating the binding to v9
- implementing this test

cc 
- JavaScript: @derekstavis
- Rust: @nia-e
- Berry: @s-hadinger
- PikaScript: @pikasTech

s-hadinger, 2024-01-14:
I haven't looked at v9 at all. Is there a summary of the major changes that could impact bindings?

kisvegabor, 2024-09-05:
Closing due to inactivity. Now only MicroPython has a test (thank you @kdschlosser) and therefore it's displayed on our website: 
[image]

If any other binding maintainers are interested in adding the test, please let me know in a comment.

closed 2024-09-05 by kisvegabor