lvgl/lvgl #4350: NULL pointer passed to lv_draw_sw_fill

kdschlosser, 2023-07-08:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

As the title states the dsc parameter is being passed aa NULL pointer

Here is the stack trace gotten from a debug build

```shell

Exception thrown: read access violation.
**dsc** was nullptr.
_lib_lvgl.dll!lv_draw_sw_fill(_lv_draw_unit_t * draw_unit=0x00000179e1298f20, const lv_draw_fill_dsc_t * dsc=0x0000000000000000, const lv_area_t * coords=0x00000179e11e40bc) Line 47
	at src\lvgl\src\draw\sw\lv_draw_sw_fill.c(47)
	
_lib_lvgl.dll!execute_drawing(lv_draw_sw_unit_t * u=0x00000179e1298f20) Line 309
	at src\lvgl\src\draw\sw\lv_draw_sw.c(309)
	
_lib_lvgl.dll!lv_draw_sw_dispatch(_lv_draw_unit_t * draw_unit=0x00000179e1298f20, _lv_layer_t * layer=0x00000179e10e1040) Line 151
	at src\lvgl\src\draw\sw\lv_draw_sw.c(151)
	
_lib_lvgl.dll!lv_draw_dispatch_layer(_lv_disp_t * disp=0x00000179e1101600, _lv_layer_t * layer=0x00000179e10e1040) Line 228
	at src\lvgl\src\draw\lv_draw.c(228)
	
_lib_lvgl.dll!lv_draw_dispatch() Line 124
	at src\lvgl\src\draw\lv_draw.c(124)
	
_lib_lvgl.dll!lv_draw_finalize_task_creation(_lv_layer_t * layer=0x00000179e10e1040, _lv_draw_task_t * t=0x00000179e11e4b00) Line 112
	at src\lvgl\src\draw\lv_draw.c(112)
	
_lib_lvgl.dll!lv_draw_rect(_lv_layer_t * layer=0x00000179e10e1040, const lv_draw_rect_dsc_t * dsc=0x00000057d81ed580, const lv_area_t * coords=0x00000179e0fd5ff8) Line 150
	at src\lvgl\src\draw\lv_draw_rect.c(150)
	
_lib_lvgl.dll!draw_knob(_lv_event_t * e=0x00000057d81ed988) Line 268
	at src\lvgl\src\widgets\slider\lv_slider.c(268)
	
_lib_lvgl.dll!lv_slider_event(const _lv_obj_class_t * class_p=0x00007ff96e11b720, _lv_event_t * e=0x00000057d81ed988) Line 230
	at src\lvgl\src\widgets\slider\lv_slider.c(230)
	
_lib_lvgl.dll!lv_obj_event_base(const _lv_obj_class_t * class_p=0x0000000000000000, _lv_event_t * e=0x00000057d81ed988) Line 89
	at src\lvgl\src\core\lv_obj_event.c(89)
	
_lib_lvgl.dll!event_send_core(_lv_event_t * e=0x00000057d81ed988) Line 307
	at src\lvgl\src\core\lv_obj_event.c(307)
	
_lib_lvgl.dll!lv_obj_send_event(_lv_obj_t * obj=0x00000179e0fd5f50, lv_event_code_t event_code=LV_EVENT_DRAW_MAIN, void * param=0x00000179e10e1040) Line 64
	at src\lvgl\src\core\lv_obj_event.c(64)
	
_lib_lvgl.dll!lv_obj_redraw(_lv_layer_t * layer=0x00000179e10e1040, _lv_obj_t * obj=0x00000179e0fd5f50) Line 103
	at src\lvgl\src\core\lv_refr.c(103)
	
_lib_lvgl.dll!refr_obj(_lv_layer_t * layer=0x00000179e10e1040, _lv_obj_t * obj=0x00000179e0fd5f50) Line 786
	at src\lvgl\src\core\lv_refr.c(786)
	
_lib_lvgl.dll!lv_obj_redraw(_lv_layer_t * layer=0x00000179e10e1040, _lv_obj_t * obj=0x00000179e132b150) Line 148
	at src\lvgl\src\core\lv_refr.c(148)
	
_lib_lvgl.dll!refr_obj(_lv_layer_t * layer=0x00000179e10e1040, _lv_obj_t * obj=0x00000179e132b150) Line 786
	at src\lvgl\src\core\lv_refr.c(786)
	
_lib_lvgl.dll!lv_obj_redraw(_lv_layer_t * layer=0x00000179e10e1040, _lv_obj_t * obj=0x00000179e132b330) Line 148
	at src\lvgl\src\core\lv_refr.c(148)
	
_lib_lvgl.dll!refr_obj(_lv_layer_t * layer=0x00000179e10e1040, _lv_obj_t * obj=0x00000179e132b330) Line 786
	at src\lvgl\src\core\lv_refr.c(786)
	
[... 260 lines cut ...]

    /*Let the render thread work*/
    lv_thread_sync_signal(&draw_sw_unit->sync);
#else
    t->state = LV_DRAW_TASK_STATE_IN_PROGRESS;
    draw_sw_unit->base_unit.target_layer = layer;
    draw_sw_unit->base_unit.clip_area = &t->clip_area;
    draw_sw_unit->task_act = t;

    if (draw_sw_unit->task_act->draw_dsc != NULL) {
        execute_drawing(draw_sw_unit);
    }

    draw_sw_unit->task_act->state = LV_DRAW_TASK_STATE_READY;
    draw_sw_unit->task_act = NULL;

    /*The draw unit is free now. Request a new dispatching as it can get a new task*/
    lv_draw_dispatch_request();

#endif
    return 1;
}
```

I think that would fix the issue. Unless draw_dsc is supposted to be populated, If that is the case then someone else will have to come up with a fix because it is above my pay grade to handle.

kisvegabor, 2023-07-09:
Hi,

How can I reproduce the `NULL` issue?

kdschlosser, 2023-07-10:
beats me... it just did it for me. I added that check for the NULL pointer which corrected the problem. There was nothing crazy about the code. I believe I was running the first animation example when it happened. Let me see if I am able to poke a stick into it on the simulator and get it to happen.

kdschlosser, 2023-07-10:
I cannot do it on the simulator because the simulator has not been compiled with the most recent changes. If you compile the online simulator with the most recent changes it should do it if you run the animation from even example.

kisvegabor, 2023-07-10:
I mean can you reproduce it in C? So that I can check and debug it on my end?

kdschlosser, 2023-07-10:
https://github.com/lvgl/lvgl/blob/08870996d15e4bf4e8f3ca976991d35a76c7054e/examples/anim/lv_example_anim_1.c

That code should cause the problem. The python version of it caused the problem so I would imagine the C version would as well.

kisvegabor, 2023-07-12:
Actually it's working in C. 

Do you see it fail immediately or only when you click the switch?

kdschlosser, 2023-07-13:
Hmmm... I will have to undo the code change I made to correct the problem and I will do some more testing on it to see when it pops up. I don't remember if it was right away or not. It might not have been immediately and only occurred after several times of toggling. Let me mess about with it and see if I can make it reproducible.

closed 2023-08-04 by lvgl-bot