lvgl/lvgl #4355: Usage of struct _xxx_t * and xxx_t

XuNeo, 2023-07-09:
### Introduce the problem

`struct _xxx_t *` is widely used in functions and struct parameters. Is there special reason not to use the `typedef` version `xxx_t`?

### Examples and cases

```
struct _lv_obj_t * lv_disp_get_layer_bottom(lv_disp_t * disp);
void lv_disp_set_theme(lv_disp_t * disp, struct _lv_theme_t * th);
```

### Suggested solution

Use `xxx_t` where possible.

kisvegabor, 2023-07-10:
We should use `xxx_t *` where possible, but on some places we cannot include the required header file due to circular includes. In these cases we use  a forward declaration (e.g. `struct _lv_theme_t;`) to avoid including the problematic header.

C47D, 2023-07-10:
Is there a way to detect unused headers? Maybe a cleanup of unused headers can reduce the number of circular dependencies.

kisvegabor, 2023-07-13:
It would be really great, I'm sure there is a lot to optimize in this regard.

XuNeo, 2023-07-15:
Let's continue in #4380

closed 2023-07-15 by XuNeo