lvgl/lvgl #4358: LVGL support  multiple running instances

bjsylvia, 2023-07-10:
### Introduce the problem
In some development patterns, we need LVGL to support the following scenarios.
1. Each app will have its own independent LVGL instance, which will not affect each other. 
2. The apps may belong to different threads in different processes, or they may belong to different threads in the same process.

However, with the presence of numerous static global variables in LVGL, supporting multiple instances becomes challenging. One solution to this issue is to first centrally manage the global variables that affect multiple instances and then bind them to the instance thread using TLS(thread-local storage, pthread_key_xxx) , thereby achieving multi-instance capability.

Defining an efficient organization and management system for global variables that impact multiple instances is a crucial prerequisite for supporting multi-instance capability. Let's focus on finding a solution that minimizes and optimizes modifications to LVGL to address this issue.

### Examples and cases

### Suggested solution

kisvegabor, 2023-07-12:
Hi,

So let's say we will have an `lv_globals_t lv_global` variable, but it's declaration will be `__thread lv_globals_t lv_global`. So when you call `lv_init` in a thread it will use the thread's `lv_global` variable. (Of course the `__thread` attribute should come from `lv_conf.h`)

If so, it seems good to me and it won't require conceptual changes into LVGL, just to introduce the `lv_global` and use it similarly to `LV_GC_ROOT`.

bjsylvia, 2023-07-12:
Yes, lv_global is exactly what we need. 

There are over 200 non-const static global variables in LVGL. Next, we need to determine how to adjust the basic strategy for these variables so that we can proceed with the coding work.

kisvegabor, 2023-07-13:
> There are over 200 non-const static global variables in LVGL.

Oh, that's a lot. Could you attach a list of these variables? I hope we can make most of them local somehow.

bjsylvia, 2023-07-15:
> > There are over 200 non-const static global variables in LVGL.
> 
> Oh, that's a lot. Could you attach a list of these variables? I hope we can make most of them local somehow.

Yes. 

The global variable data extracted from v9 master(commit 562d30694) is available in attached file: 
[LVGL Global variables (7.14 master).xlsx](https://github.com/lvgl/lvgl/files/12060804/LVGL.Global.variables.7.14.master.xlsx).

It contains three sheets corresponding to the LVGL subdirectories: src(104), demos(120), and examples(51). 
The src directory is the critical one that requires adjustments, consisting of a total of 104 entries. Let's address the src directory first.

kisvegabor, 2023-07-15:
I think we don't need to care much with the examples and demos. 

Regarding LVGL itself, I can clean it up on Monday. I think we will have ~20 variables in the end.

bjsylvia, 2023-07-16:
> I think we don't need to care much with the examples and demos.

Yes, I agree absolutely.
> 
> Regarding LVGL itself, I can clean it up on Monday. I think we will have ~20 variables in the end.

Great. We can analyze whether there is further room for optimization in the next stage.

kisvegabor, 2023-07-17:
I've removed a lot of global variables from `lvgl/src` and pushed it to the `feat/multi-instance` branch. Now we still have ~33 global variables.

`lv_tabview` tabview will be reworked and I'll remove the globals after that.   

What do you think?

bjsylvia, 2023-07-23:
Glad to see your progress! After retesting the global variables in the multi-instance branch, we found that there are only 28 variables remaining. It is amazing! (The statistics for rlottie and tabview have been removed, as their usage does not affect multi-instance support).

[image]

kisvegabor, 2023-07-23:
Cool! It seems we could still remove a 3-4 variables, but it really seems manageable now.

kisvegabor, 2023-07-24:
To move forward I think we should add an `lv_global_t` in a `src/core/lv_global.h` file and declare that variable in `lv_init.c`. 
If you agree with that can you add it and send a PR to `feat/multi-instance`?

bjsylvia, 2023-07-24:
Ok, I'll try it later.

bjsylvia, 2023-07-29:
In the previous statitics, about 40 local static variables were missed. The current implementation is [PR-4418](https://github.com/lvgl/lvgl/pull/4418) . Now there are still a few variables（8） waiting to be processed. Can you provide some ideas for them ?

[image]

For example `lv_tick_inc` requires accessing the same global variable `sys_time` across threads in SDL backend? how should we process it?

kisvegabor, 2023-07-31:
- `lv_timer.c` `run_cnt`: we can mode it to `lv_timer_state_t`
- `sdl`, `drm`: do we need to consider SDL here? At the first pass I think we can ignore it
- `indev.c` `prev_state`: We can move it [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev_private.h#L95)
- `lv_draw.c` `running`: Let's move it to among the global variables
- `lv_tick.c` `sys_time`:  Let's move it to among the global variables. What's the problem getting think in SDL? It will get the tick from it's own TLS, right?

bjsylvia, 2023-08-05:
> * `lv_timer.c` `run_cnt`: we can mode it to `lv_timer_state_t`

Done

> * `sdl`, `drm`: do we need to consider SDL here? At the first pass I think we can ignore it

Ok, let's ignore them firstly.

> * `indev.c` `prev_state`: We can move it [here](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev_private.h#L95)

I only see the usage of prev_state after initialization, without any further **assignment**. Is it necessary?
> * `lv_draw.c` `running`: Let's move it to among the global variables

Done. I move it to `lv_draw_cache_t`.

> * `lv_tick.c` `sys_time`:  Let's move it to among the global variables. What's the problem getting think in SDL? It will get the tick from it's own TLS, right?

The SDL backend creates a separate tick thread, which modifies the global variable `sys_time` through `lv_tick_inc`. This ensures the normal operation of the main thread that depends on `sys_time`. If `sys_time` is configured as TLS, `lv_tick_inc` will only modify the `sys_time` in the tick thread instead of the main thread, which will cause the program to fail to start properly. `lv_tick_get` is unable to obtain the correct sys_time.

```
lv_disp_t * lv_sdl_window_create(lv_coord_t hor_res, lv_coord_t ver_res)
{
 ...
#if !LV_TICK_CUSTOM
       SDL_CreateThread(tick_thread, "LVGL thread", NULL);
#endif
...    
}
#if !LV_TICK_CUSTOM
static int tick_thread(void * ptr)
{
    LV_UNUSED(ptr);
    static uint32_t tick_prev = 0;

    while(1) {
        uint32_t tick_now = SDL_GetTicks();
        lv_tick_inc(tick_now - tick_prev);
        tick_prev = tick_now;
        SDL_Delay(5);
    }

    return 0;
}
#endif
```

```
#if !LV_TICK_CUSTOM
/**
 * You have to call this function periodically
 * @param tick_period the call period of this function in milliseconds
 */
LV_ATTRIBUTE_TICK_INC void lv_tick_inc(uint32_t tick_period)
{
    tick_irq_flag = 0;
    sys_time += tick_period;
}
#endif

/**
 * Get the elapsed milliseconds since start up
 * @return the elapsed milliseconds
 */
uint32_t lv_tick_get(void)
{
#if LV_TICK_CUSTOM == 0

    /*If `lv_tick_inc` is called from an interrupt while `sys_time` is read
     *the result might be corrupted.
     *This loop detects if `lv_tick_inc` was called while reading `sys_time`.
     *If `tick_irq_flag` was cleared in `lv_tick_inc` try to read again
     *until `tick_irq_flag` remains `1`.*/
    uint32_t result;
    do {
        tick_irq_flag = 1;
        result        = sys_time;
    } while(!tick_irq_flag); /*Continue until see a non interrupted cycle*/

    return result;
#else
    return LV_TICK_CUSTOM_SYS_TIME_EXPR;
#endif
}
```

Error message is below:
```
[Warn]	(0.000, +0)	 lv_timer_handler: It seems lv_tick_inc() is not called. 	(in lv_timer.c line #93)
```

bjsylvia, 2023-08-05:
> * `lv_tick.c` `sys_time`:  Let's move it to among the global variables. What's the problem getting think in SDL? It will get the tick from it's own TLS, right?

 I have removed the usage of `tick_thread` and added the `lv_tick_set_cb` mechanism. The update has been included in the PR. Please confirm if the solution is feasible.

kisvegabor, 2023-08-06:
> I only see the usage of prev_state after initialization, without any further assignment. Is it necessary?

It can be really removed.

> I have removed the usage of tick_thread and added the lv_tick_set_cb mechanism. The update has been included in the PR. Please confirm if the solution is feasible.

I like the idea to make the custom tick handling run time configurable but I suggest do it this way:
1. Use an  `uint32_t tick_get_cb(void)`
2. In `lv_tick_get()` 
```c
if(state_p->tick_get_cb) return tick_get_cb();
else {
/*The current use of sys_time*/
}
``` 
3. Remove `LV_TICK_CUSTOM` from `lv_conf.h` to have only the run time configurable option

---

In general one thing is still missing from this PR: please add a few paragraph of docs in the porting section. Also mention the `__threat` and `phread_key` option as example use cases.

bjsylvia, 2023-08-06:
PR has updated `tick_get_cb` and removed `LV_TICK_CUSTOM`.

Porting docs will be updated later.

bjsylvia, 2023-08-06:
@kisvegabor Docs have been updated. Please review it.

kisvegabor, 2023-08-07:
I think I've found a problem which can cause the failure of the MicroPython CI. We haven't considered the [GC roots](https://github.com/lvgl/lvgl/blob/feat/multi-instance/src/misc/lv_gc.h#L45-L64). 

These are also global variables, and should be moved to `lv_global`. Finally `lv_global` should be also a GC_ROOT. I'm not sure bout it, but it seems ideally `lv_global` should be the only GC_ROOT.

bjsylvia, 2023-08-08:
Does MicroPython depends on GC roots?  I'm not clear how to add `lv_global` to GC roots. `lv_global` now is declared or allocated within `lv_global_default`.

We initialize `lv_global` in `lv_init`. It seems that `LV_ENABLE_GC` and `_lv_gc_clear_roots` is no longer necessary.

kisvegabor, 2023-08-08:
cc @amirgon @uraich 
Short summary: we have moved all LVGL global and static variable into an `lv_global_t` structure and now thinking about how it affects GC roots. The goal of this update is to support using multiple LVGL instance in parallel in separate threads and store  `lv_global_t` variable of each instance in TLS (Thread Local Storage). If `LV_GLOBAL_CUSTOM` is disabled `lv_global_default()` returns a `static lv_global_t` variable. If `LV_GLOBAL_CUSTOM` is enabled the user needs to implement `lv_global_default()` and return `lv_global_t` from TLS or anyhow the user wishes. 
See https://github.com/lvgl/lvgl/tree/feat/multi-instance

I'm not sure we need to support multiple instances for MicroPython so we can assume that if GC is used `LV_GLOBAL_CUSTOM` is disabled.

What do you think?

amirgon, 2023-08-09:
> we have moved all LVGL global and static variable into an `lv_global_t` structure and now thinking about how it affects GC roots.

It may simplify things because in the Micropython Bindings we already do that.  
See https://github.com/lvgl/lv_binding_micropython/blob/master/include/lv_mp_root_pointers.h where we define a struct to contain all GC roots, and later tell MP that this struct itself is a GC root. However, the struct should be defined in a special file that is scanned by MP and [can be later registered with `MP_REGISTER_ROOT_POINTER`](https://github.com/lvgl/lv_binding_micropython/blob/6369315bb7bc3dc47f0b2cc33f7e7f8d830a5749/gen/gen_mpy.py#L1030C1-L1030C75)

> `LV_GLOBAL_CUSTOM` is enabled the user needs to implement `lv_global_default()` and return `lv_global_t` from TLS or anyhow the user wishes.

I think it's best to implement `lv_global_default` for MP to return `lvgl_root_pointers`, even when there is only a single instance of LVGL.

> I'm not sure we need to support multiple instances for MicroPython so we can assume that if GC is used `LV_GLOBAL_CUSTOM` is disabled.

I also don't see the need for multiple instances in MicroPython, but still think the best way is to enable `LV_GLOBAL_CUSTOM` to return the GC allocated roots.  
In fact, after doing that, we may no longer need all [these weird macros for defining GC roots](https://github.com/lvgl/lvgl/blob/73de6a2e94378450ff7ceb13d0919ac085fd9925/src/misc/lv_gc.h#L37-L77). They were needed to conditionally define all roots in `lvgl_root_pointers` struct when building with MP. But if all globals and statics are already allocated in a struct, we just need to make sure that this struct is allocated by GC.

bjsylvia, 2023-08-09:
It looks like we need to move the global variables in GC roots to `lv_global` firstly.  I will try it later.

kisvegabor, 2023-08-09:
Thank you, Amir. Glad to hear that it will be useful or MicroPython too. 

@bjsylvia already [started to work on](https://github.com/lvgl/lvgl/pull/4440) moving the GC roots to `lv_global_t`. :+1:

kisvegabor, 2023-08-09:
Thank you! It looks good to me, but I think @amirgon can tell more about how to proceed. (#4440)

bjsylvia, 2023-08-09:
If enable `LV_GLOBAL_CUSTOM` and custom `lv_global_default` for MP,  we shouldn't need `LV_GC_INIT` anymore, right? 
`lv_global_t` has been initialized [here](https://github.com/lvgl/lvgl/blob/8e88342e0f3792c6edbb71461498bd0c04f35fb0/src/lv_init.c#L89-L92).

amirgon, 2023-08-09:
> we shouldn't need `LV_GC_INIT` anymore, right?

`LV_GC_INIT` was needed to allocate `lv_global_t` by GC (with `m_new0`) and to update it in MP state with `MP_STATE_VM`.
See: https://github.com/lvgl/lv_binding_micropython/blob/6369315bb7bc3dc47f0b2cc33f7e7f8d830a5749/lv_conf.h#L322

Now possibly MP specific `lv_global_default` can do this instead?

kisvegabor, 2023-08-09:
> Now possibly MP specific lv_global_default can do this instead?

@amirgon 
How complicated would it be to try out in MicroPython? If not so complicated could you create a new branch in `lv_binding_micropython`and add an example implementation?

FYI, now the [MP CI test is failing](https://github.com/lvgl/lvgl/actions/runs/5786996566/job/15682927000) because it can't get the default display (it was stored in a global variable which is now moved to `lv_global_t`):
[image]

Can it be because of the incorrect GC Root handling? Or is it something else?

amirgon, 2023-08-10:
> How complicated would it be to try out in MicroPython? If not so complicated could you create a new branch in `lv_binding_micropython`and add an example implementation?

I'll try this later and let you know.

> Can it be because of the incorrect GC Root handling? Or is it something else?

I really can't tell without debugging it. I'll check this after implementing MP `lv_global_default`.

kisvegabor, 2023-08-10:
Great, thank you!

amirgon, 2023-08-10:
We still need `LV_GC_INIT` to create an instance of `lv_global_t`.  

LVGL now calls `LV_GC_INIT` too late.  
In `lv_init` it must be called **before** `lv_global_init` because it calls:  
`_lv_log_add` :arrow_right: `lv_tick_get` :arrow_right: `state_p->tick_get_cb` which is `tick_state` member of `lv_global_t`....

So first, please keep `LV_GC_INIT` and move it **before anything else**.

We no longer need `src/misc/lv_gc.h` at all so you can delete it. We can also remove `LV_ENABLE_GC`, `LV_GC_INCLUDE`, `LV_GC_ROOT` etc. from `lv_conf.h`. We only need `LV_GC_INIT`.

I've created a branch on lv_micropython_bindings which enables `LV_GLOBAL_CUSTOM` and defines `lv_global_default`.
See: https://github.com/lvgl/lv_binding_micropython/tree/feat/multi-instance (and corresponding lv_micropython: https://github.com/lvgl/lv_micropython/tree/feat/multi-instance)  
This is the relevant commit: https://github.com/lvgl/lv_binding_micropython/commit/0621a3db0f55664504eb0631b9aa268e33148626

I was able to run some examples (for example advanced_demo.py) and also tried calling `gc.collect()` - it seems to work fine.  
However, there are some issues with `custom_widget_example.py`. Some things I noticed:
- `lv_is_initialized` crashes if called very early (before lvgl is initialized)
- Something changed with `LAYOUT_FLEX` ??
- In general this example no longer works (doesn't display the custom widget)

So there are some problems that should be taken care of, which don't seem directly related to this "multi-instance" feature.
  
btw why didn't you define both `LV_GLOBAL_CUSTOM` and `lv_global_default` as a **single** macro that can be used for both indicating that there are custom globals and accessing them? Something like 
```c
#define LV_GLOBAL_DEFAULT() my_globals
```
or 
```c
#define LV_GLOBAL_DEFAULT() get_my_globals()
```
the advantage of this approach is that it could improve performance by accessing the globals directly instead of going through a function indirection. We could use this optimization in Micropython since `lv_global_default` just [returns a pointer of a global variable](https://github.com/lvgl/lv_binding_micropython/blob/0621a3db0f55664504eb0631b9aa268e33148626/gen/gen_mpy.py#L1040).

bjsylvia, 2023-08-11:
> We still need `LV_GC_INIT` to create an instance of `lv_global_t`.
> LVGL now calls `LV_GC_INIT` too late. In `lv_init` it must be called **before** `lv_global_init` because it calls: `_lv_log_add` ➡️ `lv_tick_get` ➡️ `state_p->tick_get_cb` which is `tick_state` member of `lv_global_t`....
> 
> So first, please keep `LV_GC_INIT` and move it **before anything else**.
> 
> We no longer need `src/misc/lv_gc.h` at all so you can delete it. We can also remove `LV_ENABLE_GC`, `LV_GC_INCLUDE`, `LV_GC_ROOT` etc. from `lv_conf.h`. We only need `LV_GC_INIT`.

I have remove them, please review it in [PR](https://github.com/lvgl/lvgl/pull/4440/commits/1f39c0e100ad5c842de88f53642a4c38db6133e6)

> I've created a branch on lv_micropython_bindings which enables `LV_GLOBAL_CUSTOM` and defines `lv_global_default`. See: https://github.com/lvgl/lv_binding_micropython/tree/feat/multi-instance (and corresponding lv_micropython: https://github.com/lvgl/lv_micropython/tree/feat/multi-instance) This is the relevant commit: [lvgl/lv_binding_micropython@0621a3d](https://github.com/lvgl/lv_binding_micropython/commit/0621a3db0f55664504eb0631b9aa268e33148626)
> 
> I was able to run some examples (for example advanced_demo.py) and also tried calling `gc.collect()` - it seems to work fine. However, there are some issues with `custom_widget_example.py`. Some things I noticed:
> 
> * `lv_is_initialized` crashes if called very early (before lvgl is initialized)
> * Something changed with `LAYOUT_FLEX` ??

[here is `LAYOUT_FLEX`](https://github.com/lvgl/lvgl/blob/8e88342e0f3792c6edbb71461498bd0c04f35fb0/src/layouts/lv_layout.h#L39-L41)
> * In general this example no longer works (doesn't display the custom widget)
>
> So there are some problems that should be taken care of, which don't seem directly related to this "multi-instance" [...]

I agree. I also want to change the meaning of `LV_GLOBAL_CUSTOM`. It allows user define it in lv_conf.h, such as:
```
#define LV_GLOBAL_CUSTOM lv_global_default()
```
or
```
struct _lv_global_t;
extern struct _lv_global_t my_custom_global;
#define LV_GLOBAL_CUSTOM &my_custom_global
```
`extern` looks ugly. Maybe choose other macro replace it.

@kisvegabor @amirgon Is it right? If feasible, I will update it.

kisvegabor, 2023-08-11:
> Something changed with LAYOUT_FLEX ??

`LV_OBJ_FLAG_FLAX_NEW_TRACK` is also a [normal LV_OBJ_FLAG now](https://github.com/lvgl/lvgl/blob/8e88342e0f3792c6edbb71461498bd0c04f35fb0/src/core/lv_obj.h#L120).

> btw why didn't you define both LV_GLOBAL_CUSTOM and lv_global_default as a single macro that can be used for both indicating that there are custom globals and accessing them? Something like

I'm ok with this too.

bjsylvia, 2023-08-11:
PR has been updated. Please review it.

kisvegabor, 2023-08-11:
Thank you! Maybe `LV_GLOBAL_CUSTOM ` can be omitted too but it doesn't really matter now. It would be more important to see it fully working.

amirgon, 2023-08-11:
> PR has been updated. Please review it.

- I think `LV_GLOBAL_CUSTOM` should be defined with `#define LV_GLOBAL_CUSTOM()` and expanded with `LV_GLOBAL_CUSTOM()` otherwise it can't be used with a global variable (note the parenthesis).   
Please fix places like `#define LV_GLOBAL_DEFAULT() LV_GLOBAL_CUSTOM` to `#define LV_GLOBAL_DEFAULT() LV_GLOBAL_CUSTOM()`
- We might want to consider keeping `LV_GC_INCLUDE` and including it on `lib/lv_bindings/lvgl/src/lv_init.c`, since `LV_GC_INIT()` macro expansion may require some headers to be included. (now I work around this by exten in lv.conf.h)

Issues related to lv_init:

- `LV_GC_INIT` must be called in lv_init before anything else. Now it's aftert `lv_global_init`.
- `lv_initialized` must be checked before calling `lv_global_init` (but after `LV_GC_INIT`). This is required when calling `lv_init` multiple times (this can happen when running with Micropython)

Other issues:
- lv_example_ffmpeg_2.py crashes with `free(): corrupted unsorted chunks`. Maybe some related global was missed out?
- I'm getting lots of `_lv_inv_area: detected modifying dirty areas in render` when running lv_example_get_started_2.py ??
- Issues with custom_widget_example.py

bjsylvia, 2023-08-12:
> * I think `LV_GLOBAL_CUSTOM` should be defined with `#define LV_GLOBAL_CUSTOM()` and expanded with `LV_GLOBAL_CUSTOM()` otherwise it can't be used with a global variable (note the parenthesis). [...]

Done.

> * We might want to consider keeping `LV_GC_INCLUDE` and including it on `lib/lv_bindings/lvgl/src/lv_init.c`, since `LV_GC_INIT()` macro expansion may require some headers to be included. (now I work around this by exten in lv.conf.h)
> * Same as `LV_GC_INCLUDE`, consider adding optional `LV_GLOBAL_DEFAULT_INCLUDE` when `LV_GLOBAL_DEFAULT` requires some context defined elsewhere.

Now, I use `LV_GLOBAL_CUSTOM_INCLUDE` for user-defined header. For example:
```
#define LV_ENABLE_GLOBAL_CUSTOM 1
#define LV_GLOBAL_CUSTOM() (&lv_global_extern)
#define LV_GLOBAL_CUSTOM_INCLUDE "demo_gc.h"
```
If user customize `lv_global` by implementing `lv_global_default()`, only the configuration below is needed.
```
#define LV_ENABLE_GLOBAL_CUSTOM 1
```

> Issues related to lv_init:
>
> * `LV_GC_INIT` must be called in lv_init before anything else. Now it's aftert `lv_global_init`.
> * `lv_initialized` must be checked before calling `lv_global_init` (but [...]
Done. [Please check it](https://github.com/lvgl/lvgl/pull/4440/commits/7b6910bcfbc93710dd1cfd9d7e1ab4eb1dbf1099)
> Other issues:
> 
> * lv_example_ffmpeg_2.py crashes with `free(): corrupted unsorted chunks`. Maybe some related global was missed out?
> * I'm getting lots of `_lv_inv_area: detected modifying dirty areas in render` when running lv_example_get_started_2.py ??
> * Issues with custom_widget_example.py

 I will try to set up the running environment on my mac.

kisvegabor, 2023-08-12:
> I'm getting lots of _lv_inv_area: detected modifying dirty areas in render when running lv_example_get_started_2.py ??

It's already fixed in master [here](https://github.com/lvgl/lvgl/commit/0302eaa22eb855656f65300c4353909c6a78c667)

amirgon, 2023-08-12:
> It's already fixed in master [here](https://github.com/lvgl/lvgl/commit/0302eaa22eb855656f65300c4353909c6a78c667)

@bjsylvia Could you rebase?

I've updated MP `feat/multi-instance` branch. I suggest testing MP on GitHub CI by changing:

https://github.com/lvgl/lvgl/blob/ebc1f43f4042cfe3219c772fd6d2cc25151bcfe6/.github/workflows/build_micropython.yml#L26

to 

```
git checkout feat/multi-instance
```

Open issues:
- `lv_example_ffmpeg_2.py` crashes with `corrupted size vs. prev_size`
- `custom_widget_example.py` test passes but widget is not drawn (possibly not related to this PR)
- @kisvegabor LVGL master is now aligned to MP `feat/parallel-rendering` branch. After merging this PR it will be aligned with `feat/multi-instance`. lv_micropython master branch is behind because it's not tested yet with ports other than the Linux port, but at some point we'll need to align it with latest LVGL...

bjsylvia, 2023-08-13:
@kisvegabor Maybe we can merge [PR](https://github.com/lvgl/lvgl/pull/4440) firstly. Then rebase to master?

kisvegabor, 2023-08-13:
@bjsylvia 
I've merged the PR but won't have time today to rebase on master. :slightly_frowning_face:

bjsylvia, 2023-08-13:
> @bjsylvia I've merged the PR but won't have time today to rebase on master. 🙁

I have rebased it.  Here is PR: [https://github.com/lvgl/lvgl/pull/4452](https://github.com/lvgl/lvgl/pull/4452).

Can you change MP branch on GitHub CI? [lvgl/.github/workflows/build_micropython.yml](https://github.com/lvgl/lvgl/blob/ebc1f43f4042cfe3219c772fd6d2cc25151bcfe6/.github/workflows/build_micropython.yml#L26)

kisvegabor, 2023-08-13:
Merged. I've also fixed a minor error, and the changes the MicroPython CI.

> Open issues: [...]

These are probably some just some local issues. If the other things are working then it seems we are on the right track.

bjsylvia, 2023-08-16:
@kisvegabor 
I fix a merging error from `master`, [https://github.com/lvgl/lvgl/pull/4460](https://github.com/lvgl/lvgl/pull/4460)

@amirgon 

for rp2, Would you like to fix some CI compiling errors after we removed `lv_gc.h` ?
```
Error: /home/runner/work/lvgl/lvgl/ports/rp2/mpconfigport.h:210:10: fatal error: lib/lv_bindings/lvgl/src/misc/lv_gc.h: No such file or directory
  210 | #include "lib/lv_bindings/lvgl/src/misc/lv_gc.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
Error: /home/runner/work/lvgl/lvgl/ports/rp2/mpconfigport.h:207:10: fatal error: lib/lv_bindings/lvgl/src/misc/lv_gc.h: No such file or directory
  207 | #include "lib/lv_bindings/lvgl/src/misc/lv_gc.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

You can find it [here](https://github.com/lvgl/lvgl/actions/runs/5873522812/job/15926910475).

kisvegabor, 2023-08-17:
The `ffmpeg` and the `custom_widget_example` are not global issues affecting the concept in general, so let's disable them temporarily, merge this feature to master and fix  the examples later.

amirgon, 2023-08-18:
> @amirgon
> 
> for rp2, Would you like to fix some CI compiling errors after we removed `lv_gc.h` ?

Yes, done in https://github.com/lvgl/lv_micropython/commit/71254f9e1705eee17e176889ab481643b62e78c6

closed 2023-08-18 by kisvegabor

kisvegabor, 2023-08-18:
Finally I disabled `lv_example_imgfont_1.py` and `lv_example_ffmpeg_2.py` and merged `feat/multi-instance` to `master`.

There were a lot of conflicts during merging. I resolved those and the CI is passing, however some minor regression might happen.

amirgon, 2023-08-19:
> Finally I disabled `lv_example_imgfont_1.py` and `lv_example_ffmpeg_2.py` and merged `feat/multi-instance` to `master`.

1. Commenting out the code makes more sense than deleting everything, I think, and would make it easier to re-enable them when possible.
2. These failing examples very possibly point at bugs in LVGL, not in the examples.  
We can ignore this fact, or fix LVGL.  
The ffmpeg issue seems to be solved by https://github.com/lvgl/lvgl/pull/4472#event-10136981801, for example.
4. How do you track "disabled" examples? Did you add to the work plan re-enabling them? Did you open issues?

kisvegabor, 2023-08-22:
I've added it back.

> These failing examples very possibly point at bugs in LVGL, not in the examples.
We can ignore this fact, or fix LVGL.

You are right, this issue should be cough by the C test, but we need more test in LVGL, for sure.

> How do you track "disabled" examples?

We can search for `TODO` and `pass` `.py` files.

commit 0b83a82a (2023-08-18): feat(multi-instance): collect all the global variables into a struct to allow running multiple instances of LVGL

Closes #4358