lvgl/lvgl #4359: [feature] lv_disp supports dynamic buffer acquisition capability

bjsylvia, 2023-07-10:
### Introduce the problem
In certain development scenarios, the rendering buffer for the LVGL display is not during initialization. Instead, there is an independent buffer queue manager in the system that handles rendering buffers. LVGL needs to actively request a rendering buffer from the manager during rendering and commit the rendered buffer back to the manager after completion. This is similar to the "dequeueBuffer" and "queueBuffer" operations in Android's buffer management.

How can LVGL be better to meet this requirement?

### Examples and cases

### Suggested solution

kisvegabor, 2023-07-12:
Hi,

I think it would be a really useful feature. Assuming that there is infinite amount of RAM with this we will never have to wait for `flush_ready` as we can always provide a new buffer. So e.g. VSYNC-ing won't slow down the system.

So how can we implement it in an elegant way? Right now the active buffer is selected [here](https://github.com/lvgl/lvgl/blob/b38c466e8d4a59eee02c7b9a2338e468632fa6f7/src/core/lv_refr.c#L926-L934) righz after flushing.

With this update we need to acquire the new buffer right before rendering, which is ok. But how to get the buffer? 
1. Use a callback like `disp->get_draw_buffer_cb` which by default just just swaps the 2 buffers . A custom callback can malloc the buffers and even call OS_sleep or OS_wait to not block the thread while it's waiting for the buffer.
2. Use an event on `disp`, e.g. `LV_EVENT_GET_DISP_DRAW_BUF`. It'd be quite the same as a callback, but seeping in an event looks a little bit odd but might work. We already have `LV_EVENT_FLUSH_FINISH` which can be used to release the buffer.

`DIRECT_MODE` with 2 buffers really require two buffers to [synchronize them after rendering](https://github.com/lvgl/lvgl/blob/b38c466e8d4a59eee02c7b9a2338e468632fa6f7/src/core/lv_refr.c#L378-L398), but it's acceptable if this mode is not working with dynamic buffer acquisition. 

What do you think?

bjsylvia, 2023-07-12:
In my opinion, the callback approach (method one) is preferable. This is because from the perspective of the end user, they only need to configure the callback by the buffer queue manager initially and don't have to concern themselves with the implementation details such as events and logic.

There are two options for DIRECT_MODE:  either to not support it at all, or to support it by enqueuing buffer information in the buffer manager to achieve similar functionality.

kisvegabor, 2023-07-13:
> In my opinion, the callback approach (method one) is preferable. This is because from the perspective of the end user, they only need to configure the callback by the buffer queue manager initially and don't have to concern themselves with the implementation details such as events and logic.

With the callback method how can we release the buffer? Shall we add a new callback or use an event for that?
The event method seems more consistent to me. The learning curve is still shallow. Especially if we consider that dynamically allocating buffer is an advanved use case.

> There are two options for DIRECT_MODE: [...]

Let's ignore it for now. It's a special case and it seems it isn't worth the extra complication now.  If there will be areal life use case we can figure out something later.

bjsylvia, 2023-07-16:
If using the callback mechanism, the dequeueing and queuing of the buffer should be implemented through callbacks. I believe that dynamically allocating buffers can also be supported in the callback mechanism.

If using the event mechanism, the requirements can also be fulfilled.

In addition, for certain scenarios,  the ability to cancel a buffer (cancelBuffer) is needed when there is no valid drawing or if the drawing fails.

kisvegabor, 2023-07-17:
After thinking about it a little, I think the event way would be better. I see these advantages:
- We are moving LVGL's in general towards events, so it will more inline with the rest of the library 
- Events are easier to extend as we just need to add a new event code. 
- Multiple modules can subscribe to the same event. It's not possible with callbacks.

I think the overhead is negligible especially compared to the rendering time.

bjsylvia, 2023-07-18:
OK, let's support this capability based on the event mechanism.

bjsylvia, 2023-07-23:
Some ideas are below:

1. shall we get the active buffer using lazy loading mode, before [here](https://github.com/lvgl/lvgl/blob/6fba2748b6c44c2480a2835edf94e7d46286605e/src/core/lv_refr.c#L368-L370). 
    * event on disp, e.g. `LV_EVENT_GET_DISP_DRAW_BUF`
    
2. If there are no updates area in the display, cancel the buffer.
    * event on disp, e.g. `LV_EVENT_CANCEL_DISP_DRAW_BUF`
    
3. After refreshing valid area in the display, release the buffer. It is not recommended to use `LV_EVENT_FLUSH_FINISH` because this event can easily cause confusion in different modes.
    * event on disp, e.g. `LV_EVENT_RELEASE_DISP_DRAW_BUF`

a `LV_EVENT_GET_DISP_DRAW_BUF` event should correspond to either a `LV_EVENT_RELEASE_DISP_DRAW_BUF` or a `LV_EVENT_CANCEL_DISP_DRAW_BUF` event. 

Additionally, when invalidating an area on the display in `FULL_MODE`, can we also send an INVALIDATE_AREA event? This not only ensures consistency in behavior across different modes but also helps the manager to perform some preparation work in advance.

kisvegabor, 2023-07-23:
I think we should get a buffer only when really there is something to draw. So `LV_EVENT_CANCEL_DISP_DRAW_BUF` shouldn't be required. 

This way `LV_EVENT_GET_DISP_DRAW_BUF` and `LV_EVENT_RELEASE_DISP_DRAW_BUF` should be enough. I'm thinking about if we can call `LV_EVENT_RELEASE_DISP_DRAW_BUF` directly from `lv_disp_flush_ready()`. It doesn't sound like a good idea as `lv_disp_flush_ready` can be called from an interrupt too, and sending events from there might mess up things. What if there is no RELEASE event, but user should release the buffer when they call `lv_disp_flush_ready()`.

> Additionally, when invalidating an area on the display in [FULL_MODE](https://github.com/lvgl/lvgl/blob/6fba2748b6c44c2480a2835edf94e7d46286605e/src/core/lv_refr.c#L274-L279) can we also send an INVALIDATE_AREA event?

Yes, we can do that.

---

I've started to work on https://github.com/lvgl/lvgl/pull/4241. I'll finish it in Monday (tomorrow) and I think it would be better wait for it, as it will change how buffers are described.

bjsylvia, 2023-07-24:
> I think we should get a buffer only when really there is something to draw. So `LV_EVENT_CANCEL_DISP_DRAW_BUF` shouldn't be required.

Yes. let us ignore 'cancel' event.

> This way `LV_EVENT_GET_DISP_DRAW_BUF` and `LV_EVENT_RELEASE_DISP_DRAW_BUF` should be enough. I'm thinking about if we can call `LV_EVENT_RELEASE_DISP_DRAW_BUF` directly from `lv_disp_flush_ready()`. [...]

 it is recommended that the events "get" and "release" should occur in pairs within the same thread.

> I've started to work on #4241. I'll finish it in Monday (tomorrow) and I think it would be better wait for it, as it will change how buffers are described.

It sounds like good news. Looking forward to seeing your updates.

kisvegabor, 2023-07-24:
In our meeting we have found that `LV_EVENT_RENDER_START` can be used instead of `LV_EVENT_GET_DISP_DRAW_BUF` and `LV_EVENT_RELEASE_DISP_DRAW_BUF` is not required as a custom function can be called directly after `lv_disp_flush_ready()`.

Whit this in mind, I think the current features of LVGL can be used for dynamic buffer acquisition. 

The draw buffer should be stored in `disp->draw_buf_act` and by default it should be set to `NULL`. We need to check if there any issues if `disp->draw_buf_act == NULL`. 

I'll include a few things in #4241 to make this simpler.

closed 2023-08-16 by lvgl-bot

kisvegabor, 2023-08-16:
Finally did the event based approach working?

bjsylvia, 2023-08-19:
Next week, I will try it based on `multi-instance` branch and give you feedback.

bjsylvia, 2023-08-27:
@kisvegabor  The solution has been verified and can meet the requirements.

kisvegabor, 2023-08-27:
Wonderful, thank you for the feedback! :slightly_smiling_face: