lvgl/lvgl #4361: [feature] LVGL style efficiency optimization

terry0012, 2023-07-11:
### Introduce the problem

There are two efficiency issues with LVGL style queries. One is the sequential search of obj styles, and the other is the recursive query of parent-child styles

### Examples and cases

### Suggested solution

  To improve search efficiency of obj styles,  we can put frequently used props in the front of the prop array, while non-frequently used props are put in the back.
  we can add a new flag in meta data to indicate the prop is a frequently used or not.

kisvegabor, 2023-07-12:
It'd be very important to improve this indeed.

> One is the sequential search of obj styles

There are 2 things here:
1. Find the relevant styles based on `state` and `part`. 
2. Find the styles property in the relevant styles

For 1) we can group the styles related to the same `state` and `part` in to a list. E.g.
```c
typedef struct {
    uint32_t selector:24; //Part + state
    uint8_t style_cnt; //Number of styles in the `style_list`
    lv_style_t * style_list; //Dynamically allocated array to store the styles.
} lv_style_part_state_t;  //This type doesn't exist now

typedef struct {
  ...
  lv_style_part_state_t * style_part_state_list;
  uint8_t style_part_state_cnt;
  ...
}lv_obj_t;
``` 

We can store `style_part_state_list`s in a specific order. E.g. same parts together starting with the `MAIN` part. This way if we see a non-MAIN part for the first time, we can finish searching.

----

For 2) I think simply storing the most common properties at the beginning of the list won't help much. It's because let's say we look for  the `text_opa` property. We still need to check all relevant styles and all properties, and in the end we might conclude that this property is not set, so we shall use the default value. Most of the `style_property_get` calls end up in a "miss" because most of the properties are used with the default value. So I think we should this fact and find a way to quickly decide if we need to search for a property or not.
 
My idea is to extend `lv_style_part_state_t` from above like this:
```c
typedef struct {
   //For most common properties
    uint32_t prop_is_set;
    //means 
    //uint32_t bg_opa_is_set   :1;
    //uint32_t text_opa_is_set  :1;
    //uint32_t text_font_is_set  :1;
    //uint32_t clip_corner_is_set  :1;
    //etc, for the 32 most common properties
           
    uint32_t selector:24; //Part + state
    uint8_t style_cnt; //Number of styles in the `style_list`
    lv_style_t * style_list; //Dynamically allocated array to store the styles.
} lv_style_part_state_t;`
```

We need to set the actual value of the style properties to match the bit position of the property in `prop_is_set`. E.g.  for the above example
```
LV_STYLE_BG_OPA         0
LV_STYLE_TEXT_OPA     1
LV_STYLE_TEXT_FONT   2
LV_STYLE_CLIP_CORNER_FONT   3
...
```

And when we get a property (pseudo code):
```c
for(i = 0; i < obj->style_part_state_cnt; i++) {
  lv_style_part_state_t * sp = obj->style_part_state_cnt[i]
  if(is_relevant(sp->selector) {
     if(style_prop >= 32 || (sp->prop_is_set & (1 << style_prop))) {
         for(j = 0; j < sp->style_cnt; j++) {
               found = find_prop_in_style(sp->style[j], style_prop, &result);
               if(found) return result;
         }
     }
  }
}
```

We can do the same `prop_is_set` in each `lv_style_t` as well, not only in `lv_style_part_state_t`. Or only in styles. 

We can also store the style properties in an ascending order an apply logarithmic search is there are let's say > 16 properties.

----

> the other is the recursive query of parent-child styles

I think the above mentioned things should highly improve this as well. However I was thinking about dropping the style property inheritance  feature too. What we would lose is mainly related to the text properties. So we need to set the color of a label on a button explicitly, instead of setting the label color on the button. This we will need more styles, on the label widgets, which will use slightly more RAM. ~8..24  extra byte per label, I guess. Removing inheritance would break CSS compatibility, which is not a good thing though.

---

I think the `prop_is_set` thing used in `lv_style_part_state_t` and/or `lv_style_t` would have a  huge impact, but we need to benchmark all these and make experiments. 

What do you think?

terry0012, 2023-07-17:
For 1)  
should we record the count of each  part in style_part_state_list?
E.g.  If same parts together starting as the definition of lv_part_t,   when we are searching a LV_PART_ITEMS part, we can quickly get the start index of LV_PART_ITEMS  part position.

    LV_PART_MAIN         = 0x000000,   /**< A background like rectangle*/
    LV_PART_SCROLLBAR    = 0x010000,   /**< The scrollbar(s)*/
    LV_PART_INDICATOR    = 0x020000,   /**< Indicator, e.g. for slider, bar, switch, or the tick box of the checkbox*/
    LV_PART_KNOB         = 0x030000,   /**< Like handle to grab to adjust the value*/
    LV_PART_SELECTED     = 0x040000,   /**< Indicate the currently selected option or section*/
    LV_PART_ITEMS        = 0x050000,   /**< Used if the widget has multiple similar elements (e.g. table cells)*/
    LV_PART_TICKS        = 0x060000,   /**< Ticks on scale e.g. for a chart or meter*/
    LV_PART_CURSOR       = 0x070000,   /**< Mark a specific place e.g. for text area's cursor or on a chart*/
    LV_PART_CUSTOM_FIRST = 0x080000,    /**< Extension point for custom widgets*/
    LV_PART_ANY          = 0x0F0000,    /**< Special value can be used in some functions to target all parts*/

For 2)

I think we can quickly get the position of the attribute based on the variable 'prop_is_set', if we also store the most common 32 attributes in the head.

    for(i = 0; i < obj->style_part_state_cnt; i++) {
        lv_style_part_state_t * sp = obj->style_part_state_cnt[i]
        if(is_relevant(sp->selector) {
            int count = __builtin_popcountl(sp->prop_is_set);
    
            if(stryle_prop < 32) {
                if( 1 << style_prop) {
                    for(j = 0; j < count ; j++) {
                        found = find_prop_in_style(sp->style[j], style_prop, &result);
                        if(found) return result;
                    }
                } 
            } else {
                for(j = count; j < sp->style_cnt; j++) {
                    found = find_prop_in_style(sp->style[j], style_prop, &result);
                    if(found) return result;
                }
            }
        }
    }

> I think the above mentioned things should highly improve this as well. However I was thinking about dropping the style property inheritance feature too. What we would lose is mainly related to the [...]

Drop the inheritance feature of style property also have a lot impact on the existing applications.

kisvegabor, 2023-07-28:
I'm thinking about approaching getting style properties via hash tables like this
```
index = hash(property_id)
return values[index]
```

In the most simple case can use the `property_id` directly:
```
return values[property_id]
``` 

But it in case of max 150 style propeties it means 150 x 4 bytes = 600 byte per style which is obviously not acceptable.

Ideally we need to have a magical hash function like this:
- let's say we have this `property_id`s in a style: 16, 35, 48, 77 (e.g. bg color, border width, text font, text color)
- the hash function needs to convert  
```
16 -> 0
35 -> 1
48 -> 2
77 -> 3
any others -> -1 or other invalid value
``` 
- however in an other style we will have different style properties, e.g. 23, 27 and the hash function should convert
```
23 -> 0
27 -> 1
```
- so the hash function needs to be dynamic somehow by using a seed or something like 
`index = hash(seed_for_this_style, property_id)`

I'm also not sure if it will be faster. Typically we have < 5 style properties in a style. Even a simple linear search should be very fast on such a small array. With `popcount` trick it should be event better. 

Knowing if we should skip a style or not  (because we know if the property is there or not, or the part - state doesn't match) might matter much more then searching the styles.

I'm not sure this magic is possible, I just wanted to bring it up, maybe it gives some ideas for you.

kisvegabor, 2023-07-29:
Something more realistic and related thing just came to my mind. There are a few issues with the current theme system:
- all theme styles are added to widgets which
   - uses extra RAM
   - hard to swap theme styles
 - `&lv_<widget_name>_class` is [used](https://github.com/lvgl/lvgl/blob/db0dbee435677292cfb81871f054c476d384487e/src/themes/default/lv_theme_default.c#L750) in the themes which makes it impossible for the linker to drop the unused widgets
- the styles of the theme uses a lot of RAM
- it's hard to add new styled components theme, i.e. extend the theme

What I recommend is using a `widget_type_id` for each widget. It just replaces `&lv_<widget_name>_class`.  E.g. `#define LV_BTN_ID 25` in `lv_btn.h`. 

We don't add the theme styles to the widgets, but when `lv_obj_get_style_prop` finds that the property doesn't exists in widget's styles it asks the theme's callback by passing the widget's pointer, selector, and property to get.

Besides saving some RAM it opens the door for amazing things:
- return different value based on some properties of the widget, its parent, or its children, etc. For example making every second child darker (something like `:nth-child(odd)`) can be easily realized by 
```c
return (lv_obj_get_index(obj) % 2) ? white : gray;
``` 
or adjust the radius based on the size of the widget (we could do it by having `lv_pct` support for radius too though)
```c
return lv_obj_get_width(obj) / 20
```
- Handling dark/light theme switch is super easy:
```c
if(dark) { search in dark styles }
else { search in light styles }
```
- Styles can be organized anyhow to optimize performance. As with this suggestion widgets will have only a few styles we can add the whole `lv_style_part_state_t` trick only to the theme implementation. It might seem that we just pushed the problem to the theme from the widgets, but in the themes we have much more flexibility to handle styles. All styles would be stored only on one place so having larger look up tables or hash maps is way more feasible.
- We can make it very simple to add new "components" to the themes
```c
lv_theme_extend(theme, callback); //Add the callback to theme's callback linked list 

bool callback(lv_theme_t * theme, lv_obj_t * obj, lv_style_prop_t prop, lv_style_selector_t selector, lv_style_value_t * v)
{
   if(obj->widget_type_id == CUSTOM_BUTTON_ID) {
      if(selector == LV_PART_MAIN | LV_STATE_PRESSED) {
         if(prop == LV_STYLE_BG_COLOR) {
              v->color = lv_color_white();
              return true;
          }
      } else if(selector == LV_PART_SCROLLBAR | LV_STATE_SCROLLED) {
          if(find_in_style(&some_style, prop, v) return true;
          else return false; 
     }
  }
  return false;
}
```
We can provide some helpers to integrate new components effectively into the theme, but I think you get the idea. Maybe EACH `widget_type_id` can have it's own callback list to speed things up.

If you agree with this direction, I recommend:
- leaving `lv_obj_style_t * styles` in `lv_obj_t` as it is (i.e. do not add `lv_style_part_state_t`)
- store the styles in `lv_obj_style_t * styles` in `lv_obj_t` sorted by part to early skip some iterations 
- adding the `prop_is_set` to each style
- rework `lv_theme_default` and apply all style organization and optimization magic there

kisvegabor, 2023-07-30:
I think I've found something simple yet effective. 

 The widgets usually have 0...5 styles (it's only a few iterations to check) and each style has 5-10 properties (not that long either). However LVGL tries to get a style property ~3000 times from widgets on one full refresh in the demo widget. Even if getting the properties would be support optimal starting something with 3000 times has a lot of overhead. 

And the important thing is that  86% of the style property get calls are misses. It means 7 out of 8 [`get_prop_core()`](https://github.com/lvgl/lvgl/blob/db0dbee435677292cfb81871f054c476d384487e/src/core/lv_obj_style.c#L297) (called in `lv_obj_get_style_...()` to get a style property from a widget)  call will use either the default value or if the property is inheritable, check the parent widget.

If we could immediately decide if the widget has a property or not, the whole process would be a lot faster. 

This is what I recommend: 
- With the new theme idea that I described above, we can move the majority of the styles to the theme. So these styles are stored only once and on each widget insatnce. 
- In the theme we store the styles in a [`_lv_obj_style_t`](https://github.com/lvgl/lvgl/blob/master/src/core/lv_obj_style.h#L52-L57) and also store a bit field of 256 bits for each property. If a property is set in any state or part in the given style list set the bit of the property. 
- When getting a style property, immediately check the bit of the property and return "no found" if the property is not there. 
- This 256 bits means 16 extra byte per widget type (not widget instance), so not a big deal. Besides the styles of the theme are kind of static (styles are not added dynamically) so invalidating these bit fields is not big issue.

---

Regarding this new theme idea:
We should also support cascading components. For example "Red button" should be based on the "Norma button" widget. For this we just need  `new_component_id = lv_theme_add_component(base_id, callback)`. When the style of a widget with `new_component_id` is being get and the property is not found we should search in `base_id` too. Of course we need to be careful as this "component inheritance"  also affect performance. However as each component aggregates 5-10 styles there are much less iterations. 

I really like this concept as it seems much more flexible, allows great new features, and has a lot of optimization opportunities. Unless there is no objection, I'll try it out next week.

kisvegabor, 2023-08-08:
After a lot of testing and benchmarking  I could do the followings with the current theme concept (add the theme styles to the widgets)

On an STM32F769 (200 MHz, 800x480) rendering the first page of the  widgets demo 1000 times, the style related functions calls needed 24% of the CPU time.
By adding
- `uint32_t prop_set_for_main_part` to `lv_obj_t *` to save if a proeprty is set for the `MAIN` part. As it has 32 bits but we have 128 style properties, each bit means a group of 4 properties
- similarly adding `uint32_t prop_set_for_other_part` for other than `MAIN` part properties.
- similarly `uint32_t prop_set` in `lv_style_t`
- removed the `INHERIT` and `INITIAL` style properties (not the inheritance behavior, just the "inherit" value)
- optimized the loops and functions, inlined functions, removed function calls, etc 

And the result is 9%, so 15% better. The original 30ms (33FPS) rendering time was changed to 25ms (40FPS).

I tried to lower the number of style property gets, but it changed only 0.2%. 

Next I'll add the new theme idea. I've already made some experiments but I'm not happy with the result so far. 

Anyway, I think this < 10% is already ok. <5% would be better of course, but that 5% change would mean only 1.5ms from the 30ms rendering time. We will see.

kisvegabor, 2023-08-15:
The changes are merged to master. The theme improvement is not that critical so we can do it later. 

You can enable style property caching with `LV_OBJ_STYLE_CACHE`.

I close this issue now, please comment here if you find any issues.

closed 2023-08-15 by kisvegabor