lvgl/lvgl #4366: Consider adding LV_ OBJ_ FLAG_ FOCUSABLE and LV_ OBJ_ FLAG_ LONG_ CLICKABLE

incity, 2023-07-12:
In some cases, it is necessary to use directional keys (similar to those on a TV remote control or keyboard) to locate interface elements,

kisvegabor, 2023-07-13:
I don't get it.  What would be point of making a widget non long press clickable? You can just not react on the `LV_EVENT_LONG_PRESS` event.

incity, 2023-07-25:
if an object has LV_ OBJ_ FLAG_ LONG_ CLICKABLE flag, the lv_indev can send LV_EVENT_LONG_CLICKED. I think this is more better than now: send a LV_EVENT_SHORT_CLICKED if did not long press and always send LV_EVENT_CLICKED.
```
        /*Send RELEASE Call the ancestor's event handler and event*/
        lv_obj_send_event(indev_obj_act, LV_EVENT_RELEASED, indev_act);
        if(indev_reset_check(indev)) return;

        /*Send CLICK if no scrolling*/
        if(scroll_obj == NULL) {
            if(indev->long_pr_sent == 0) {
                lv_obj_send_event(indev_obj_act, LV_EVENT_SHORT_CLICKED, indev_act);
                if(indev_reset_check(indev)) return;
            }

            lv_obj_send_event(indev_obj_act, LV_EVENT_CLICKED, indev_act);
            if(indev_reset_check(indev)) return;
        }
        else {
            lv_obj_send_event(scroll_obj, LV_EVENT_SCROLL_THROW_BEGIN, indev_act);
            if(indev_reset_check(indev)) return;
        }
```
why not like THIS:
```
        /*Send RELEASE Call the ancestor's event handler and event*/
        lv_obj_send_event(indev_obj_act, LV_EVENT_RELEASED, indev_act);
        if(indev_reset_check(indev)) return;

        /*Send CLICK if no scrolling*/
        if(scroll_obj == NULL) {
            if(indev->long_pr_sent == 0) {
                lv_obj_send_event(indev_obj_act, LV_EVENT_CLICKED, indev_act);
                if(indev_reset_check(indev)) return;
            } else {
                if (lv_obj_has_flag(indev_obj_act, LV_OBJ_FLAG_LONG_CLICKABLE)) {
                    lv_obj_send_event(indev_obj_act, LV_EVENT_LONG_CLICKED, indev_act);
                    if(indev_reset_check(indev)) return;
                }
            }
        }
```

clicked always means short click. and long click means long click.

incity, 2023-07-25:
also for LV_OBJ_FLAG_CLICKABLE and LV_OBJ_FLAG_CLICK_FOCUSABLE.
why not just use LV_OBJ_FLAG_CLICKABLE  and LV_OBJ_FLAG_FOCUSABLE. if an object has LV_OBJ_FLAG_CLICKABLE and LV_OBJ_FLAG_FOCUSABLE, then the object can be focused when clicked.

kisvegabor, 2023-07-25:
I think we can add `LV_EVENT_LONG_CLICKED` without any issues and keep `LV_EVENT_CLICKED` and `LV_EVENT_SHORT_CLICKED` too.

Regarding `LV_OBJ_FLAG_FOCUSABLE` I tink we should unify it with `LV_OBJ_FLAG_CLICK_FOCUSABLE` somehow.

incity, 2023-08-01:
LV_ EVENT_ LONG_ CLICKED is required. Sometimes we need to simulate (short) clicked and long clicked events.

closed 2023-08-24 by lvgl-bot