lvgl/lvgl #4368: I am porting LVGL to K230 which has a VGLite GPU same as NXP, can I reuse the NXP code?

MrThanlon, 2023-07-13:
### Introduce the problem

The GPU of the K230 runs on its little core and operates on the Linux operating system. This means that user-level programs use virtual addresses, while the GPU uses physical addresses or so called bus addresses and requires memory mapping for access. However, the VGLite draw ctx provided by NXP directly sends the buffer as a virtual address to the GPU, which does not work properly on the K230. I'm wondering how to reuse this part of the code? Should I rewrite it?

Here is [k230_sdk](https://github.com/kendryte/k230_sdk)

### Examples and cases

### Suggested solution

kisvegabor, 2023-07-14:
I haven't used Kendryte chips so far so can't really comment on the details. However what I can tell is that VGLite and PXP GPU are meant to run only on NXP devices and they are not generic solutions.

If you already know where to touch the drivers I encourage you to make some experiments and please report back to us.

MrThanlon, 2023-07-15:
I tried the full_refresh mode and directly exported the DRM buffer as a DMA-buf to be drawn by VGLite and LVGL. However, the result was too slow, achieving only about 3 FPS at a resolution of 1080x1920. The profiling showed that most of the time was spent on drawing lines. VGLite GPU doesn't directly support line drawing and instead simulates lines with polygons, which is not very efficient. I attempted to switch to partial refresh mode, but it seemed to pose some challenges when using double buffering.

kisvegabor, 2023-07-15:
If software rendered line rendering is faster feel free comment out that part from VG-Lite.

PeterBee97, 2023-07-17:
@

> ### Introduce the problem
> The GPU of the K230 runs on its little core and operates on the Linux operating system. This means that user-level programs use virtual addresses, while the GPU uses [...]

Hi @MrThanlon ,
I have some experience with LVGL on VGLite GPU and I am intererested in contributing a better driver to the community, however I wasn't able to buy NXP RT595 or RT1170 in China. I see that the K230 chip is still not quite available on market. Maybe I can help you if I have access to the board? You may add my WeChat: ptr_b_

MrThanlon, 2023-07-18:
VGLite GPU requires the width of the drawing buffer to be aligned to 16. Is there a way to force LVGL to use a drawing buffer with a width that is a multiple of 16? I noticed the LV_ATTRIBUTE_MEM_ALIGN_SIZE in lv_conf.h, but it doesn't seem to work.

PeterBee97, 2023-07-18:
> VGLite GPU requires the width of the drawing buffer to be aligned to 16. Is there a way to force LVGL to use a drawing buffer with a width that is a multiple of 16? I noticed the LV_ATTRIBUTE_MEM_ALIGN_SIZE in lv_conf.h, but it doesn't seem to work.

@MrThanlon Short answer no, as we don't want to force a width but want to add stride info to the image. For more discussion see #4241 .

closed 2023-08-09 by lvgl-bot

XuNeo, 2023-11-15:
Following up discussion here: https://github.com/lvgl/lvgl/issues/4813