lvgl/lvgl #4370: Optimizer prevents font rendering

the-programmer, 2023-07-13:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) (not possible, platform/compiler specific)

### Describe the bug
In certain compilers the following check gets optimized out when using -Os.
https://github.com/lvgl/lvgl/blob/29ed7c5717f6cfbca5ce888bf4497221525aae85/src/font/lv_font_fmt_txt.c#L152-L155
This gets converted to (at least)
```c
unicode_letter = ' ';
``` 
The proof for this can be found at the following link.
https://forum.lvgl.io/t/no-text-on-buttons-labels/12287/27?u=theprogram01

### To Reproduce
Have a specific compiler version `(Sourcery CodeBench Lite 2013.05-23) 4.7.3` and set the optimization to -Os. In that case the call to `get_glyph_dsc_id()` will contain 32 as `letter`.
Note: don't add logging to 'lv_font_get_glyph_dsc_fmt_txt()' since this will tell the optimizer that it is important and will also "fix" the issue.

### Expected behavior
That I can see my text

### Screenshots or video
<see topic>

### Possible solution
If I update the code to the following it works as intended.
```c
bool lv_font_get_glyph_dsc_fmt_txt(const lv_font_t * font, lv_font_glyph_dsc_t * dsc_out, uint32_t unicode_letter,
                                   uint32_t unicode_letter_next)
{
    volatile uint32_t temp_unicode_letter = unicode_letter;
    volatile bool is_tab = false;
    if(temp_unicode_letter == '\t') {
        unicode_letter = ' ';
        is_tab = true;
    }
    lv_font_fmt_txt_dsc_t * fdsc = (lv_font_fmt_txt_dsc_t *)font->dsc;
```
(has been done [before](https://github.com/lvgl/lvgl/commit/388a5d7f975a91eb964e419e8d3ab81fd8a306c5))

kisvegabor, 2023-07-14:
Wow, really strange. Two similar issues in a week. 

On the forum I saw that your compiler is 10 years old, so the font issue is probably a compiler bug. I found the [other optimizer issue](https://github.com/lvgl/lvgl/commit/388a5d7f975a91eb964e419e8d3ab81fd8a306c5) in an STM project which is using: 
```
arm-none-eabi-gcc (GNU Tools for STM32 10.3-2021.10.20211105-1100) 10.3.1 20210824 (release)
```
So it's not that old. 

---

Anyway adding volatile here should affect the performance much but it's hacky. :slightly_frowning_face: 

cc @pete-pjb @FASTSHIFT, what do you think, what would be the best way to workaround this? Add the `volatile` keyword as @the-programmer suggested? Ignore this issue as the compiler is too old? Anything else?

the-programmer, 2023-07-14:
Just to verify, I assumed you meant "shouldn't affect the performance" but you typed "should".

kisvegabor, 2023-07-14:
Oh sorry, yes I meant shouldn't. Updated :slightly_smiling_face:

kisvegabor, 2023-07-14:
BTW, what is the use-case in which you have a 10 years old compiler? I'm just asking to better understand in what kind of environments LVGL is being used.

the-programmer, 2023-07-14:
The manufacturer of the device delivers that specific compiler with it's toolchain. If you want I can send you more details (but that is something I want to keep as "private" as possible). (A DM on the form is fine for me)

Also, currently the code is running out of external ram, so the performance is horrible. I am currently working with the manufacturer to get LVGL to the flash level (so we can have a better performance).

kisvegabor, 2023-07-15:
Sure, please send a DM on the Forum.

the-programmer, 2023-07-19:
My suggestion would be
``` c 
//Check if we have a 'tab', if so we only need a visual space from the font
bool is_tab = unicode_letter == '\t';
if(is_tab) {
    unicode_letter = ' ';
}

The 2nd part is what the compiler/optimizer made out of it (but not what would suggest).

kisvegabor, 2023-07-19:
Got it, thank you. 

Could you send a Pull request with:
```c
bool is_tab = unicode_letter == '\t';
if(is_tab) {
    unicode_letter = ' ';
}
```

pete-pjb, 2023-07-19:
Hi @kisvegabor & @the-programmer ,

I think this is a good solution. Do you think a comment above the code change to explain why it is written in that way would be good for a reminder in the future?

Kind Regards,

Pete

the-programmer, 2023-07-20:
@kisvegabor PR Created

@pete-pjb, To be honest I can't think of a correct comment for this.

the-programmer, 2023-07-21:
Just tested the with the latest version of LVGL 8.3 and the code works.

Issue closed.

closed 2023-07-21 by the-programmer

kisvegabor, 2023-07-23:
:+1: 

> @pete-pjb, To be honest I can't think of a correct comment for this.

Finally we added a comment too.

PR #4385 (2023-07-21): Update lv_font_fmt_txt.c
Fixes  #4370, a possible optimizer bug that prevents font rendering.

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if [... cut]