lvgl/lvgl #4372: label default text

fuxiujun, 2023-07-13:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The "Text" text is created by default in the lv_label, but I don't need it because reading fonts is a time-consuming operation

### To Reproduce

```c
    lv_obj_t* label = lv_label_create(NULL);
    lv_obj_set_style_text_font(label, font, 0);
    lv_label_set_text(label, "abc");
```

### Expected behavior

```c
static void lv_label_constructor(const lv_obj_class_t * class_p, lv_obj_t * obj)
{
....
    lv_obj_clear_flag(obj, LV_OBJ_FLAG_CLICKABLE);
    lv_label_set_long_mode(obj, LV_LABEL_LONG_WRAP);
    lv_label_set_text(obj, ""); 
}
```

### Screenshots or video

kisvegabor, 2023-07-14:
I think it should be a generic option which affects multiple widgets that have default value. For example: roller, drop down, button matrix, etc. 

We could call it `LV_WIDGETS_HAS_DEAFULT_VALUE` and set it to `1` default. 

What do you think?

fuxiujun, 2023-07-16:
Thanks for your reply, I think it makes more sense and solves the problem I'm having. I have been trying to modify it like this.

kisvegabor, 2023-07-17:
Let continue in #4371.

closed 2023-07-17 by kisvegabor