lvgl/lvgl #4376: Sets the default focused child component when scrolling is used

BestStark, 2023-07-14:
### Introduce the problem

When I use the scroll function, I set to allow horizontal scrolling. There are panel 1, panel 2, and panel 3 respectively. By default, panel 1 will be focused, but I want to set panel 2 as the default focus. Every time I enter this interface, I will see it first. It is panel 2, slide left and right to see panel 1 and panel 3 respectively

### Examples and cases

[image]

### Suggested solution

Sorry, I didn't think of a good way, but I want to achieve this function and hope to get some suggestions or discussions,thanks.

BestStark, 2023-07-14:
oh, `lv_obj_scroll_to_view(lv_obj_get_child(parent, 0), LV_ANIM_OFF); ` This method can. thanks

BestStark, 2023-07-14:
But I found a strange problem. For example, I have three panels in the scrolling area, and one of the panels has a button. **When I click the button**, there will be a msgbox. When I click yes or no, the current entire scrolling area is activated after the event is triggered. Refresh, it will return to the last panel, what is the reason?

kisvegabor, 2023-07-14:
> When I click yes or no, the current entire scrolling area is activated after the event is triggered. Refresh, it will return to the last panel, what is the reason?

I believe you have  set a default group with `lv_group_set_default`. Due to this all interactive elements (e.g. buttons) are automatically added to default group (so you can control them e.g. with a keyboard too).

When you delete the message box LVGL looks for a new widget to focus and it finds the last created widget, which is the last button. 

You can solve it by
- removing `lv_group_set_default`
- manually selecting which widget to focus when the message box is closed

BestStark, 2023-07-17:
> I believe you have set a default group with `lv_group_set_default`. Due to this all interactive elements (e.g. buttons) are automatically added to default group (so you can control them e.g. with a [...]

oh , yes !! 

Thank you very much for your help. It was very useful to me 😄

kisvegabor, 2023-07-17:
Happy to help! :slightly_smiling_face:

closed 2023-07-17 by kisvegabor