lvgl/lvgl #4378: The Chinese text of lv_label_set_text has the problem of inaccurate display or garbled code~

daironxuan, 2023-07-15:
### The test code is as follows  Label_Test.c
```c++

lv_obj_t * cz_label = lv_label_create(lv_scr_act());
    lv_label_set_text(cz_label,
                      "嵌入式系统（Embedded System），\n是一种嵌入机械或电气系统内部、具有专一功能和实时计算性能的计算机系统。");
    lv_obj_set_style_text_font(cz_label, &lv_font_simsun_16_cjk, 0);
    lv_obj_set_width(cz_label, 310);
    lv_obj_align(cz_label, LV_ALIGN_BOTTOM_LEFT, 5, -5);
```
### Examples and cases

### Here's my lv_conf.h, which looks like this:

```c++
#define LV_FONT_DEFAULT &lv_font_montserrat_14
...
#define LV_FONT_MONTSERRAT_12_SUBPX      0
#define LV_FONT_MONTSERRAT_28_COMPRESSED 0
#define LV_FONT_DEJAVU_16_PERSIAN_HEBREW 1
#define LV_FONT_SIMSUN_16_CJK            1
....
#define LV_FONT_CUSTOM_DECLARE LV_FONT_DECLARE(my_Chinese_font)
#define LV_FONT_DEFAULT &lv_font_montserrat_14
```

### My questions are as follows：

<h3>My hardware is STM32_F407ZGT6.

Touch screen: TTFLCD 2.8(width 240, height 320)

The compiler is: arm-none-ebi-gcc

After burning, display: □□□□□□□□..

Is there a good solution?</h3>

daironxuan, 2023-07-15:
**_Mine is: v8.3.8-release_**

daironxuan, 2023-07-15:
**_My version is: v8.3.8-release_**

daironxuan, 2023-07-15:
**_Yes, he has no problems on the PC, but my STM32F407ZGT6 will have problems~_**

kisvegabor, 2023-07-15:
Are you sure the IDE has UTF-8 encoding? 

You can check it like [this](https://github.com/lvgl/lvgl/blob/b466045f1396b96824c2b42f791d9dc33492cc2b/src/lv_init.c#L114-L119).

daironxuan, 2023-07-15:
**_ Yes, it did solve my problem ~_**

_** Or you can convert the Font File to UTF-8 in Clion IDE **_

如下：
[image]

kisvegabor, 2023-07-15:
Great!

closed 2023-07-15 by kisvegabor