lvgl/lvgl #4381: Need someone to test lv_img using the current master branch

kdschlosser, 2023-07-16:
I am having an issue with the CPython binding that I have been tearing my hair out over for the last few days. I am starting to wonder if it is not an issue with something I am doing but possibly something not functioning properly in LVGL.

The issue I am having is I create an lv_img_dsc_t structure. I place into the `data` field the data from a PNG file. I set the `data_size` field to the length of the data. I then cast the structure if a `void` pointer and use the `lv_img_set_src` function passing the pointer to the function.

Inside the function one of the first thing done is a call is made to `lv_img_src_get_type` to get the type of data that has been passed in the pointer.  Inside the `lv_img_src_get_type` function the first thing that is done is a check to make sure the passed pointer is not `NULL`, It passes this check. The next thing that is done is the pointer is cast to a uint8_t pointer and the first byte is evaluated to determine the type. This reports back as being variable which is a `0` which is what the function returns. 

Back in the `lv_img_set_src` function the next thing that gets done is the header for `lv_img_dsc_t` gets populated. This is done via the `lv_img_decoder_get_info` function. Inside that function the pointer is once again tested to make sure it is not `NULL` which it passes The data type is gotten again and zero is returned. The pointer is then cast to an `lv_img_dsc_t` structure and the data in the structure gets tested to see if it is NULL and that is where it fails. I am able to check the data in the original structure and it checks out, it is there and it is not corrupt or anything like that. I am unsure of why the data is showing up as a NULL pointer.

For the sake of my sanity I would like someone that has the ability to test `lv_img` in the same capacity as above to see if it is working properly. If it works I can check that off the list as to a possible cause. If it doesn't work then we will have to figure out why it is not working. 

Thanks in advance for any help that is given.

kisvegabor, 2023-07-17:
Please copy how you initialize the `lv_img_dsc_t` in Python.

kdschlosser, 2023-07-18:
Here is the Structure in Python

```python
class img_header_t(ctypes.Structure): 
    def __init__(
        self, 
        cf: Optional["uint32_t"] = _DefaultArg, 
        always_zero: Optional["uint32_t"] = _DefaultArg, 
        reserved: Optional["uint32_t"] = _DefaultArg, 
        chroma_keyed: Optional["uint32_t"] = _DefaultArg, 
        w: Optional["uint32_t"] = _DefaultArg, 
        h: Optional["uint32_t"] = _DefaultArg
    ):
        ctypes.Structure.__init__(self)

        kwargs = {}
        if cf != _DefaultArg:
            kwargs['cf'] = cf
        if always_zero != _DefaultArg:
            kwargs['always_zero'] = always_zero
        if reserved != _DefaultArg:
            kwargs['reserved'] = reserved
        if chroma_keyed != _DefaultArg:
            kwargs['chroma_keyed'] = chroma_keyed
        if w != _DefaultArg:
            kwargs['w'] = w
        if h != _DefaultArg:
            kwargs['h'] = h

        for key, value in kwargs.items():
            setattr(self, key, value)

class img_dsc_t(ctypes.Structure): 
    
    def __init__(
        self, 
        header: Optional["img_header_t"] = _DefaultArg, 
        data_size: Optional["uint32_t"] = _DefaultArg, 
        data: Optional["uint8_t"] = _DefaultArg
    ):
        ctypes.Structure.__init__(self)

        kwargs = {}
        if header != _DefaultArg:
            kwargs['header'] = header
        if data_size != _DefaultArg:
            kwargs['data_size'] = data_size
        if data != _DefaultArg:
            kwargs['data'] = data

        for key, value in kwargs.items():
            setattr(self, key, value)

img_header_t._fields_ = [
    ('cf', uint32_t),
    ('always_zero', uint32_t),
    ('reserved', uint32_t),
    ('chroma_keyed', uint32_t),
    ('w', uint32_t),
    ('h', uint32_t)
]

img_dsc_t._fields_ = [
    ('header', img_header_t),
    ('data_size', uint32_t),
    ('data', _ctypes.POINTER(uint8_t))
]

```

Here is the `img_set_src` function

```python
_lib_lvgl.lv_img_set_src.argtypes = [_ctypes.POINTER(obj_t), src=void_t]
_lib_lvgl.lv_img_set_src.restype = None

def img_set_src(obj: obj_t, src: Any) -> None:
    obj = _convert_to_ctype(obj, _lib_lvgl.py_lv_img_set_src.argtypes[0])
    src = _convert_to_ctype(src, _lib_lvgl.py_lv_img_set_src.argtypes[1])

    _lib_lvgl.py_lv_img_set_src(obj, src)
```

The `_convert_to_ctype` function converts the python objects to the ctype it needs to be. in the case of the image widget or "obj" the passed in variable which is not a pointer because of how python works gets turned into a pointer using `ctypes.byref`. The passed in "src" parameter if it is a byte string or a bytearray it gets turned into a uint8_t array and then the address of that array is what gets passed back. If the "src" is an `lv_img_dsc_t` structure then the address is gotten using `ctypes.addressof` and that is returned. If the passed in src is a string it gets converted to bytes and 0x00 added onto the end and then it gets put into a char_t array and the address to that array gets passed to the function.

I am able to take the structure and convert it to a void pointer and then convert it back to the lv_img_dsc_t structure and the data is there and it is accessible but for whatever reason it is not getting converted properly on the C code side of things. I am not sure if it is because of the `const` or maybe because the cat is not really explicit.  What I mean by not being explicit is the cast is done like this. `const lv_img_dsc_t *dsc = src;` and this would be explicit. `const lv_img_dsc_t *dsc = (const lv_img_dsc_t *)src;` There are other places in LVGL where the cast is done explicitly and it is also done without the `const`.

I do know that a pointer that gets passed as a parameter for the purposes of being an "out" the parameter type needs to be a pointer to a pointer of the type. This is a problem with the file system driver and reading the file. The callback function has a parameter that needs to be filled with the data and I am not able to fill it because the type is not a pointer to a pointer. There is also a goofy issue that only allows me to return primitives from callback functions. no structures or unions or arrays can be returned. There is only one callback where there would be an issue.

How I am constructing the structure is as follows.

```python

with open('some file.png', 'rb') as f:
    img_data = f.read()

img_dsc = lv.img_dsc_t()
img_dsc.data = (lv.uint8_t * len(img_data))(*img_data)
img_dsc.data_size = len(img_data)

img = lv.img_create(lv.scr_act())

lv.img_set_src(img, img_dsc)
```

the function call ends up like this.
```python 
lv.img_set_src(ctypes.byref(img), lv.void_t(ctypes.addressof(img_dsc)))
```

If I do this the data is accessible

````python

vp = lv.void_t(ctypes.addressof(img_dsc))

new_img_dsc = ctypes.cast(vp, ctypes.POINTER(lv.img_dsc_t)).contents
test_array= new_img_dsc.data[:new_img_dsc.data_size]
test_data = bytes(bytearray([item.value for item in test_array]))

print(test_data == img_data)
```

and that outputs `True`. So the structure is being cast to a void pointer properly and it is able to be reverse back into an `img_dsc_t` structure and the data is valid inside of it.

kisvegabor, 2023-07-19:
I can't see where you set the `w`, `h`, `cf` and ``always_zero` in `new_img_dsc.header`. Am I missing something?

kdschlosser, 2023-07-19:
I am setting PNG data into the data field.

kisvegabor, 2023-07-19:
I've just tested this in C and it really works:
```c

const lv_img_dsc_t img_wink_png = {
//    .header.always_zero = 0,
//    .header.w = 50,
//    .header.h = 50,
    .data_size = 5158,
//    .header.cf = LV_IMG_CF_RAW_ALPHA,
    .data = img_wink_png_map,
};
```

So it's enough to set the `.data` and `.data_size`.

Please add some `printf` in `lv_png.c` to see if it's executed and at least LVGL tries to decode the PNG image. 

**BTW, is `LV_USE_PNG` enabled in `lv_conf.h`?**

kdschlosser, 2023-07-19:
Even if the decoder is not turned on where it is failing is when the data is being tested as a NULL pointer.

I do have a question. If I do this in C code what is the result

```c

typedef struct {
    uint8_t reserved;
    uint8_t *data;
} test_struct;

test_struct * test1;

uint8_t data[5] = {1, 1, 1, 1, 1};

test1->data = data;
void * test2 = (void *) test1;
uint8_t * test3 = test2;
test_struct * test4 = test2;

```

how would these evaluate?

```
test3[0] == 0
test4->data == NULL
```

This is what is happening in the C code.

The `test3[0] == 0` evaluates as `true` which is should. But if `test1` was a NULL pointer it would also evaluate as `true` correct? Where it is failing is when the void pointer is cast back to the structure and the data field is checked to see if it is a NULL pointer.

if I call `lv_img_decoder_get_info` which is called from the `lv_img_set_src` function where the failure happens is easier to see.

Here is the code for the `lv_img_decoder_get_info` function. I modified the return values from this function so I could easily see what was happening. and where it was happening. 

```c
lv_res_t lv_img_decoder_get_info(const void * src, lv_img_header_t * header)
{
    lv_memzero(header, sizeof(lv_img_header_t));
    if(src == NULL) return 1;

    lv_img_src_t src_type = lv_img_src_get_type(src);
    if(src_type == LV_IMG_SRC_VARIABLE) {
        lv_img_dsc_t * img_dsc = src;
        if(img_dsc->data == NULL) return 2;
    }

    lv_res_t res = 3;
    lv_img_decoder_t * d;
    _LV_LL_READ(&LV_GC_ROOT(_lv_img_decoder_ll), d) {
        if(d->info_cb) {
            res = d->info_cb(d, src, header);
            if(res == LV_RES_OK) break;
        }
    }

    return res;
}
```

this line

```c
    lv_img_src_t src_type = lv_img_src_get_type(src);
```
returns a `0` which is what it should return when passing a pointer to an lv_img_dsc_t structure.

This is the code block that fails. The returned value I am getting is 2.

```c
if(src_type == LV_IMG_SRC_VARIABLE) {
        lv_img_dsc_t * img_dsc = src;
        if(img_dsc->data == NULL) return 2;
    }
```

kisvegabor, 2023-07-20:
I think I see the problem:
```py
img_header_t._fields_ = [
    ('cf', uint32_t),
    ('always_zero', uint32_t),
    ('reserved', uint32_t),
    ('chroma_keyed', uint32_t),
    ('w', uint32_t),
    ('h', uint32_t)
]
```

E.g. `cf` is not an `uint32_t` variable but a bitfield. 
```
 uint32_t cf : 5; 
```

So 5 bits from an `uint32_t`. I think his is not handles correctly in the Python version.

kdschlosser, 2023-07-20:
oye!!!! i forgot about the bitfields!!!

python does have a way of handling the bitfields I just forgot to code that in.

kdschlosser, 2023-07-22:
THANK YOU!!!!!

That now works properly!!! I can't believe I forgot to add the bitfields. I have been fighting with that for days now all because for some crazy reason I didn't see the bitfields. Once I added the code in to get those added. Poof it started working properly!!!

Sometimes all it takes is a second set of eyes.

I have to run some more tests and see if it will compile and run on Linux and if it does then I am going to push the code changes to the repo. Do a little more testing and see if I can't get others to give it a go as well. I have a bunch of the basic type examples running. I am going to see if I can get the audio mixer demo working. I was having a problem getting the images to load but everything else was working so hopefully it will work properly now. Would be nice.

kdschlosser, 2023-07-22:
OOOoOoooO I got the audio mixer running. That is a decently advanced demo. It looking really good...

[image]

closed 2023-07-23 by kdschlosser

bryghtlabs-richard, 2025-09-29:
`uint32_t cf : 5; `

On a related note, not all LVGL8 color-formats can be stored here, only those `<LV_IMG_CF_USER_ENCODED_2`

kdschlosser, 2025-09-30:
This had to do with V9.x and writing a Python binding to a shared LVGL library using the ctypes module in Python.

bryghtlabs-richard, 2025-10-01:
Agreed - I just meant that LVGL8 has a very similar issue internally.