lvgl/lvgl #4384: vglite freezes when variables are stored in the memory of the FlexSPI interface on NXP

Vinolzy, 2023-07-20:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
platform: NXP RT1176
LVGL version: have tried both version of 8.3.2 and 8.3.8

I have two boards, one with variables stored in the SDRAM via the SEMC interface, and the other with variables stored in the HyperRAM via the FlexSPI interface. Both boards have VGLite enabled. In the board with SDRAM, everything runs smoothly. However, in the board with HyperRAM, there are different behaviors each time it is powered on with VGLite enabled. For some specific components like buttons and switches, there is a 90% chance they will not display correctly, while components like labels and image buttons display properly. If I have two screens, namely screen1 and screen2, there is a high possibility that the program will freeze when entering screen1 after power-up, but it is less likely to freeze if I first enter screen2 and then enter screen1.

To investigate the issue, I created a very simple project with only one label and one button. Each time the board is powered on, the behavior varies, and there is a low probability of GUI rendering correctly.each time after power-on, it shows one of the following four behaviors.
[image]
The program always gets stuck at “vg_lite_os_wait,” waiting for a semaphore released by GPU interrupt, but the GPU interrupt fails to occur as expected.

I have checked multiple times, the programs for HyperRAM and SDRAM are essentially the same. I am certain that I have correctly configured the MPU and linked the relevant LVGL variables to cacheable and non-cacheable regions.

So, I suspect there might be a compatibility issue between FlexSPI and GPU, such as the GPU interrupting FlexSPI timing, or perhaps there is a specific location where cache needs to be flushed.

When vglite is disabled, all components on the HyperRAM board can display normally, and there are no issues or freezes on any interface.But there is also a peculiar issue that when vglite is disabled, displaying the camera image on the canvas shows short green or white horizontal lines at the left and right edges. But when vglite is enabled, the display appears normal. Interestingly, on the previous SDRAM board, there were no such horizontal lines when vglite was disabled.

### To Reproduce
To create an NXP project with HyperRAM, manually allocate all variables used by LVGL and VGLite to the cacheable or non-cacheable regions of HyperRAM.
There are mainly four variables as below:
```
(1)lv_mem.c    cacheable
static LV_ATTRIBUTE_LARGE_RAM_ARRAY MEM_UNIT work_mem_int[128*1024 / sizeof(MEM_UNIT)];

(2)lvgl_support.c    cacheable
static uint8_t s_frameBuffer[2][DEMO_FB_SIZE];

(3)vglite_support.c    non-cacheable
static uint8_t contiguous_mem[MAX_CONTIGUOUS_SIZE];

(4)if you have a mipi camera     both non-cacheable
uint8_t s_cameraBuffer[DEMO_CAMERA_BUFFER_COUNT][DEMO_CAMERA_HEIGHT][DEMO_CAMERA_WIDTH *DEMO_CAMERA_BUFFER_BPP]
uint8_t s_lcdBuffer[DEMO_LCD_BUFFER_COUNT][DEMO_BUFFER_HEIGHT][DEMO_BUFFER_WIDTH * DEMO_LCD_BUFFER_BPP]
```

The information about VGLite is too scarce, and it would take a considerable amount of time to deeply understand its principles. Could you provide me with some advice on how to solve this issue? It has been bothering me for quite a while.

kisvegabor, 2023-07-20:
cc  @nicusorcitu

Vinolzy, 2023-07-21:
> 1. The vglite heap needs to stay in non-cached and aligned region. Is that respected in your project setup

1.Yes,I've linked contiguous_mem to non-cacheable and 32 bit align region using AT_NONCACHEABLE_SECTION_ALIGN.
2.Because the default MAX_CONTIGUOUS_SIZE is 0x20000 which is 2048KB,It's too large to be linked to the internal SRAM.

nicusorcitu, 2023-07-21:
1.  It might not be enough, you have to check the flash (hyper ram) configuration from ConfigMPU.
  2.  The question is why don t you use external sdram instead of flash for vglite heap?

________________________________
De la: Vinolzy ***@***.***>
Trimis: Friday, July 21, 2023 4:17:25 AM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] vglite freezes when variables are stored in the memory of the FlexSPI interface on NXP (Issue #4384)

Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

  1.  The vglite heap needs to stay in non-cached and aligned region. Is that respected in your project setup

1.Yes,I've linked contiguous_mem to non-cacheable and 32 bit align region using AT_NONCACHEABLE_SECTION_ALIGN.
2.Because the default MAX_CONTIGUOUS_SIZE is 0x20000 which is 2048KB,It's too large to be linked to the internal SRAM.

Vinolzy, 2023-07-21:
> 1. It might not be enough, you have to check the flash (hyper ram) configuration from ConfigMPU. 2. The question is why don t you use external sdram instead of flash for vglite heap?
> […](#)

1.The default size of contiguous_mem is 0x200000 (equivalent to 2MB) which is sufficient memory on my SDRAM board. Even if I set the size to 16MB on the hyperram board, the problem remains unresolved.
2.SDRAM is connected through the SEMC interface, which has a larger package size and requires many pins. Considering the product dimensions, our latest board has switched to HyperRAM with a FlexSPI interface.So there is no SDRAM on the current board.

nicusorcitu, 2023-07-21:
1. I would try to play a bit more with the mapping of the flash. I would use the same parameters as the sdram non cached memory defaults. **Memory with Device type, not shareable, non-cacheable.**
I am not 100% sure about the issue but I still believe it shall be related to the memory access type. I would pick up device memory type instead of normal memory type.
[image]
2. I understand now, but you shall expect an decrease of performance. Flash is the slowest memory type.

Vinolzy, 2023-07-21:
> 1. I would try to play a bit more with the mapping of the flash even I have not tried to put the gpu heap into flash.
>    1.1. I would exclude the overlaping region method, just be sure the memory is configured as you want. Configure it one single time correctly.
>    1.2. I would use the same parameters as the sdram non cached memory defaults. **Memory with Device type, not shareable, non-cacheable.**
>    I am not 100% sure about the issue but it shall be related to the memory access type. I would pick up device memory type instead of normal memory type.
>    [image]
> 2. I understand now, but you shall expect an decrease of performance. Flash is the slowest memory type.

1.The MPU overlap method was a test I performed to ensure proper configuration, but in reality, I have always configured it only once. The previous SDRAM board's MPU configuration was set to Normal memory type. I just tested configuring HyperRAM as a device memory type and even as strongly ordered one, but the issue still remains unresolved.

As I mentioned earlier, even when VGLITE is disabled, if I display the camera's frame in LVGL's canvas, some stripes appear on both sides of the screen. However, this phenomenon does not occur on the SDRAM board when VGLITE is disabled. 
[image]

I'm not sure if this is due to CPU overload.On the SDRAM board, even without VGLITE and compiler optimization enabled, the CPU usage reached over 80%, causing some image display stuttering. On the HyperRAM board, I also enabled compiler optimization level -O3, which resulted in CPU usage over 90% and severe stuttering (FPS in the image just represents a certain moment,mostly ranging from 1 to 7).

Nevertheless, it is true that if VGLITE is disabled, all components render correctly, and the program does not get stuck at any point.

2.The HyperRAM datasheet mentions that at 200MHz, the data rate is 400MB/s, which is similar to the speed of the SDRAM I used before, which was 200M words/s mentioned in datasheet.
HyperRAM:[https://www.issi.com/WW/pdf/66-67WVH64M8DALL-BLL.pdf](https://www.issi.com/WW/pdf/66-67WVH64M8DALL-BLL.pdf)
SDRAM:[https://www.micro-semiconductor.com/datasheet/76-W9825G6KH-5-TR.pdf](https://www.micro-semiconductor.com/datasheet/76-W9825G6KH-5-TR.pdf)

But there is one difference: on the SDRAM board, I directly ran the code from the SDRAM, whereas on the current board, the code runs on the NOR Flash through the FlexSPI interface at a speed of 200MHz. Therefore, the HyperRAM board's FPS and CPU usage may indeed be slightly worse than the SDRAM board's performance, but it is still acceptable.

nicusorcitu, 2023-07-21:
> As I mentioned earlier, even when VGLITE is disabled, if I display the camera's frame in LVGL's canvas, some stripes appear on both sides of the screen. However, this phenomenon does not occur on the SDRAM board when VGLITE is disabled.
>
So the problem is not related only to the vglite, but also CPU itself have issues when the framebuffer is in HyperRAM, correct?

> I'm not sure if the stripes are due to CPU overload.On the SDRAM board, even without VGLITE and compiler optimization enabled, the CPU usage reached over 80%, only causing some image display stuttering. On the HyperRAM board, I also enabled compiler optimization level -O3, which resulted in CPU usage over 90% and severe stuttering (FPS in the image just represents a certain moment,mostly ranging from 1 to 7).
>
I believe you actually reached the limits of the CPU. I am not confident of the 33FPS with 92% CPU.
Please change the lines:
```
void _lv_disp_refr_timer(lv_timer_t * tmr) from file lv_refr.c
--     if(lv_tick_elaps(perf_monitor.perf_last_time) < 300) {
++   if(lv_tick_elaps(perf_monitor.perf_last_time) < 900) {
```
And remove the fps_limit
```
        // if(fps > fps_limit) {
        //     fps = fps_limit;
        // }
```

Vinolzy, 2023-07-21:
> So the problem is not related only to the vglite, but also CPU itself have issues when the framebuffer is in HyperRAM, correct?
>
>
> I believe you actually reached the limits of the CPU. I am not [...]

Yes, if I directly display the camera's image on the LCD, it's very smooth, and there are no stripes on both sides.
However, when displaying it on the LVGL's canvas, it seems like there's an additional step of converting the original LCD buffer to be displayed on the canvas, which results in more CPU usage.

Before making the changes you suggested, the CPU usage is either 78% or 96%, and the FPS is either 4 or 33.
Now FPS is always 3 and CPU usage is either 58% or 81%.Most of time is 58%.

nicusorcitu, 2023-07-21:
> Now FPS is always 3 and CPU usage is either 58% or 81%.Most of time is 58%.
>
This is your real FPS, after applying the changes I mentioned.

>However, I always get the following warnings during compilation. Is this normal?
>
No, it is not normal. I also got into these warnings recently and the fix is easy:
```
static void _get_subarc_control_points(vg_arc * arc, int32_t span)
{
    vg_arc fullarc = {0};
```
and
```
static void _get_arc_control_points(vg_arc * arc, bool start)
{
    vg_arc fullarc = {0};
```
If I would have seen them in the past I would fix them. Anyway they are just warnings, will never get into default case to be used uninitialized.

Vinolzy, 2023-07-25:
I had solved the issue just now.It's the issue of FlexSPI configuration.

Vinolzy, 2023-07-25:
I have solved it by myself.It's not related to the LVGL or VGLITE code,but nxp interface configurations.Thanks for your help.

closed 2023-07-25 by Vinolzy

nicusorcitu, 2023-07-31:
Can you please elaborate what do you mean by "nxp interface configuration"? What is the fix?

Vinolzy, 2023-07-31:
> Can you please elaborate what do you mean by "nxp interface configuration"? What is the fix?

```
bool enableReadAddressOpt;    /*!< Enable/disable remove AHB read burst start address alignment limitation.
                             when enable, there is no AHB read burst start address alignment limitation. */
```
This field in FlexSPI should be true to remove alignment limitation.