lvgl/lvgl #4388: [Feature] Adding vector API support

XuNeo, 2023-07-20:
### Introduce the problem

Adding vector API support has many benefits like building vector based scenes, creating fancy animation and drawing scalable images, saving spaces.

I haven't done any work yet, but want to provide some usecases and bring up the discussion.

I think supporting it in `lvgl` means porting an existing libraries like [`ThorVG`](https://github.com/thorvg/thorvg) and providing acceleration APIs porting layer.

GPUs like `vglite`, `nemaGFX` that support 2D vector operations can be used for acceleration.

### Examples and cases

#### Fancy vector based images from ThorVG
We have impelemented it together on lv_canvas using builtin software raster engine. It looks cool on simulator but laggy on device.
[image]

#### Cubic curve on watch
This curve is done by custom widget and call GPU APIs for drawing path directly in `DRAW_MAIN` event.

https://github.com/lvgl/lvgl/assets/25867991/be0cc942-21c5-496f-907d-2ed377c79a76

### Suggested solution

For reference, `vglite` supports below path operations.
```c
/* Path command (op code). */
#define VLC_OP_END          0x00
#define VLC_OP_CLOSE        0x01
#define VLC_OP_MOVE         0x02
#define VLC_OP_MOVE_REL     0x03
#define VLC_OP_LINE         0x04
#define VLC_OP_LINE_REL     0x05
#define VLC_OP_QUAD         0x06
#define VLC_OP_QUAD_REL     0x07
#define VLC_OP_CUBIC        0x08
#define VLC_OP_CUBIC_REL    0x09

#define VLC_OP_SCCWARC      0x0A
#define VLC_OP_SCCWARC_REL  0x0B
#define VLC_OP_SCWARC       0x0C
#define VLC_OP_SCWARC_REL   0x0D
#define VLC_OP_LCCWARC      0x0E
#define VLC_OP_LCCWARC_REL  0x0F
#define VLC_OP_LCWARC       0x10
#define VLC_OP_LCWARC_REL   0x11
```

For freetype, it provides below outline API. [Link to original code.](https://github.com/freetype/freetype/blob/4d8db130ea4342317581bab65fc96365ce806b77/include/freetype/ftimage.h#L664)
```c
  typedef struct  FT_Outline_Funcs_
  {
    FT_Outline_MoveToFunc   move_to;
    FT_Outline_LineToFunc   line_to;
    FT_Outline_ConicToFunc  conic_to;
    FT_Outline_CubicToFunc  cubic_to;
    int                     shift;
    FT_Pos                  delta;
  } FT_Outline_Funcs;
```

kisvegabor, 2023-07-21:
Thank you for opening this topic. Establishing a vector API is on our [ROADMAP for v9](https://github.com/lvgl/lvgl/issues/4016).

I think it's a good idea to getting started with the feature set of VG-Lite. All vector APIs I saw so far work with pushing commands in a stack and not via function calls like ˙`draw_bezier3(p1, p2, p3, dash, color, width, cap, ...)`. So probably we should also do a similar stack based API. 

It'd be very important to have a built in SVG parser which can convert SVG commands into our API. 

Implementing ThorVG would be amazing and it's great that you have already integrated it. In the end it'd be important to make it work with:
- CMake
- Make
- Simple managed build e.g. in Eclipse. I think it's the most complicated. Do you think it's possible?

I think the performance of SW rasterizer is not a big issue because
1. On embedded probably you will use vector graphics if you have a capable 2D GPU as well
2. Desktop should be fast enough for SW render too
3. The result can be cached, so it needs to be rasterized only once.

PeterBee97, 2023-07-24:
Here is a draft header for vector APIs:
```
typedef struct {
    lv_color_t color;
    lv_coord_t width;
    lv_opa_t opa;
    const void * img_src;
} lv_vector_fill_dsc_t;

typedef struct {
    lv_color_t color;
    lv_coord_t width;
    lv_opa_t opa;
    lv_coord_t dash_width;
} lv_vector_stroke_dsc_t;

enum {
    LV_VECTOR_PATH_FORMAT_VGLITE,
    LV_VECTOR_PATH_FORMAT_CUSTOM
};
typedef uint8_t lv_vector_path_format_t;

struct _lv_vector_path_t {
    uint8_t * path;
    lv_vector_path_format_t format;
    uint8_t * stroked_path; /*For caching stroked path where native stroke is unsupported (i.e. VGLite)*/
    lv_matrix_t * matrix;
    uint32_t id;

    /*Path operations*/
    uint8_t* (*create)(struct _lv_vector_path_t * path, size_t length);
    void (*destroy)(struct _lv_vector_path_t * path);
    void (*move_to)(struct _lv_vector_path_t * path, lv_point_t * p);
    void (*line_to)(struct _lv_vector_path_t * path, lv_point_t * p);
    void (*quad_to)(struct _lv_vector_path_t * path, lv_point_t * p1, lv_point_t * p2);
    void (*cubic_to)(struct _lv_vector_path_t * path, lv_point_t * p1, lv_point_t * p2, lv_point_t * p3);
    void (*arc_to)(struct _lv_vector_path_t * path, lv_point_t * c, lv_point_t * p);
    void (*close)(struct _lv_vector_path_t * path);
    lv_res_t (*append_rect)(struct _lv_vector_path_t * path, lv_area_t *rect, lv_coord_t r);
    lv_res_t (*append_circle)(struct _lv_vector_path_t * path, lv_point_t * c, lv_coord_t r);
    /**
     * @param[in] start_angle The start angle of the arc given in degrees, measured counter-clockwise from the horizontal line.
     * @param[in] sweep The central angle of the arc given in degrees, measured counter-clockwise from @p start_angle.
     * @param[in] pie Specifies whether to draw radii from the arc's center to both of its end-point - drawn if @c true.
     */
    lv_res_t (*append_arc)(struct _lv_vector_path_t * path, lv_point_t * c, lv_coord_t r, uint16_t start_angle, uint16_t sweep, bool pie);
    lv_res_t (*append_path)(struct _lv_vector_path_t * path, struct _lv_vector_path_t * subpath);

    /*Draw path commands*/
    void (*fill)(struct _lv_vector_path_t * path, lv_vector_fill_dsc_t * dsc);
    void (*stroke)(struct _lv_vector_path_t * path, lv_vector_stroke_dsc_t * dsc);
    void (*scissor)(struct _lv_vector_path_t * path, lv_area_t * area);
    void (*transform)(struct _lv_vector_path_t * path, lv_matrix_t * m);
};
typedef struct _lv_vector_path_t lv_vector_path_t;
```
Furthur perfection is expected. What do you think?

kisvegabor, 2023-07-24:
```c
enum {
    LV_VECTOR_PATH_FORMAT_VGLITE,
    LV_VECTOR_PATH_FORMAT_CUSTOM
};
typedef uint8_t lv_vector_path_format_t;
```

Why do we need this? Ideally the vector API should be vendor independent. 

I'm thinking about if we need to support the SW vector API and e.g. VGLite API at the same time. Is there a use case when we should make them work in parallel? Making them working on a the same image (e.g. path1 to SW, path2 to VGLite) doesn't make sense, but it might work if they are working on different images in parallel.  But it doesn't sound too reasonable either and might result in flickering if an image is not taken by the same engine.

So I think we can just select a single vector engine and all vector operations will be passed to it. 
---

I think `/*Path operations*/` callbacks shouldn't be stored for each path. Instead we can create functions like 
```c
lv_vglite_vector.c

#if LV_USE_VECTOR_ENGINE == LV_VECTOR_ENGINE_VGLITE
    uint8_t* lv_vector_create(struct _lv_vector_path_t * path, size_t length) { }
    void lv_vector_destroy(struct _lv_vector_path_t * path) { }
    void lv_vector_move_to(struct _lv_vector_path_t * path, lv_point_t * p) { }
    void lv_vector_line_to(struct _lv_vector_path_t * path, lv_point_t * p) { }
    ...
#endif
```
```c
lv_thorvg_vector.c

#if LV_USE_VECTOR_ENGINE == LV_VECTOR_ENGINE_THORVG
   uint8_t* lv_vector_create(struct _lv_vector_path_t * path, size_t length) { }
    void lv_vector_destroy(struct _lv_vector_path_t * path) { }
    void lv_vector_move_to(struct _lv_vector_path_t * path, lv_point_t * p) { }
    void lv_vector_line_to(struct _lv_vector_path_t * path, lv_point_t * p) { }
#endif
```

It similar to e.g. malloc API. See [here](https://github.com/lvgl/lvgl/blob/master/src/stdlib/clib/lv_mem_core_clib.c).

---

I'm still thinking about how to make it fit into the current `draw_task` based rendering pipeline. 
If I understand correctly the `/*Path operations*/` don't require  the GPU itself, these are just functions to manipulate the parameters of the path. The GPU should be required only for `/*Draw path commands*/`. Is the correct? 

If so, probably we can create `LV_DRAW_TASK_PATH_FILL/STROKE/SCISSOR/TRANSFORM`. What do you think?

cc @nicusorcitu

kisvegabor, 2023-07-27:
FYI, Lottie support is added to ThorVG: https://forum.lvgl.io/t/rive-create-and-ship-interactive-animations-to-any-platform/5819/18

PeterBee97, 2023-07-31:
You're right, we should select path format at compile time.

> I'm still thinking about how to make it fit into the current `draw_task` based rendering pipeline. If I understand correctly the `/*Path operations*/` don't require the GPU itself, these are just [...]

Yes. It's like a canvas where creating content is computationally simple.
> If so, probably we can create `LV_DRAW_TASK_PATH_FILL/STROKE/SCISSOR/TRANSFORM`. What do you think?

The real draw task has only two types, fill and stroke, so we may only need LV_DRAW_TASK_PATH_FILL and LV_DRAW_TASK_PATH_STROKE. The scissor and transform are just parameters of the path.

XuNeo, 2023-08-15:
Still on going.

onecoolx, 2023-08-28:
I suggest we should define a full featured 2D vector graphics API which is independent of the limitations of 2D GPU capabilities. 
For my experience, a full featured 2D graphic API needs to contains the following objects at least:
* GraphicContext
* Matrix
* Path
* Gradient
* Pattern
* Font
* Image

Among **GraphicsContext** is the core object, which is used to save various states and attributes during the graphics drawing process. Such as fill color, fill mode, transformation, stroke width, stroke color, dashed line, clipping region, font family, etc. The **Path** object is used to describe the vector geometric data. and is the main object for drawing.
  
For examples use:
```
    ...
    // draw a red rectangle.

    lvg_context* ctx = lvg_create_context(canvas);
    lvg_clear(ctx);

    lvg_rect rect = {0, 0, 100, 100};
    lvg_set_fill_color(ctx, 0xFF000000);
    lvg_rectangle(ctx, &rect);
    lvg_fill(ctx);

    lvg_flush(); // finish draw.

    ...

    lvg_context_unref(ctx); // release contxt memory.
```

```
    ...
    // draw a user define path.
    lvg_context* ctx = lvg_create_context(canvas);
    lvg_clear(ctx);

    lvg_path* path = lvg_create_path();
    lvg_rect rect = {0, 0, 100, 100};
    lvg_rect rect = {20, 20, 50, 50};
    lvg_path_add_rect(path, &rect);
    lvg_path_add_rect(path, &rect2);

    lvg_set_path(ctx, path);
    lvg_set_fill_color(ctx, 0xFF000000);
    lvg_set_fill_rule(ctx, LVG_FILL_RULE_EVEN_ODD);
    lvg_fill(ctx);

    lvg_flush(); // finish draw.

    ...

    lvg_path_unref(path);
    lvg_context_unref(ctx);
```

```
    ...
    // draw multi path and transformation.
    lvg_context* ctx = lvg_create_context(canvas);
    lvg_clear(ctx);

    lvg_identity(ctx); // reset transform.
    lvg_translate(ctx, 200, 200); // translate

    lvg_rect rect = {0, 0, 100, 100};
    lvg_set_fill_color(ctx, 0xFF000000);
    lvg_rectangle(ctx, &rect);
    lvg_fill(ctx); // fill red rectangle

    lvg_rotate(ctx, 0.32f); // rotate transform.

    lvg_rect rect = {20, 20, 50, 50};
    lvg_set_fill_color(ctx, 0x0000FF00);
    lvg_rectangle(ctx, &rect);
    lvg_stroke(ctx); // stroke blue rectangle

    lvg_flush(); // finish draw.

    ...

    lvg_context_unref(ctx); 
```

Such  API design let LVGL  have a rich drawing capabilities, independent the limitations of the GPU, Regardless of whether the backporting  is hw or sw, the user code is consistent.
Anyone have an ideas?

PeterBee97, 2023-08-28:
Refined previously proposed vector API as follows (mostly following parameter sets from ThorVG):
```c
src/draw/lv_draw_vector.h

enum {
    LV_VECTOR_PATH_FILL_NONZERO = 0,
    LV_VECTOR_PATH_FILL_EVENODD
}
typedef uint8_t lv_vector_path_fill_t;

enum {
    LV_VECTOR_PATH_PATTERN_MODE_COLOR = 0,
    LV_VECTOR_PATH_PATTERN_MODE_PAD,
}
typedef uint8_t lv_vector_path_pattern_mode_t;

enum {
    LV_VECTOR_STROKE_CAP_STYLE_SQUARE = 0,
    LV_VECTOR_STROKE_CAP_STYLE_ROUND,
    LV_VECTOR_STROKE_CAP_STYLE_BUTT,
}
typedef uint8_t lv_vector_stroke_cap_style_t;

enum {
    LV_VECTOR_STROKE_JOIN_STYLE_BEVEL = 0,
    LV_VECTOR_STROKE_JOIN_STYLE_ROUND,
    LV_VECTOR_STROKE_JOIN_STYLE_MITER,
}
typedef uint8_t lv_vector_stroke_join_style_t;

typedef struct {
    lv_color_t color;
    lv_opa_t opa;
    lv_vector_path_fill_t fill_type;
    lv_vector_path_pattern_mode_t pattern_mode;
    const void * img_src;
    lv_grad_dsc_t grad;
} lv_vector_fill_dsc_t;

typedef struct {
    lv_color_t color;
    lv_coord_t width;
    lv_opa_t opa;
    float * dash_pattern;
    lv_coord_t dash_pattern_count;
    lv_vector_stroke_cap_style_t cap;
    lv_vector_stroke_join_style_t join;
    lv_coord_t miter_limit;
    lv_grad_dsc_t grad;
} lv_vector_stroke_dsc_t;

typedef struct {
    float m[3][3];
} lv_matrix_t;

struct _lv_vector_path_t {
    uint8_t * path;
    uint8_t * stroked_path; /*For caching stroked path where native stroke is unsupported (i.e. VGLite)*/
    lv_matrix_t * matrix;
};
typedef struct _lv_vector_path_t lv_vector_path_t;

/*Path operations*/
uint8_t* lv_vector_create(struct _lv_vector_path_t * path, size_t length);
void lv_vector_destroy(struct _lv_vector_path_t * path);
void lv_vector_move_to(struct _lv_vector_path_t * path, lv_point_t * p);
void lv_vector_line_to(struct _lv_vector_path_t * path, lv_point_t * p);
void lv_vector_quad_to(struct _lv_vector_path_t * path, lv_point_t * p1, lv_point_t * p2);
void lv_vector_cubic_to(struct _lv_vector_path_t * path, lv_point_t * p1, lv_point_t * p2, lv_point_t * p3);
void lv_vector_arc_to(struct _lv_vector_path_t * path, lv_point_t * c, lv_point_t * p);
void lv_vector_close(struct _lv_vector_path_t * path);
void lv_vector_set_scissor(struct _lv_vector_path_t * path, lv_area_t * area);
void lv_vector_set_transform(struct _lv_vector_path_t * path, lv_matrix_t * m);
void lv_vector_set_stroke_width(struct _lv_vector_path_t * path, lv_coord_t width);
void lv_vector_set_stroke_dash(struct _lv_vector_path_t * path, float * dash_pattern, lv_coord_t dash_pattern_count);
void lv_vector_set_stroke_cap(struct _lv_vector_path_t * path, lv_vector_stroke_cap_style_t cap);
void lv_vector_set_stroke_join(struct _lv_vector_path_t * path, lv_vector_stroke_join_style_t join);
void lv_vector_set_stroke_miter_limit(struct _lv_vector_path_t * path, lv_coord_t miter_limit);
lv_res_t lv_vector_append_rect(struct _lv_vector_path_t * path, lv_area_t *rect, lv_coord_t r);
lv_res_t lv_vector_append_circle(struct _lv_vector_path_t * path, lv_point_t * c, lv_coord_t r);
/**
 * @param[in] start_angle The start angle of the arc given in degrees, measured counter-clockwise from the horizontal line.
 * @param[in] sweep The central angle of the arc given in degrees, measured counter-clockwise from @p start_angle.
 * @param[in] pie Specifies whether to draw radii from the arc's center to both of its end-point - drawn if @c true.
 */
lv_res_t lv_vector_append_arc(struct _lv_vector_path_t * path, lv_point_t * c, lv_coord_t r, uint16_t start_angle, uint16_t sweep, bool pie);
lv_res_t lv_vector_append_path(struct _lv_vector_path_t * path, struct _lv_vector_path_t * subpath);

/*Draw path commands*/
void lv_vector_fill(struct _lv_vector_path_t * path, lv_vector_fill_dsc_t * dsc);
void lv_vector_stroke(struct _lv_vector_path_t * path, lv_vector_stroke_dsc_t * dsc);
```
It is similar to what @onecoolx has proposed, except for the main object name being lv_vector_t instead of a more generalized lvg_context. The "flush" job is left for the parallel rendering architecture draw task handling.

nicusorcitu, 2023-08-28:
The API refined looks good from vglite perspective. It covers also the vector font support.
It will be another story how to integrate those VG API functionalities into the draw context - as a separate task.

@kisvegabor but LV_DRAW_TASK_TYPE_FILL (for rectangles, rounded rectangle, circles), LV_DRAW_TASK_TYPE_LINE, LV_DRAW_TASK_TYPE_ARC, LV_DRAW_TASK_TYPE_TRIANGLE will be still in place, correct? even if there will be VG API support in v9.

kisvegabor, 2023-08-28:
It looks good to me too.

I see that in `lv_vector_fill_dsc_t` we can pass an `const void * img_src`. Is it a background image for the path, right? So we can mask out any vector shape from an image. If so do we need some other parameters for the image like: tile, zoom, center, etc? I'm not sure if it's supported in the the GPUs that we are targeting now, just asking. 

Would it be possible (or is it required) to support  an A8 image mask during stroke and fill? Or shall it be a different stage and generic feature for any draw task?

>  It covers also the vector font support.

Where is it covered? Or do you mean this feature set is enough for vector font drawing too?

>  LV_DRAW_TASK_TYPE_LINE, LV_DRAW_TASK_TYPE_ARC, LV_DRAW_TASK_TYPE_TRIANGLE will be still in place, correct? even if there will be VG API support in v9.

Yes,  on "draw level" we should have 2 new draw task types:
- `LV_DRAW_TASK_TYPE_VECTOR_FILL` with `lv_vector_path_t` and `lv_vector_fill_dsc_t` parameters
- `LV_DRAW_TASK_TYPE_VECTOR_STROKE` with `lv_vector_path_t` and `lv_vector_stroke_dsc_t` parameters

So you put together a `lv_vector_path_t` in a widget, and call e.g. `lv_draw_vector_fill(&vector, &fill_dsc, &area)`.  We should have an `lv_vector` widget (with a better name) similar to `lv_line` where the users can easily put together any vector graphics. 

I wonder if can define some helpers for matrix transformation, e.g. `lv_vector_scale(vector, zoom, pivot_x, pivot_y) ;`
or `lv_vector_rotate(vector, angle, pivot_x, pivot_y) ;`

PeterBee97, 2023-08-29:
> I see that in `lv_vector_fill_dsc_t` we can pass an `const void * img_src`. Is it a background image for the path, right? So we can mask out any vector shape from an image. If so do we need some [...]

The VGLite API does support different transformation matrices for both the path and the image, so yes.

> Would it be possible (or is it required) to support an A8 image mask during stroke and fill? Or shall it be a different stage and generic feature for any draw task?

The VGLite does not support bitmap masking even with regular BLIT operations.

> I wonder if can define some helpers for matrix transformation, e.g. `lv_vector_scale(vector, zoom, pivot_x, pivot_y) ;` or `lv_vector_rotate(vector, angle, pivot_x, pivot_y) ;`

Sure.

onecoolx, 2023-08-29:
I have a suggestion, can we put the filldst and strokedst  into a whole struct? 

like this:

```

typedef struct {
    lv_color_t color;
    lv_opa_t opa;
    lv_vector_path_fill_t fill_type;
    lv_vector_path_pattern_mode_t pattern_mode;
    const void * img_src;
    lv_grad_dsc_t grad;
    lv_matrix_t  pattern_matrix;
} lv_vector_fill_dsc_t;

typedef struct {
    lv_color_t color;
    lv_coord_t width;
    lv_opa_t opa;
    float * dash_pattern;
    lv_coord_t dash_pattern_count;
    lv_coord_t dash_offset;
    lv_vector_stroke_cap_style_t cap;
    lv_vector_stroke_join_style_t join;
    lv_coord_t miter_limit;
    lv_grad_dsc_t grad;
} lv_vector_stroke_dsc_t;

typedef struct {
    float m[3][3];
} lv_matrix_t;

struct _lv_vector_path_t {
    uint8_t * path;
    uint8_t * stroked_path; /*For caching stroked path where native stroke is unsupported (i.e. VGLite)*/
};
typedef struct _lv_vector_path_t lv_vector_path_t;

typedef enum  {
    LV_DRAW_TYPE_FILL = 0x01;
    LV_DRAW_TYPE_STROKE = 0x02;
} draw_type;

typedef struct {
    lv_vector_path_t path; // draw path
    lv_matrix_t  matrix; // global matrix
    uint8_t draw_type; // draw type
    lv_vector_fill_dsc_t fill_dsc;
    lv_vector_stroke_dsc_t stroke_dsc;
} lv_vector_draw_info;
```

The draw_type can be LV_DRAW_TYPE_FILL, LV_DRAW_TYPE_STROKE or (LV_DRAW_TYPE_FILL | LV_DRAW_TYPE_STROKE).

This can change the drawing task into a single interface.
```
void lv_vector_draw(lv_vector_draw_info * info);

```
This can support the fill and stroke at the same time in future.

```
VG_LITE_DRAW_FILL_STROKE_PATH 
```

nicusorcitu, 2023-08-29:
@kisvegabor 

> Where is it covered? Or do you mean this feature set is enough for vector font drawing too?
>
Yes, by those APIs:
```
void lv_vector_move_to(struct _lv_vector_path_t * path, lv_point_t * p);
void lv_vector_line_to(struct _lv_vector_path_t * path, lv_point_t * p);
void lv_vector_quad_to(struct _lv_vector_path_t * path, lv_point_t * p1, lv_point_t * p2);
void lv_vector_cubic_to(struct _lv_vector_path_t * path, lv_point_t * p1, lv_point_t * p2, lv_point_t * p3);
void lv_vector_arc_to(struct _lv_vector_path_t * path, lv_point_t * c, lv_point_t * p);
```
Those shall cover the font which is composed by lines, cubics... But it does not cover the text layout.

@PeterBee97 
> The VGLite API does support different transformation matrices for both the path and the image, so yes.
>
This is correct. So it would be useful to have the transformation on paths too. It does not make sense to draw paths to a layer and then to rotate that layer (as it behaves right now in the FILL task). The GPU can rotate it in a single shot.

> The VGLite does not support bitmap masking even with regular BLIT operations.
>
RT1170 and RT500 does not support bitmap masking. But the next RT700 (release plan for the next year) will support it.

PeterBee97, 2023-08-29:
Merging the fill and stroke look good since it is also how ThorVG did this. Also adding other parameter-setting functions and matrix helpers, now the draft looks like this:
```c
enum {
    LV_VECTOR_PATH_FILL_NONZERO = 0,
    LV_VECTOR_PATH_FILL_EVENODD
}
typedef uint8_t lv_vector_path_fill_rule_t;

enum {
    LV_VECTOR_PATH_PATTERN_MODE_COLOR = 0,
    LV_VECTOR_PATH_PATTERN_MODE_PAD,
}
typedef uint8_t lv_vector_path_pattern_mode_t;

enum {
    LV_VECTOR_STROKE_CAP_STYLE_SQUARE = 0,
    LV_VECTOR_STROKE_CAP_STYLE_ROUND,
    LV_VECTOR_STROKE_CAP_STYLE_BUTT,
}
typedef uint8_t lv_vector_stroke_cap_style_t;

enum {
    LV_VECTOR_STROKE_JOIN_STYLE_BEVEL = 0,
    LV_VECTOR_STROKE_JOIN_STYLE_ROUND,
    LV_VECTOR_STROKE_JOIN_STYLE_MITER,
}
typedef uint8_t lv_vector_stroke_join_style_t;

typedef struct {
    float m[3][3];
} lv_matrix_t;
typedef struct {
    lv_color_t color;
    lv_opa_t opa;
    lv_vector_path_fill_rule_t fill_rule;
    lv_vector_path_pattern_mode_t pattern_mode;
    lv_draw_img_dsc_t * img_dsc;
    lv_grad_dsc_t * grad;
    lv_matrix_t * matrix;
} lv_vector_fill_dsc_t;

typedef struct {
    lv_color_t color;
    lv_coord_t width;
    lv_opa_t opa;
    float * dash_pattern;
    lv_coord_t dash_pattern_count;
    lv_coord_t dash_offset;
    lv_vector_stroke_cap_style_t cap;
    lv_vector_stroke_join_style_t join;
    lv_coord_t miter_limit;
    lv_grad_dsc_t * grad;
} lv_vector_stroke_dsc_t;

struct _lv_vector_path_t {
    uint8_t * path;
    uint32_t path_length;
    uint8_t * stroked_path; /*For caching stroked path where native stroke is unsupported (i.e. VGLite)*/
    uint32_t stroked_path_length;
    lv_area_t * scissor_area;
    lv_matrix_t * matrix;
    lv_vector_fill_dsc_t * fill_dsc;
    lv_vector_stroke_dsc_t * stroke_dsc;
    bool stroke_first;
};
typedef struct _lv_vector_path_t lv_vector_path_t;

/*Path operations*/
struct _lv_vector_path_t * lv_vector_create(uint8_t * path, uint32_t allocated_size);
struct _lv_vector_path_t * lv_vector_copy(struct _lv_vector_path_t * path);
[... 58 lines cut ...]
void lv_vector_rotate(struct _lv_vector_path_t * path, float angle, lv_coord_t pivot_x, lv_coord_t pivot_y);
void lv_vector_translate(struct _lv_vector_path_t * path, lv_coord_t x, lv_coord_t y);
```

A user is expected to write something like this:
```c
lv_vector_path_t * v = lv_vector_create(NULL, 0);
lv_vector_move_to(v, 0, 0);
lv_vector_line_to(v, 0, 100);
lv_vector_line_to(v, 100, 100);
lv_vector_line_to(v, 100, 0);
lv_vector_close(v);
lv_vector_append_circle(v, 50, 50, 50);
lv_vector_init_fill_dsc(v, NULL);
lv_vector_set_fill_rule(v, LV_VECTOR_PATH_FILL_EVENODD);
lv_vector_set_fill_color(v, lv_color_hex(0xFF00FF));
lv_vector_init_stroke_dsc(v, NULL);
lv_vector_set_stroke_width(v, 10);
lv_vector_set_stroke_color(v, lv_color_hex(0x00FF00));
lv_vector_translate(0, 0);
lv_vector_draw(v);
```

And get something like this:
[image]

onecoolx, 2023-08-29:
```
/*Helper functions*/
void lv_vector_scale(struct _lv_vector_path_t * path, float zoom, lv_coord_t pivot_x, lv_coord_t pivot_y);
void lv_vector_rotate(struct _lv_vector_path_t * path, float angle, lv_coord_t pivot_x, lv_coord_t pivot_y);
void lv_vector_translate(struct _lv_vector_path_t * path, lv_coord_t x, lv_coord_t y);
```

Is this better? The first parameter is matrix？

```
void lv_vector_scale(lv_matrix_t * matrix,  float zoomX, float zoonY, lv_coord_t pivot_x, lv_coord_t pivot_y);
void lv_vector_rotate(lv_matrix_t * matrix,  float angle, lv_coord_t pivot_x, lv_coord_t pivot_y);
void lv_vector_translate(lv_matrix_t * matrix,  lv_coord_t x, lv_coord_t y);
```

PeterBee97, 2023-08-30:
> > typedef enum  {
> > LV_DRAW_TYPE_FILL = 0x01;
> > LV_DRAW_TYPE_STROKE = 0x02;
> > } draw_type;
> 
> I like the idea of combining fill and stroke but the user can easily set fill_dsc.opa = 0 and stroke_dsc.width = 0 to skip them. So `LV_DRAW_TYPE_*` feels redundant to me.

True. Only one bit for indicating fill and stroke order then. Edited back to `bool stroke_first;` above.

onecoolx, 2023-08-31:
> True. Only one bit for indicating fill and stroke order then. Edited back to `bool stroke_first;` above.

OK, It just suggestion.  This design is fine too for me.

onecoolx, 2023-08-31:
> ```c
> void lv_vector_move_to(struct _lv_vector_path_t * path, lv_coord_t x, lv_coord_t y);
> void lv_vector_line_to(struct _lv_vector_path_t * path, lv_coord_t x, lv_coord_t y);
> void lv_vector_quad_to(struct _lv_vector_path_t * path, lv_coord_t x1, lv_coord_t y1, lv_coord_t x2, lv_coord_t y2);
> void lv_vector_cubic_to(struct _lv_vector_path_t * path, lv_coord_t x1, lv_coord_t y1, lv_coord_t x2, lv_coord_t y2, lv_coord_t x3, lv_coord_t y3);
> void lv_vector_arc_to(struct _lv_vector_path_t * path, lv_coord_t x1, lv_coord_t y1, lv_coord_t x2, lv_coord_t y2, lv_coord_t r);
> ```

For these functions, is it more reasonable to change the parameters x and y to point?  
For example 
```
 void lv_vector_move_to(struct _lv_vector_path_t * path, lv_point_t* point);
```

nicusorcitu, 2023-08-31:
> For these functions, is it more reasonable to change the parameters x and y to point?
>
I would say yes.

kisvegabor, 2023-08-31:
>> For these functions, is it more reasonable to change the parameters x and y to point?

> I would say yes.

I also agree :+1:

PeterBee97, 2023-08-31:
Updated previous code with lv_point_t. Also adding blend_mode to dsc and lv_vector_clear() function (and also the missing lv_layer_t argument).

onecoolx, 2023-09-04:
> ```c
> struct _lv_draw_vector_dsc_t {
>     lv_vector_fill_dsc_t fill_dsc;
>     lv_vector_stroke_dsc_t stroke_dsc;
>     lv_matrix_t matrix;
>     lv_blend_mode_t blend_mode;   <------ this type
>     lv_area_t scissor_area;
>     bool stroke_first;
> };
> typedef struct _lv_draw_vector_dsc_t lv_draw_vector_dsc_t;
> ```

I think the lv_blend_mode_t here needs to be modified to a new type to support all SVG blend types.  Right ?
```
enum {
    LV_VECTOR_BLEND_MODE_SRC_OVER,
    LV_VECTOR_BLEND_MODE_SRC_IN,
    LV_VECTOR_BLEND_MODE_DST_OVER,
    LV_VECTOR_BLEND_MODE_DST_IN,
    LV_VECTOR_BLEND_MODE_SCREEN, 
    LV_VECTOR_BLEND_MODE_MULTIPLY,
    LV_VECTOR_BLEND_MODE_NONE,
    LV_VECTOR_BLEND_MODE_ADDITIVE,
    LV_VECTOR_BLEND_MODE_SUBTRACTIVE
};
typedef uint8_t lv_vector_blend_mode_t;   
```

PeterBee97, 2023-09-05:
There are more blending modes in [Porter-Duff definitions](http://ssp.impulsetrain.com/porterduff.html) such as src-atop, dst_atop, xor, etc. Is there any plan to support all blending modes consistently across LVGL? @kisvegabor

kisvegabor, 2023-09-05:
> Is there any plan to support all blending modes consistently across LVGL?

I was thinking about it earlier, however on a general UI with widgets I couldn't imagine any use case for Porter-Duff blend modes other than "src over". I see that it's useful for vector drawing but I can't see use case when we build up a UI from buttons, sliders, etc. Do you know any real life use cases?

onecoolx, 2023-09-06:
> I was thinking about it earlier, however on a general UI with widgets I couldn't imagine any use case for Porter-Duff blend modes other than "src over". I see that it's useful for vector drawing but [...]

Yes，I think you are right, for basic UI, "SRC OVER" is enough, other blending methods are only for vector drawing.  So we need to define a separate set of enums for the drawing api.

PeterBee97, 2023-09-07:
I have opened a draft pull request #4528 . We can discuss the details there.

PeterBee97, 2023-09-11:
> I was thinking about it earlier, however on a general UI with widgets I couldn't imagine any use case for Porter-Duff blend modes other than "src over". I see that it's useful for vector drawing but [...]

Not really. But I think if we want to introduce definitions of other blending modes, expanding lv_blend_mode_t would be cleaner than defining another enum. SW renderer may just return error on unsupported modes or be optionally expanded later.

kisvegabor, 2023-09-11:
This [page](http://ssp.impulsetrain.com/porterduff.html) in the "Blending" section mentions exactly this use case. 

It says the  Porter-Duff and Additive, Multiply, Difference, etc blend modes can exist independently and the Blend modes can be applied on the "Both" region of  Porter-Duff layering. 

It means Blend modes and  Porter-Duff are independent and there should be two separate enums to configure them .

PeterBee97, 2023-09-11:
> This [page](http://ssp.impulsetrain.com/porterduff.html) in the "Blending" section mentions exactly this use case.
>
> It says the Porter-Duff and Additive, Multiply, Difference, etc blend modes can [...]

You're right. Porter-Duff are a complete set of definitions on its own. Should we put the Porter-Duff enum in vector API headers or somewhere more common?

kisvegabor, 2023-09-11:
> Should we put the Porter-Duff enum in vector API headers or somewhere more common?

Until we see a solid example of  Porter-Duff  can be used in other places as well,  I think we can keep them in a vector API header.

XuNeo, 2023-09-26:
Continue on PR lvgl#4528

closed 2023-09-26 by XuNeo

raveslave, 2024-02-08:
nice feature! 
- are there any performance numbers running lvgl+thorvg?
- any plans to support fixed point mcu's?

kisvegabor, 2024-02-08:
> are there any performance numbers running lvgl+thorvg?

I've just made a few measuements:
This line was rendered to 240x240 ARGB8888 canvas in 10ms on a Cortex-M7 200 MHz MCU. 
[image]

> any plans to support fixed point mcu's?

We use ThorVG (mostly) as it is, so unless the ThorVG adds fixed point math, I'm afraid it won't be supported.