lvgl/lvgl #4390: expected '=', ',', ';', 'asm' or '__attribute__' before lv_color_int_t'

hayschan, 2023-07-21:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Build error for the file `lv_color.h`. 

### To Reproduce

When I build my ESP-IDF project, I got the error for `lv_color_int_t`. The project lists LVGL as a managed component, and it pulls LVGL's latest version automatically.

When I comment the line of code out, it solves the issue. But that is probably not the best way to solve it.

I included LVGL in my `main.c` as follow:

```c
#include "lvgl.h"
```

### Expected behavior

```
managed_components\lvgl__lvgl\src\misc\lv_color.h:234:45: error: expected '=', ',', ';', 'asm' or '__attribute__' before lv_color_int_t'

  234 | typedef LV_CONCAT3(uint, LV_COLOR_SIZE, _t) lv_color_int_t;
```

### Screenshots or video

kisvegabor, 2023-07-24:
Hi,

What if you change this line to:
```c

typedef LV_CONCAT3(uint, 16, _t) lv_color_int_t;
```

hayschan, 2023-07-25:
Nope. It does not work. It gives the same error by changing from

```c
typedef LV_CONCAT3(uint, LV_COLOR_SIZE, _t) lv_color_int_t;
```

to

```c
typedef LV_CONCAT3(uint, 16, _t) lv_color_int_t;
```

## My other attempts

The error does not occur if I have either piece of code. This is by adding an `#undef` for `uint`.

### Attempt 1

```c
#undef uint
typedef LV_CONCAT3(uint, 16, _t) lv_color_int_t;
```

### Attempt 2 

```c
#undef uint
typedef LV_CONCAT3(uint, LV_COLOR_SIZE, _t) lv_color_int_t;
```

hayschan, 2023-07-25:
Just found the reason of the problem. It turns out this is an issue on my side, not an issue in LVGL.

## Reason of the problem

In one of my component, I have this line to define `uint` to an unsigned integer. As this happens in preprocessor, all the `uint` in the project is replaced as `unsigned int`. The scope is the whole project, including LVGL, even as managed component.

```c
#define uint unsigned int
```

This should cause issue to LVGL as the definition of `uint` has changed.

## Solution

I solve this my removing/commenting out the line:

```c
// #define uint unsigned int
```

closed 2023-07-25 by hayschan

kisvegabor, 2023-07-25:
Ah, good to hear that it's resolved!