lvgl/lvgl #4391: Setting LV_ATTRIBUTE_FAST_MEM spawns linker warnings on ESP-IDF v5.1

fronders, 2023-07-21:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I'm using ESP-IDF v5.1 (stable) with lvgl v8.3 from component manager. When I enable `Set IRAM as LV_ATTRIBUTE_FAST_MEM` linker produces a bunch of warnings about conflicting iram sections like this:

```
Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/misc/lv_math.c.obj
../managed_components/lvgl__lvgl/src/misc/lv_math.c:49:1: warning: ignoring attribute 'section (".iram1.3")' because it conflicts with previous 'section (".iram1.0")' [-Wattributes]
   49 | {
      | ^
In file included from ../managed_components/lvgl__lvgl/src/misc/lv_math.c:9:
../managed_components/lvgl__lvgl/src/misc/lv_math.h:47:31: note: previous declaration here
   47 | LV_ATTRIBUTE_FAST_MEM int16_t lv_trigo_sin(int16_t angle);
      |                               ^~~~~~~~~~~~
```

You can clearly see different sections are assigned to function and to its prototype because `IRAM_ATTR` expands with `_SECTION_ATTR_IMPL(".iram1", __COUNTER__)` macro that uses `__COUNTER__`. This was not the case for v5.0 supposedly because v5.1 updated to GCC12.2.0 and this was ignored

[This issue comment](https://github.com/espressif/esp-idf/pull/10773#issuecomment-1432602363) mentions what can be a fix for this problem:

> solution we would like to have is to remove IRAM_ATTR from function declarations in the header file, except for the static inline definitions

And indeed all ESP-IDF sources follow that practice

---

Here is the full output of the linkage:

```
esptool.py v4.6.2
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /home/fronders/esp/mp3-player/build/bootloader/bootloader.bin
[96/96] cd /home/fronders/esp/mp3-player/build/bootloader/esp-idf/esptool_py && /home/fronders/.espressif/python_env/idf5.1_py3.8_env/bin/python /home/fronders/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/fronders/esp/mp3-player/build/bootloader/bootloader.bin
Bootloader binary size 0x6eb0 bytes. 0x150 bytes (1%) free.
[728/1068] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/draw/sw/lv_draw_sw_gradient.c.obj
../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_gradient.c:298:1: warning: ignoring attribute 'section (".iram1.12")' because it conflicts with previous 'section (".iram1.8")' [-Wattributes]
  298 | {
      | ^
In file included from ../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_gradient.c:9:
../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_gradient.h:72:39: note: previous declaration here
   72 | LV_ATTRIBUTE_FAST_MEM lv_grad_color_t lv_gradient_calculate(const lv_grad_dsc_t * dsc, lv_coord_t range,
      |                                       ^~~~~~~~~~~~~~~~~~~~~
[743/1068] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/draw/sw/lv_draw_sw_img.c.obj
../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_img.c:47:1: warning: ignoring attribute 'section (".iram1.19")' because it conflicts with previous 'section (".iram1.17")' [-Wattributes]
   47 | {
      | ^
In file included from ../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_img.c:9:
../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.h:64:28: note: previous declaration here
   64 | LV_ATTRIBUTE_FAST_MEM void lv_draw_sw_img_decoded(struct _lv_draw_ctx_t * draw_ctx, const lv_draw_img_dsc_t * draw_dsc,
      |                            ^~~~~~~~~~~~~~~~~~~~~~
[770/1068] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/draw/sw/lv_draw_sw_blend.c.obj
../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_blend.c:105:1: warning: ignoring attribute 'section (".iram1.21")' because it conflicts with previous 'section (".iram1.11")' [-Wattributes]
  105 | {
      | ^
In file included from ../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.h:16,
                 from ../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_blend.c:9:
../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_blend.h:59:28: note: previous declaration here
   59 | LV_ATTRIBUTE_FAST_MEM void lv_draw_sw_blend_basic(struct _lv_draw_ctx_t * draw_ctx, const lv_draw_sw_blend_dsc_t * dsc);
      |                            ^~~~~~~~~~~~~~~~~~~~~~
../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_blend.c:242:1: warning: ignoring attribute 'section (".iram1.22")' because it conflicts with previous 'section (".iram1.19")' [-Wattributes]
  242 | {
      | ^
../managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_blend.c:29:35: note: previous declaration here
   29 | LV_ATTRIBUTE_FAST_MEM static void fill_normal(lv_color_t * dest_buf, const lv_area_t * dest_area,
[... 178 lines cut ...]
  647 | LV_ATTRIBUTE_FAST_MEM void lv_color_fill(lv_color_t * buf, lv_color_t color, uint32_t px_num);
      |                            ^~~~~~~~~~~~~
[1033/1068] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/misc/lv_math.c.obj
../managed_components/lvgl__lvgl/src/misc/lv_math.c:49:1: warning: ignoring attribute 'section (".iram1.3")' because it conflicts with previous 'section (".iram1.0")' [-Wattributes]
   49 | {
      | ^
In file included from ../managed_components/lvgl__lvgl/src/misc/lv_math.c:9:
../managed_components/lvgl__lvgl/src/misc/lv_math.h:47:31: note: previous declaration here
   47 | LV_ATTRIBUTE_FAST_MEM int16_t lv_trigo_sin(int16_t angle);
      |                               ^~~~~~~~~~~~
../managed_components/lvgl__lvgl/src/misc/lv_math.c:110:1: warning: ignoring attribute 'section (".iram1.4")' because it conflicts with previous 'section (".iram1.2")' [-Wattributes]
  110 | {
      | ^
../managed_components/lvgl__lvgl/src/misc/lv_math.h:87:28: note: previous declaration here
   87 | LV_ATTRIBUTE_FAST_MEM void lv_sqrt(uint32_t x, lv_sqrt_res_t * q, uint32_t mask);
      |                            ^~~~~~~
[1067/1068] Generating binary image from built executable
esptool.py v4.6.2
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /home/fronders/esp/mp3-player/build/mp3-player.bin
[1068/1068] cd /home/fronders/esp/mp3-player/build/esp-idf/esptool_py && /home...n_table/partition-table.bin /home/fronders/esp/mp3-player/build/mp3-player.bin
mp3-player.bin binary size 0x70280 bytes. Smallest app partition is 0x100000 bytes. 0x8fd80 bytes (56%) free.
```

kisvegabor, 2023-07-24:
Hi,

Thank you for opening this issue!

> solution we would like to have is to remove IRAM_ATTR from function declarations in the header file, except for the static inline definitions

Are we sure that the compiler in general will like having the `LV_ATTRIBUTE_FAST_MEM` only in the C files? Will they know how to call RAM functions without without having any attributes  in the prototypes? 

What do you think?

kisvegabor, 2023-07-24:
> OTOH if an attribute is mentioned only in header prototypes, wouldn't it create a new section on each inclusion of a header file because of the counter macro? So maybe keeping it in C files only isn't a bad idea after all, I doubt the compiler would have problems with C-files-only attributes since this work is done by the linker after the fact anyways

I agree with that. Do you have time to send a Pull request with the required changes?

fronders, 2023-07-24:
Sure, should I use release/v8.3 for that? 
Also maybe it would be reasonable to do to the same to all of the `LV_ATTRIBUTE_*` attributes?