lvgl/lvgl #4397: what happened to LV_OBJ_FLAG_OVERFLOW_VISIBLE

kdschlosser, 2023-07-23:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
`LV_OBJ_FLAG_OVERFLOW_VISIBLE` seems to be missing. Maybe it was removed? If it was removed is there a replacement?

kisvegabor, 2023-07-24:
It was really removed because it blocked a some optimization opportunities. 

You can set an extra draw size instead, which is more explicit (it's a little bit cumbersome now though):   
1. You need to add an `LV_EVENT_REFR_EXT_DRAW_SIZE` to the which should allow drawing out of the bounds
2. Inside the event call `lv_event_set_ext_draw_size(e, px);` where `px` is the extra draw area around the widget.

kdschlosser, 2023-07-25:
Oh that's just going to be a pain in the ass to do now.

XuNeo, 2023-07-25:
Do you plan to add it back? It has some real use cases:
- When image(E.g. analog time hand) rotating out of its parent(E.g. image in position 0,0), need this flag for its parent
- Making object as a pure container, usually needed when UI is complex and divided to several modules/layers

XuNeo, 2023-07-26:
[image]

I hope this example can explain it better.

This is a simple watchface UI. The number `04` in red rectangle is `month`. Here's the description of UI.
```xml
<DataItemImageNumber id="id_123" source="dateDay" space="0" align="left" rotation="76" leadingZero="true" totalDigits="2" ref="@DataItemImageNumber_in96qe_ref" />
....
<Layout ref="@id_123" x="439" y="174" _w="72" _h="52"/>
```

The attribute `DataItemImageNumber` is mapped to `lv_obj_t`. It has two image children rotating 76 degree, out of its parent's normal position/size. 
In this case, I simply add `LV_OBJ_FLAG_OVERFLOW_VISIBLE` flag to parent. Now the parent acts a container and decides its children's position but alllows children to draw outside.

Without`LV_OBJ_FLAG_OVERFLOW_VISIBLE `, I agree using `LV_EVENT_REFR_EXT_DRAW_SIZE ` can also solve it.

kdschlosser, 2023-07-26:
I just notice all of the rendering functions are gone from lv_canvas. How do we go about rendering to the canvas?

XuNeo, 2023-07-26:
> I just notice all of the rendering functions are gone from lv_canvas. How do we go about rendering to the canvas?

With help of `layer`, now there is no need to use special APIs for canvas. Just init the layer from canvas and call draw methods directly. Check examples like `lvgl/examples/widgets/canvas/lv_example_canvas_1.c`

```c
    lv_layer_t layer;
    lv_canvas_init_layer(canvas, &layer);

    lv_area_t coords_rect = {30, 20, 100, 70};
    lv_draw_rect(&layer, &rect_dsc, &coords_rect);
```

kdschlosser, 2023-07-26:
That is great and all if there was a function for drawing a polygon which there isn't.

kdschlosser, 2023-07-26:
so there is no longer a need to set up a buffer  and how does one set the size of the area being rendered to?

kisvegabor, 2023-07-27:
> That is great and all if there was a function for drawing a polygon which there isn't.

It's a different story. Polygon rendering is a complex thing and there are a lot of non trivial cases. Therefore we have decided to remove it from pure software rendering. Instead, polygon drawing will be supported via the [vector API](https://github.com/lvgl/lvgl/issues/4388).

> so there is no longer a need to set up a buffer and how does one set the size of the area being rendered to?

You can see here how it works: https://github.com/lvgl/lvgl/blob/master/examples/widgets/canvas/lv_example_canvas_3.c

kdschlosser, 2023-07-27:
how come `lv_draw_rect` doesn't conform to the same API as `lv_draw_line` and `lv_draw_triangle`?

kisvegabor, 2023-07-27:
I'm thinking about if we should add `coords` in `lv_draw_rect` into the `dsc`. So the short answer is: it's still not final :slightly_smiling_face:

kdschlosser, 2023-07-27:
keeping them separate i think is a better thing to do because it allows the user to create only a single dsc that can be used to draw multiple items.

C-Ackerman, 2023-08-15:
> It was really removed because it blocked a some optimization opportunities.
>
> You can set an extra draw size instead, which is more explicit (it's a little bit cumbersome now though):
>
> 1. You need to [...]

Hi Gabor,

I use LV_EVENT_REFR_EXT_DRAW_SIZE to test here. The effect of LV_OBJ_FLAG_OVERFLOW_VISIBLE cannot be achieved. The reason is that lv_obj_redraw() in lv_refr.c will cut off the rendering range of the child component. After trying to modify clip_coords_for_children, the child component can be rendered beyond the scope of the parent component.
Version v8.3

```
if(lv_obj_has_flag(obj, LV_OBJ_FLAG_OVERFLOW_VISIBLE)) {
        clip_coords_for_children  = *clip_area_ori;
    }
    else {
        if(!_lv_area_intersect(&clip_coords_for_children, clip_area_ori, &obj->coords)) {
            refr_children = false;
        }
    }
```

demo:
```
static void ext_event_cb(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);

    /*Provide some extra space for the value*/
    if(code == LV_EVENT_REFR_EXT_DRAW_SIZE) {
        lv_event_set_ext_draw_size(e, 20);
    } else if(code == LV_EVENT_DRAW_MAIN_END) {
        lv_obj_refresh_ext_draw_size(obj);
    }
}

void lv_demo_widgets(void)
{
    static lv_style_t style_base;
    lv_style_init(&style_base);
    lv_style_set_bg_color(&style_base, lv_palette_lighten(LV_PALETTE_GREY, 1));

    lv_obj_t * parent = lv_obj_create(lv_scr_act());   /*Create a parent object on the current screen*/
    lv_obj_set_size(parent, 200, 200);	                 /*Set the size of the parent*/
    lv_obj_set_pos(parent, 50, 50);	/*Move the parent. The child will move with it.*/
    lv_obj_add_style(parent, &style_base, 0);
    // lv_obj_add_flag(parent, LV_OBJ_FLAG_OVERFLOW_VISIBLE);
    lv_obj_add_event_cb(parent, ext_event_cb, LV_EVENT_ALL, NULL);
    lv_obj_refresh_ext_draw_size(parent);
    
    static lv_style_t style_chld;
    lv_style_init(&style_chld);
    lv_style_set_bg_color(&style_chld, lv_palette_lighten(LV_PALETTE_LIGHT_BLUE, 3));

    lv_obj_t * obj1 = lv_obj_create(parent);	         /*Create an object on the previously created parent object*/
    lv_obj_set_size(obj1, 250, 50);
    lv_obj_set_pos(obj1, 10, 10);
    lv_obj_add_style(obj1, &style_chld, 0);

    return;
}
```

[image]

kisvegabor, 2023-08-16:
Ahhh, you are right. It really won't work. :cry: 

Than let's figure out how to approach it. 

My concert with adding `LV_OBJ_FLAG_OVERFLOW_VISIBLE` back: 
> The main reason I've removed it is when we need to create a layer from a widget with LV_OBJ_FLAG_OVERFLOW_VISIBLE we don't know the size of the layer as it can be any large because the children can be anywhere.  

> I guess we can add it back somehow, but an infinite large draw area when LV_OBJ_FLAG_OVERFLOW_VISIBLE is enabled doesn't sound embedded friendly. It also mean that when a widget with LV_OBJ_FLAG_OVERFLOW_VISIBLE is invalidated its parent area needs to be invalidated (if the parent also has LV_OBJ_FLAG_OVERFLOW_VISIBLE we need to invalidate the grandparent).

I don't know what to do if a widget with `OVERFLOW_VISIBLE` is zoomed to 10%. Do we need to render a 10 times larger parent area to be sure that after zooming  to 10% everything will be visible?

Let's collect as many real life use cases as we can to see if how we can solve them.

kdschlosser, 2023-08-17:
Treat the children exactly the same as you do currently except allow the children to be visible if it is outside the rect of the parent.

kisvegabor, 2023-08-17:
Hmm, we can't do it automatically because e.g. if a button has shadow, you still might want to clip its content text to the border of the button. But if we add  `OVERFLOW_VISIBLE` flag back again it can really use the same extra draw size. Why didn't is came to my mind? It so obvious. I should sleep more :upside_down_face:

kdschlosser, 2023-08-17:
sometimes it is just a matter of having someone tell you something that triggers the thought on how to do it. This happens to me all the time. It's one of those moments where you smack your forehead.

There has to be a routine that is used to set the boundaries for a child object, That boundary test would get passed if the user is wanting to make the children visible if they are not within that boundary.

kisvegabor, 2023-08-22:
I've added back `LV_OBJ_FLAG_OVERFLOW_VISIBLE` in c7e8b8447.

kdschlosser, 2023-08-23:
It doesn't conflict with the new design at all??

kisvegabor, 2023-08-23:
FYI, this complex case is working and tested as well: https://github.com/lvgl/lvgl/commit/5d604b46abb732b99068a06b58cec56ba0b16677#diff-8e0c875d5e85aae9b9ff1a5d6fd0905f9a63938a2050c436c51cf076939c3c43

The red rectangle has 100 px of extra draw area, and with rotation some widgets are clipped there

C-Ackerman, 2023-08-25:
hi Gabor,
LV_OBJ_FLAG_OVERFLOW_VISIBLE needs to be used in conjunction with ext_draw_size_event_cb, which is different from the V8 version. Assuming there are 5 layers, the subcomponent is at the top and is larger than the bottom parent component. How does the bottom parent component get the ext draw size of the top subcomponent?

kisvegabor, 2023-08-25:
You need to set `LV_OBJ_FLAG_OVERFLOW_VISIBLE` on each layer. But it was required in v8 as well, wan't it?

bjsylvia, 2023-08-26:
In v8, setting `LV_OBJ_FLAG_OVERFLOW_VISIBLE` is only required for the parent. It is not necessary to set `ext_draw_size`. Perhaps [the refr part of the code](https://github.com/lvgl/lvgl/blob/9ec3b8706b098f60940cee6ac211d3cb3141aedf/src/core/lv_refr.c#L166-L173) should also be restored?

bjsylvia, 2023-08-26:
Furthermore, if the parent has the scrollable property, should we not allow the child to be drawn outside the parent's clipping area?

kisvegabor, 2023-08-27:
> In v8, setting LV_OBJ_FLAG_OVERFLOW_VISIBLE is only required for the parent. It is not necessary to set ext_draw_size. Perhaps [the refr part of the code](https://github.com/lvgl/lvgl/blob/9ec3b8706b098f60940cee6ac211d3cb3141aedf/src/core/lv_refr.c#L166-L173) should also be restored?

The equivalent part should be here:
https://github.com/lvgl/lvgl/blob/538bae5de8dc5d012188d48421168be634c9024b/src/core/lv_refr.c#L119-L124

> Furthermore, if the parent has the scrollable property, should we not allow the child to be drawn outside the parent's clipping area?

I think it's up to the user/use-case. If the content is small and not scrollable,  `OVERFLOW_VISIBLE` has no effect. If the content is large and scrollable you  you can decide wheter to enable or disable to `OVERFLOW_VISIBLE` flag case by case.

kisvegabor, 2023-09-11:
Do we need anything else here?

C-Ackerman, 2023-09-11:
There is still a problem. I have a dial here but it is a bit complicated. I have not found out the reason. As shown in the figure, the pointer is truncated. Currently, there is a problem with the area x1:206 x2:285 calculated by coords_tmp.

[image]

[image]

XuNeo, 2023-09-11:
@C-Ackerman Let's try to build up a test example firstly.

kisvegabor, 2023-09-11:
A code snippet to reproduce the issue would be really welcome :slightly_smiling_face:

C-Ackerman, 2023-09-19:
@kisvegabor 
I'm very sorry. This problem was caused by our internal modification of the image decoder. It has been solved. There is no problem with LV_OBJ_FLAG_OVERFLOW_VISIBLE.

kisvegabor, 2023-09-20:
I see, thank you for the feedback!

closed 2023-09-20 by kisvegabor