lvgl/lvgl #4398: External Signal Interrupt to the animation

CodeOn-ArK, 2023-07-23:
### Introduce the problem
Hi, I would like to interface external input devices such as buttons/sensors to my LVGL project which is based on RPi 4. What should be the correct procedure for this

### Examples and cases
interfacing LVLG based UI with external input devices (buttons/sensors)

### Suggested solution
NULL

C47D, 2023-07-23:
You can use LVGL input devices: [docs](https://docs.lvgl.io/8.3/overview/indev.html)

CodeOn-ArK, 2023-07-27:
> You can use LVGL input devices: [docs](https://docs.lvgl.io/8.3/overview/indev.html)

Don't we have anything more generic, which can be set to & used for any input, like adc input or uart message from another board?

kisvegabor, 2023-07-27:
How would you like to control the UI exactly with these external data? So far all real life use cases I've seen were doable with the 4 input devices types of LVGL.

CodeOn-ArK, 2023-07-28:
Well, In my use case, I am using an Arduino for interfacing with the sensors (all sorts of data acquisition) and It would be sending uart commands to beaglebone, which would be running the GUI.
 Ik we can simplify this scenario to fit the current methods of interfacing which lvgl provides,  but that won't be feasible in this project. 
1 way is to control gpios to send signals to the beaglebone which would read the signals as external buttons, but for huge number of commands, the number of gpios used will be high.

kisvegabor, 2023-07-28:
With the [BUTTON indev type](https://docs.lvgl.io/8.3/porting/indev.html#button) you can handle `my_btn_read` can return  the pressed button based on UART commands too.

CodeOn-ArK, 2023-07-28:
So you are saying to handle uart through the btn indev API right?

kisvegabor, 2023-07-31:
Yes, could it work for you?

CodeOn-ArK, 2023-07-31:
As of now, I used buttons for the projects, didn't test uart, but if you want I will test it after sometime

closed 2023-08-22 by lvgl-bot