lvgl/lvgl #4401: textbox update high cpu usage on esp32

cyberumut, 2023-07-24:
```
 lv_obj_t *textbox = lv_textarea_create(lv_scr_act());
	lv_obj_align(textbox, NULL, x, y);
	lv_obj_set_style_radius(textbox, 0, 0);
        lv_obj_set_size(textbox, sizex, sizey);
	lv_obj_set_style_bg_color(textbox, lv_color_make(255, 255, 255), LV_PART_MAIN);
	lv_textarea_set_text(textbox, "");    /*Set an initial text*/
	lv_textarea_set_align(textbox, LV_TEXT_ALIGN_LEFT); 
	lv_obj_set_style_border_opa(textbox,LV_STATE_DEFAULT, 0);
	

and my update textbox task

lv_textarea_set_text(textbox , str); 
```

3 textobx and update cpu %50 usage is this normal ?

[image]

kisvegabor, 2023-07-24:
How often do you update the text areas?

pete-pjb, 2023-07-24:
Hi @kisvegabor ,

This seems to be a duplication of [this](https://forum.lvgl.io/t/esp32-lvgl-high-cpu-usage-problem/12409/18) forum post, it might be worth closing this and continuing on the forum if you have anything to add your contribution would be appreciated... 🙂 

Kind Regards,

Pete

kisvegabor, 2023-07-25:
Okay, I've just commented in the forum.

closed 2023-07-25 by kisvegabor