lvgl/lvgl #4407: How to use lvgl jpg decoder with esp32 spiffs correctly?

edword01, 2023-07-26:
Device: ESP32s3 N8R8
already config spiffs correctly

### step
1、config lv_conf.h to enable ```LV_USE_SJPG``` and ```LV_USE_FS_STDIO``` , like this:
```
#define LV_USE_FS_STDIO 1
#if LV_USE_FS_STDIO
    #define LV_FS_STDIO_LETTER 'A'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_STDIO_PATH "/"         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_STDIO_CACHE_SIZE 4096    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif
```
2、test file system . like this:
```
lv_fs_file_t f;
    lv_fs_res_t res;
    res = lv_fs_open(&f, "A:spiffs/test.txt", LV_FS_MODE_RD);
    if(res != LV_FS_RES_OK)
        ESP_LOGE(TAG, "lv file err, %d", res);
    else
    {
        char buf[6];
        lv_fs_read(&f, buf, 4, NULL);
        ESP_LOGI(TAG, "%s", buf);
```
3、 then try to decode jpg file,  
```
lv_img_set_src(objpg, "A:spiffs/sc1.jpg");
```

### Result
it should be a picture there, but nothing happen, if i change to use C array of jpg, it display correctly.
how to fix that.?

edword01, 2023-07-26:
i also try to convert jpg to sjpg, then ```lv_img_set_src(objpg, "A:spiffs/sc1.sjpg");``` ,  it still don't work.

edword01, 2023-07-26:
lvgl doc shows: 
``` Currently only 16 bit image format is supported (TODO),  ```
what does this mean??  support jpg with RGB565 format?? how to generate this kind of jpg file?  PS software only support 8bit for each channel, that mean 3*8=24bit, so, my question is: how to use this jpg decoder now????

edword01, 2023-07-27:
may i ask when will this decoder add into lvgl ? any plan?

edword01, 2023-07-27:
> To be honest, the current SJPG decoder has some limitations and bug. I'll probably add a normal JPG decoder instead.
> 
> I suggest using PNG until that. 🙁

i try png decode, fail again, where to find the correct config ?? i just enable ```LV_USE_PGG```， and confg stack size of ui task: 240kb  ,  which larger than 240*240*4 bytes,  and config ```LV_FS_STDIO_CACHE_SIZE``` larger than 240KB

kisvegabor, 2023-07-27:
Please enable the [logging](https://docs.lvgl.io/master/porting/log.html) of LVGL to see what could be the problem.

PGNetHun, 2023-07-28:
Hello!
There is a bug in JPEG decoding indeed, since the driver architecture change (in which color formats are also changed): https://github.com/lvgl/lvgl/commit/124f9b0f9f045025360e9167ddcf8e7122a083a4

The image itself is decoded, but the place of image is black.

The problem occurs only when the `lv.img` object reads the JPEG file from file system (SPIFFS). 
When the JPEG is explicitly decoded into `LV_COLOR_FORMAT_NATIVE` array buffer, and we set the `lv.img` source to this decoded data, then the image is shown.

**The fix** is to change `LV_COLOR_FORMAT_RAW` to `LV_COLOR_FORMAT_NATIVE` everywhere in https://github.com/lvgl/lvgl/blob/master/src/libs/sjpg/lv_sjpg.c file.

Sorry not mentioning this earlier, I was not sure if it affected only me (I'm using only MicroPython binding, not the C language).

One way to unit test the correctness of image decoders (not only JPEG and PNG, but others also), to render images into an LVGL screen, and compare the rendered screen/image with a reference raw data file. But the unit test should set the display color format, and not the decoder target color format! So we can test, whether the image is really correctly decoded to the display color format (what is the problem in our case).
These color formats are most used in embedded systems (I assume): RGB565, RGB888, ARGB888

kisvegabor, 2023-07-31:
Hi @PGNetHun 

Thanks for the hit. 

> One way to unit test the correctness of image decoders (not only JPEG and PNG, but others also), to render images into an LVGL screen, and compare the rendered screen/image with a reference raw data file. 

Yeah, we need this for sure! We will add some updated to image decoding in general soon, as part of it I'll take care of the unit tests as well.

PGNetHun, 2023-08-20:
Hi @kisvegabor , @edword01 ,
Sorry, the color format change will also not fix this issue.
I tried to add tests for the PNG and JPEG decoders (and compare result screen with reference images), the PNG tests pass, but it seems the call to JPEG decoder is removed in the parallel rendering architecture commit: 
https://github.com/lvgl/lvgl/commit/f753265a799bdd910881238cb36f8d731f6d8727#diff-90f14ebcd1b2e071c69ef3ad4020cb675f2484b26db670b0e455b1aec6cabdb0L78  
In original `src/draw/lv_draw_img.c` file, line 78 there is a call to `decode_and_draw()`, which calls the decoder's `lv_img_decoder_read_line()` function if image is not decoded yet. 
This is missing from the new changes.
(PNG is rendered correctly because it decodes the image in `decoder_open()` function, and there is no `decoder_read_line()` function in `lv_png.c` file, like `lv_sjpg.c` has)

Image decoder (PNG & JPG) tests are added in my PR, the JPEG test is "ignored" till the issue is fixed: https://github.com/lvgl/lvgl/pull/4473

closed 2023-09-14 by lvgl-bot