lvgl/lvgl #4409: The Input device may not be rotated as the display assigned to it

asypost, 2023-07-26:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Recently I tried to port the LVGL to old 3DS console which has two screens and the bottom screen has touch support,but on old 3DS the screens should be rotated 270 degrees to display properly,But the touch layer did not need to do this.So I suggest to add a flag to indicates if the input device should be rotated as the screen which is assigned to it. 

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2023-07-26:
I think it shouldn't be a flag but a `touch_default_orientation` field.

However, this is the first time I see such a request and before adding a new feature we need to be sure it solves frequent issues. 

So what do you think about a aligning the orientation to the display in the `indev_read_cb` manually?

asypost, 2023-07-27:
I tried to do so in `indev_read_cb ` but find it hard, because the bottom screen of the old 3DS need to be rotated 270 degrees to display properly and in the `lv_indev.c` there are codes:
```
if (disp->driver->rotated == LV_DISP_ROT_180 || disp->driver->rotated == LV_DISP_ROT_270)
{
    data->point.x = disp->driver->hor_res - data->point.x - 1;
    data->point.y = disp->driver->ver_res - data->point.y - 1;
}
if (disp->driver->rotated == LV_DISP_ROT_90 || disp->driver->rotated == LV_DISP_ROT_270)
{
    lv_coord_t tmp = data->point.y;
    data->point.y = data->point.x;
    data->point.x = disp->driver->ver_res - tmp - 1;
}
```
I need to undo all the operations the above codes do which should not be there.I do not known whether it is a more common issue or not🤔.

kisvegabor, 2023-07-27:
I would say add the rotation in the `read_cb` to make the touch pad aligned with the display first. After the LVGL display rotation can be applied to to rotate both.

asypost, 2023-07-28:
Yeah,I did that.I didn't know lvgl rotated the touch device the same way as the associated display until I look into the lvgl source code,It's better to note this behavior in the document.

kisvegabor, 2023-07-28:
I've just updated the docs.

closed 2023-08-20 by lvgl-bot